espressif/esp-idf #12363: CMake Error at run_serial_tool.cmake:66 (message):

Pklug, 2023-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am getting the following error when trying to flash the ESP-IDF firmware to my ESP32 device:

CMake Error at run_serial_tool.cmake:66 (message):
[image]
Environment:

Operating system: Ubuntu 

on virtual machine ESP-IDF version: 5.1.1 ESP32 device: esp32c6-devkitc- v1.2

I have tried the following troubleshooting steps: Made sure that the ESP32 device is connected to my computer and in the correct mode to be flashed. Tried flashing the ESP-IDF firmware using the esptool.py command-line tool. Tried flashing the ESP-IDF firmware using a different IDE or build system. Tried flashing the ESP-IDF firmware to a different ESP32 device. I am still getting the error message. Question: What can I do to fix this error?
@shubhamdp @dhrishi @espressif-bot

dobairoland, 2023-10-09:
> on virtual machine

If you are trying to flash a device from a virtual machine then you have to make it available there. When you connect a device to a host machine, it is available there but not in the virtual machines. Usually, the virtual environment software has a setting page when the serial device can be forwarded to the virtual machine.

dobairoland, 2023-10-16:
Related hints with suggestions were added on the esptool side: https://github.com/espressif/esptool/commit/d61da7785e29ab8a926d1f9e24314b6ea9d3f933

dobairoland, 2023-10-23:
I'm closing this for inactivity.

closed 2023-10-23 by dobairoland

Pklug, 2023-11-08:
Hello @dobairoland I tried the recommended it is yet not working.
`[1/5] cd /home/ubu/esp/esp-idf/esp32c6-blink-led/build/esp-idf/esptool_py && /home/ubu/.espressif/python_env/idf5.3_py3.11_env/bin/python /home/ubu/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/ubu/esp/esp-idf/esp32c6-blink-led/build/partition_table/partition-table.bin /home/ubu/esp/esp-idf/esp32c6-blink-led/build/esp32c6-blink-led.bin
esp32c6-blink-led.bin binary size 0x224d0 bytes. Smallest app partition is 0x100000 bytes. 0xddb30 bytes (87%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/ubu/esp/esp-idf/esp32c6-blink-led/build/bootloader/esp-idf/esptool_py && /home/ubu/.espressif/python_env/idf5.3_py3.11_env/bin/python /home/ubu/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/ubu/esp/esp-idf/esp32c6-blink-led/build/bootloader/bootloader.bin
Bootloader binary size 0x5410 bytes. 0x2bf0 bytes (34%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /home/ubu/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/ubu/esp/esp-idf -D "SERIAL_TOOL=/home/ubu/.espressif/python_env/idf5.3_py3.11_env/bin/python;;/home/ubu/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c6" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/ubu/esp/esp-idf/esp32c6-blink-led/build -P /home/ubu/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32c6 -p Port -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 esp32c6-blink-led.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.dev2
Serial port Port

A fatal error occurred: Could not open Port, the port is busy or doesn't exist.
([Errno 2] could not open port Port: [Errno 2] No such file or directory: 'Port')

FAILED: CMakeFiles/flash /home/ubu/esp/esp-idf/esp32c6-blink-led/build/CMakeFiles/flash 
cd /home/ubu/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/ubu/esp/esp-idf -D "SERIAL_TOOL=/home/ubu/.espressif/python_env/idf5.3_py3.11_env/bin/python;;/home/ubu/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c6" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/ubu/esp/esp-idf/esp32c6-blink-led/build -P /home/ubu/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
`

dobairoland, 2023-11-09:
```
Serial port Port

A fatal error occurred: Could not open Port, the port is busy or doesn't exist.
([Errno 2] could not open port Port: [Errno 2] No such file or directory: 'Port')
```

Port is not a valid port name. Usually it is either `/dev/ttyUSB0` or `/dev/ttyACM0` (or some greater number than 0 in the name).

Zeni241, 2024-09-18:
In my case it was the wires I was using to connect USB-to-UART Bridge to my custom esp32 board.