espressif/esp-idf #12364: Seeking Clarification on ESP Thread Library Integration in ESP-IDF 4.4.x

Mikeadaneg, 2023-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,I hope this is the appropriate platform for my inquiry. Please accept my apologies if it is not.

Hardware Platform:

ESP32 as the host processor
SiLabs EFR32 as a radio co-processor
Codebase:

esp-idf 4.4.x ([GitHub Commit](https://github.com/espressif/esp-idf/commit/6a7d83af1984b93ebaefa7ca9be36304806c3dc8/))
I'm currently working on updating our codebase to esp-idf v4.4.5. In our older esp-idf codebase, we don't appear to have a link to the ESP Thread Library, and it seems that everything is built from source.

I've reviewed the README.rst in the repository, but I'm still missing some context:

Purpose of Pre-Compiled Binary:

What is the purpose of this pre-compiled binary? Why not just publish the source differences?
Custom Components/Openthread:

If we want to create custom components/openthread in our esp-idf project, what considerations should we keep in mind?
Future of Precompiled Binaries:

What is the intended future of the precompiled binaries? Are they going to be integrated into the mainline?
When I compare snapshots at the following URLs:

[Snapshot at esp-idf-v4.4.5](https://github.com/espressif/esp-thread-lib/tree/9a8d34d8f698cad2c9468468b473e26a3dda51b9)
[Current Master](https://github.com/espressif/esp-thread-lib/tree/ca32b6597397e2d7e50cf5cd54684fef112500e7)
I observe that the number of files tracked by the precompiled library is increasing (from 6 to 16 in this case). This raises concerns that the Espressif OpenThread ports might be diverging further from the upstream and potentially becoming (partially) closed source.

Is there any user guide or technical article that I might be missing, which could provide the context I need?

Thank you for your time and assistance.

chshu, 2023-10-07:
@Mikeadaneg Please refer to this similar issue: https://github.com/espressif/esp-thread-lib/issues/1#issuecomment-1751577684

chshu, 2023-10-24:
@Mikeadaneg Please reopen if any further questions.

closed 2023-10-24 by chshu