espressif/esp-idf #12365: ESP32 Access Point example issue

CKarlGrech, 2023-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My aim is to create an access point using the ESP32 using the ESP-IDF. As a start up I used the softAP example, however the softAP is configured for WPA3 and I only want password. 

When I deactivate "Enable WPA3-Personal" using the menu config, I am capable of seeing the SSID of the ESP32 from my phone with only the password, but when I  enter the password, the phone does not try to connect to the AccessPoint.

What can I do to create an access point o the ESP32, using only the password and the softAP example as a start point?

Thank you

baylf2000, 2023-10-07:
@CKarlGrech There are a lot of examples on the internet of using ESP32 devices as WiFi access points. Many of them use the Arduino system as it's much simpler to get started. Have you tried googling? You might get faster help that way. I don't want to post external links here as I'm not sure what is allowed.

Xiehanxin, 2023-10-07:
hi, @CKarlGrech could you please provide your IDF-version? Ihave just tried the the softAP example in IDF and it works https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c  or you can check a stable branch like release/4.4 but not master

CKarlGrech, 2023-10-07:
@baylf2000, yes I tried with Arduino and it worked as expected,but I would like to use the esp-idf. Thank you.

@Xiehanxin, I am currently using the master. Which version are you using, please? Thank you

Xiehanxin, 2023-10-07:
@CKarlGrech  maybe you can use the release/4.4 for try

CKarlGrech, 2023-10-07:
I tried with release/4.4 and it worked as expected. 

Thank you

CKarlGrech, 2023-10-07:
I tried with release/4.4 and it worked as expected.

Thank you.

I will know close the issue.

closed 2023-10-07 by CKarlGrech