espressif/esp-idf #12373: ESP-IDF ESP32 WPS Example cant connect to AP

Tomvdsch, 2023-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Im trying to run the ESP32 WPS example on my ESP_Devkitc_V4 but i never wants to connect to my AP. I tried this in Arduino at first but kept getting the ASSOC_LEAVE error followed by a 4WAY_HANDSHAKE_TIMEOUT error in a loop. Now that i tried it in the ESP-IDF the monitor keeps saying the following:
I (802) example_wps: start wps...
I (802) example_wps: WIFI_EVENT_STA_START
I (912) main_task: Returned from app_main()
I (13382) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (13382) wifi:state: init -> auth (b0)
I (13382) wifi:state: auth -> assoc (0)
I (13392) wifi:state: assoc -> run (10)
I (18402) wifi:state: run -> init (0)
I (18402) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (18412) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (18412) example_wps: WIFI_EVENT_STA_WPS_ER_FAILED
W (18512) wifi:sta_scan: STA is connecting, scan are not allowed!
I (20822) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (23232) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (23232) example_wps: Failed to connect!

Does anyone have any idea how i can solve this? I tried looking online but could not find anything to solve it.

kapilkedawat, 2023-10-09:
Hi @Tomvdsch, can you please enable wpa_supplicant logs and share those? Please use [this](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/Kconfig#L600) config option to enable that.

Also mention your IDF version and commit ID.

Tomvdsch, 2023-10-09:
[image]
I selected version 5.1.1(stable) in VSCode and if i execute the idf.py --version command i get v1.0.3

nishanth-radja, 2023-10-13:
@Tomvdsch can we also have sniffer capture during the connection.

AxelLin, 2023-10-19:
@Tomvdsch Any update regarding the comment in https://github.com/espressif/esp-idf/issues/12373#issuecomment-1761005022 ?

AxelLin, 2023-12-05:
@Tomvdsch 
Any update? BTW, what's your esp-idf version? (git describe --tags output?)

closed 2024-10-18 by Sherry616