espressif/esp-idf #12377: ESP32-S3 OTA "Warning: checksum mismatch between flashed and built applications" despite running image

alexanderturner, 2023-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

Warning: ESP32-S3 has no Chip ID. Reading MAC instead.

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Lovely Binary ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I'm writing an OTA image to the device over BLE. The BLE app breaks the app into sectors and performs a checksum on every 2000 byte sector as it writes it to Flash. On completion, the app runs the app without issue; however there is a warning when the OTA image is run: 

```
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 1a75c346f808303ac8b4f2b874f204795f25a4b2fe575d79b9966c91ce94aa8a
```

### What is the actual behavior?

Successfully running OTA image has checksum mismatch warning

### Steps to reproduce.

Unsure at this point

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3ae0,len:0x1f1c
load:0x403c9700,len:0xd30
load:0x403cc700,len:0x4428
SHA-256 comparison failed:
Calculated: d3a893eee9a1b6ed021e98e09cf617a313bae82d732a5dfd54f3ef0f163feb88
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c99a8
W (35) boot.esp32s3: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (61) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (61) boot: compile time 18:54:22
I (61) boot: chip revision: v0.1
I (64) boot.esp32s3: Boot SPI Speed : 80MHz
I (68) boot.esp32s3: SPI Mode       : DIO
I (73) boot.esp32s3: SPI Flash Size : 16MB
I (78) boot: Enabling RNG early entropy source...
I (83) boot: Partition Table:
I (87) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (102) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (109) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (117) boot:  3 ota_0            OTA app          00 10 00010000 004e0000
I (124) boot:  4 ota_1            OTA app          00 11 004f0000 004e0000
I (132) boot:  5 nvs_key          WiFi data        01 02 009d0000 00004000
I (139) boot:  6 storage          Unknown data     01 82 009d4000 0062c000
I (147) boot: End of partition table
I (152) boot: Enabled a check secure version of app for anti rollback
I (158) boot: Secure version (from eFuse) = 0
I (165) esp_image: segment 0: paddr=004f0020 vaddr=3c090020 size=270b0h (159920) map
I (191) esp_image: segment 1: paddr=005170d8 vaddr=3fc99a00 size=0374ch ( 14156) load
I (194) esp_image: segment 2: paddr=0051a82c vaddr=40374000 size=057ech ( 22508) load
I (201) esp_image: segment 3: paddr=00520020 vaddr=42000020 size=8e970h (584048) map
I (275) esp_image: segment 4: paddr=005ae998 vaddr=403797ec size=10168h ( 65896) load
I (286) esp_image: segment 5: paddr=005beb08 vaddr=600fe000 size=0003ch (    60) load
I (297) boot: Loaded app from partition at offset 0x4f0000
I (297) boot: Disabling RNG early entropy source...
I (308) cpu_start: Pro cpu up.
I (308) cpu_start: Starting app cpu, entry point is 0x403756c4
0x403756c4: pthread_mutex_unlock at /Users/aturner/esp/esp-idf/components/pthread/pthread.c:693

I (0) cpu_start: App cpu up.
I (323) cpu_start: Pro cpu start user code
I (323) cpu_start: cpu freq: 240000000 Hz
I (323) cpu_start: Application information:
I (326) cpu_start: Project name:     bms-main
I (331) cpu_start: App version:      d338d5e-dirty
I (336) cpu_start: Compile time:     Oct 10 2023 18:00:44
I (342) cpu_start: ELF file SHA256:  f0ab8259b74c1814...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 1a75c346f808303ac8b4f2b874f204795f25a4b2fe575d79b9966c91ce94aa8a
I (348) cpu_start: ESP-IDF:          v5.0.1-dirty
I (354) cpu_start: Min chip rev:     v0.0
I (359) cpu_start: Max chip rev:     v0.99
I (363) cpu_start: Chip rev:         v0.1
I (368) heap_init: Initializing. RAM available for dynamic allocation:
I (375) heap_init: At 3FCA1388 len 0002EC78 (187 KiB): D/IRAM
I (382) heap_init: At 3FCD8000 len 00011710 (69 KiB): D/IRAM
I (388) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (395) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (401) heap_init: At 600FE03C len 00001FC4 (7 KiB): RTCRAM
I (408) spi_flash: detected chip: gd
I (412) spi_flash: flash io: dio
I (427) sleep: Configure to isolate all GPIO pins in sleep state
I (427) coexist: coexist rom version e7ae62f
I (428) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```
```

### More Information.

I've tried reading the OTA partitions from flash, however the esptool command is dumping the entire partition, not just the bin so I'll need to trim the binary and unsure how.

KonstantinKondrashov, 2023-10-10:
Hi @alexanderturner!
This msg does not say that the image is not correct. The `Warning: checksum mismatch between flashed and built applications. Checksum of built application is 1a75c346f808303ac8b4f2b874f204795f25a4b2fe575d79b9966c91ce94aa8a` message means that the ELF file (sha256(elf_file)) in your project mismatches to this running image. This message is printed by the esp_idf_monitor tool, the tool reads the line `ELF file SHA256:  f0ab8259b74c1814...` and compares it to sha256 of the ELF file in the build folder. During building, the build system adds sha256 of the ELF file into the image (see the [app_elf_sha256](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#application-description)). There is a recommendation to save the ELF file to be able to get correct backtrace.

You can take a look at `ELF file SHA256` of an image by running `esptool.py image_info  -v 2 build/file.bin `.

alexanderturner, 2023-10-10:
Super helpful and thank you! Had no idea this came from esp-monitor. I'll close this as it's not a bug, but can we leave the issue in existence for other to find in future

Thank you @KonstantinKondrashov

closed 2023-10-10 by alexanderturner

oclyke, 2024-09-05:
This was helpful to me when I noticed the same. Thanks!

GaoJia123-long, 2025-04-01:
Thank you, I encountered this problem and saw that you solved it