espressif/esp-idf #12381: abort on mbedtls_sha256_update if inside critical section with acceleration on (sha512 also affected)

greenaddress, 2023-10-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 tdisplay s3

### Power Supply used.

USB

### What is the expected behavior?

I expect mbedtls_sha256_update to not cause panic

### What is the actual behavior?

mbedtls_sha256_update causes a panic/abort

### Steps to reproduce.

Run the code below to reproduce:

```
/*                                                                              
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD        
 *                                                                              
 * SPDX-License-Identifier: CC0-1.0                                             
 */                                                                             
                                                                                
#include <stdio.h>                                                              
#include <inttypes.h>                                                           
#include "sdkconfig.h"                                                          
#include "freertos/FreeRTOS.h"                                                  
#include "freertos/task.h"                                                      
                                                                                
#include <mbedtls/sha256.h>                                                     
                                                                                
/* tested using esp-idf 5.1.1 */                                                
#define SIZE_OF_DATA 256                                                                                                                                                                                           
                                                                                
static portMUX_TYPE resource_mutex;                                             
                                                                                
static void new_f(void* ignored) {                                              
    uint8_t hash_result[32];                                                    
    mbedtls_sha256_context ctx = {};                                            
    mbedtls_sha256_init(&ctx);                                                  
    portENTER_CRITICAL(&resource_mutex);                                        
    uint8_t* data = calloc(1, SIZE_OF_DATA);                                    
    mbedtls_sha256_update(&ctx, (const unsigned char *)data, SIZE_OF_DATA);     
    portEXIT_CRITICAL(&resource_mutex);                                         
    mbedtls_sha256_finish(&ctx, hash_result);                                   
    mbedtls_sha256_free(&ctx);                                                  
    free(data);                                                                 
}                                                                               
                                                                                
void app_main(void)                                                             
{                                                                               
    spinlock_initialize(&resource_mutex);                                       
    new_f(NULL);                                                                
    vTaskSuspend(NULL);                                                         
} 
```

### Debug Logs.

```plain
I (867) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (867) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (868) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (868) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (869) spi_flash: detected chip: winbond
I (869) spi_flash: flash io: qio
I (869) app_start: Starting scheduler on CPU0
I (869) main_task: Started on CPU0
I (869) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (869) main_task: Calling app_main()

abort() was called at PC 0x4037680b on core 0
0x4037680b: lock_acquire_generic at /COMPONENT_NEWLIB_DIR/locks.c:130

Backtrace: 0x403753de:0x3fc96cb0 0x40379219:0x3fc96cd0 0x4037d8b7:0x3fc96cf0 0x4037680b:0x3fc96d70 0x40376925:0x3fc96db0 0x42007fa1:0x3fc96dd0 0x42007fbb:0x3fc96df0 0x420070bf:0x3fc96e10 0x42006fa3:0x3fc0x403753de: panic_abort at /COMPONENT_ESP_SYSTEM_DIR/panic.c:452

0x40379219: esp_system_abort at /COMPONENT_ESP_SYSTEM_DIR/port/esp_system_chip.c:84

0x4037d8b7: abort at /COMPONENT_NEWLIB_DIR/abort.c:38

0x4037680b: lock_acquire_generic at /COMPONENT_NEWLIB_DIR/locks.c:130

0x40376925: _lock_acquire_recursive at /COMPONENT_NEWLIB_DIR/locks.c:158

0x42007fa1: esp_crypto_ds_lock_acquire at /COMPONENT_ESP_HW_SUPPORT_DIR/port/esp32s3/esp_crypto_lock.c:31

0x42007fbb: esp_crypto_sha_aes_lock_acquire at /COMPONENT_ESP_HW_SUPPORT_DIR/port/esp32s3/esp_crypto_lock.c:49

0x420070bf: esp_sha_acquire_hardware at /COMPONENT_MBEDTLS_DIR/port/sha/dma/sha.c:118

0x42006fa3: mbedtls_sha256_update at /COMPONENT_MBEDTLS_DIR/port/sha/dma/esp_sha256.c:168
```

### More Information.

if the data being hashed is smaller this doesn't happen.

also on esp32 vanilla this does not happen, with or without acceleration.

greenaddress, 2023-10-11:
For what is worth, this doesn't happen on esp32 with sha acceleration, just on esp32s3 with Sha acceleration

Harshal5, 2023-10-12:
Hello @greenaddress ,

Thank you for reporting the issue.
I am able to reproduce the issue on my end. Attaching a small patch [avoid_acquiring_recursive_locks.patch](https://github.com/espressif/esp-idf/files/12877789/avoid_acquiring_recursive_locks.patch), could you apply it and check if the issue gets resolved?
Will raise an internal MR to fix the issue.

greenaddress, 2023-10-12:
@Harshal5 unfortunately even with the patch the problem still presents itself in a multi task/cores environment

see this code to reproduce (with the patch applied on top of v5.1.1)
```
#include "freertos/FreeRTOS.h"                                                  
#include "freertos/task.h"                                                      
#include "sdkconfig.h"                                                          
#include <inttypes.h>                                                           
#include <stdio.h>                                                              
                                                                                
#include <esp_log.h>                                                            
                                                                                
#include <mbedtls/sha256.h>                                                     
                                                                                
/* tested using esp-idf 5.1.1 */                                                
#define SIZE_OF_DATA 256                                                        
                                                                                
static portMUX_TYPE resource_mutex;                                             
static char* TAG = "TEST_MULTI MUTEX";                                          
                                                                                
static void new_f(void* ignored)                                                
{                                                                               
    for (;;) {                                                                  
        uint8_t hash_result[32];                                                
        mbedtls_sha256_context ctx = {};                                        
        mbedtls_sha256_init(&ctx);                                              
        mbedtls_sha256_starts(&ctx, 0);                                         
        portENTER_CRITICAL(&resource_mutex);                                    
        uint8_t* data = calloc(1, SIZE_OF_DATA);                                
        mbedtls_sha256_update(&ctx, (const unsigned char*)data, SIZE_OF_DATA);  
        portEXIT_CRITICAL(&resource_mutex);                                     
        mbedtls_sha256_finish(&ctx, hash_result);                               
        mbedtls_sha256_free(&ctx);                                              
        free(data);                                                             
    }                                                                           
}                                                                               
                                                                                
void app_main(void)                                                             
{                                                                               
    spinlock_initialize(&resource_mutex);                                       
    xTaskCreatePinnedToCore(new_f, "appcpu", 5 * 1024, NULL, tskIDLE_PRIORITY + 1, NULL, 0);
    xTaskCreatePinnedToCore(new_f, "appcpu", 5 * 1024, NULL, tskIDLE_PRIORITY + 1, NULL, 1);
    vTaskSuspend(NULL);                                                         
}
```

It results in this:
```
abort() was called at PC 0x4037665f on core 0
0x4037665f: lock_acquire_generic at /COMPONENT_NEWLIB_DIR/locks.c:130

Backtrace: 0x403752ae:0x3fc99670 0x403788e9:0x3fc99690 0x4037cf87:0x3fc996b0 0x4037665f:0x3fc99730 0x40376769:0x3fc99770 0x42007981:0x3fc99790 0x4200799b:0x3fc997b0 0x42006acf:0x3fc997d0 0x420069b3:0x3fc0x403752ae: panic_abort at /COMPONENT_ESP_SYSTEM_DIR/panic.c:452

0x403788e9: esp_system_abort at /COMPONENT_ESP_SYSTEM_DIR/port/esp_system_chip.c:84

0x4037cf87: abort at /COMPONENT_NEWLIB_DIR/abort.c:38

0x4037665f: lock_acquire_generic at /COMPONENT_NEWLIB_DIR/locks.c:130

0x40376769: _lock_acquire at /COMPONENT_NEWLIB_DIR/locks.c:154

0x42007981: esp_crypto_ds_lock_acquire at /COMPONENT_ESP_HW_SUPPORT_DIR/port/esp32s3/esp_crypto_lock.c:31

0x4200799b: esp_crypto_sha_aes_lock_acquire at /COMPONENT_ESP_HW_SUPPORT_DIR/port/esp32s3/esp_crypto_lock.c:49

0x42006acf: esp_sha_acquire_hardware at /COMPONENT_MBEDTLS_DIR/port/sha/dma/sha.c:118

0x420069b3: mbedtls_sha256_update at /COMPONENT_MBEDTLS_DIR/port/sha/dma/esp_sha256.c:168

```

igrr, 2023-10-12:
In general, RTOS blocking functions (such as the mutex functions used by `_lock_acquire`) can't be used inside critical sections. I don't think what you are trying to do can be made to work. (The technical reason is: the mechanism which blocking RTOS functions use to block the task and schedule a different task relies on interrupts. When the interrupts are disabled, it is not possible to perform a context switch.)

If you replace the critical section with an RTOS mutex, it should work.

greenaddress, 2023-10-12:
@igrr thank you, i replaced all instances of blocking/spinlock mutex to rtos ones and so far things seem ok. I will test more and provide feedback if I find anything. Up to you if you want to keep the issue open for now or close and reopen if needed; cheers

Harshal5, 2023-10-13:
Thanks @igrr for the help!

@greenaddress closing the issue for now, please feel free to re-open if needed.

closed 2023-10-13 by Harshal5