espressif/esp-idf #12383: ESP HTTP Client digest auth uses MD5 even if server requests SHA-256

christophhebeisen, 2023-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Tested on v5.1.1

I am using the ESP HTTP Client to make a request to a server that uses digest authentication with SHA-256. The Authorization: header in the followup request always uses algorithm="MD5", no matter what algorithm the server's WWW-Authenticate: header requested. While some servers tolerate this behaviour (notably httpbin.org), the one I am working with does not. I believe that this is a bug in the ESP HTTP Client.

The sequence of requests and responses is shown below:

Initial request:
```
GET /digest-auth/auth/user/passwd/SHA-256/never HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0
```

401 response indicating `algorithm=SHA-256`:
```
HTTP/1.1 401 UNAUTHORIZED
Date: Wed, 11 Oct 2023 03:07:40 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 0
Connection: keep-alive
Server: gunicorn/19.9.0
WWW-Authenticate: Digest realm="me@kennethreitz.com", nonce="f5bd5c31bf3ce86f1d77d5a5fe01962b5d37e23870fa0a36aaf25c89722ad485", qop="auth", opaque="db0c5ba3eebb3f2cb88bac808dcf7fe79b7a0733f54913244e3ac52bbedcd013", algorithm=SHA-256, stale=FALSE
Set-Cookie: stale_after=never; Path=/
Set-Cookie: fake=fake_value; Path=/
Access-Control-Allow-Origin: *
Access-Control-Allow-Credentials: true
```

Request incorrectly using `algorithm="MD5"`
```
GET /digest-auth/auth/user/passwd/SHA-256/never HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0
Authorization: Digest username="user", realm="me@kennethreitz.com", nonce="f5bd5c31bf3ce86f1d77d5a5fe01962b5d37e23870fa0a36aaf25c89722ad485", uri="/digest-auth/auth/user/passwd/SHA-256/never", algorithm="MD5", response="b7308f1a82e0b40473b22fc562c44d3c", qop=auth, nc=00000001, cnonce="f851c0cf5c4d830a", opaque="db0c5ba3eebb3f2cb88bac808dcf7fe79b7a0733f54913244e3ac52bbedcd013"
```

hmalpani, 2023-10-19:
Hello @christophhebeisen 
Thanks for reporting the issue.
I have created an internal MR for fixing this issue. I am sharing the patch with you. Can you please try it and check if it works for you?

Thanks!
[IDFGH_11217.txt](https://github.com/espressif/esp-idf/files/13043044/IDFGH_11217.txt)

christophhebeisen, 2023-11-04:
Hello Harshit,

Thank you for your response and my apologies for the delayed response - I was traveling and didn't have a chance to test the code until now. Sadly, I am not familiar with how I can properly test the change in ESP IDF but I tested the header generation by the code and compared the result against the expected outcome. I was able to confirm that the code generates the correct response.

I did notice that the header is slightly different from what RFC-7616 indicates (see examples in [https://datatracker.ietf.org/doc/html/rfc7616#section-3.9.1](https://datatracker.ietf.org/doc/html/rfc7616#section-3.9.1)) - the _algorithm_ parameter is supposed to be an unquoted string (it is quoted in the header generated by this code) and the _cnonce_ parameter is supposed to be quoted (it is not quoted as generated by this code). I don't know if this will make a difference in practice when used with any real web server though.

Cheers,
Christoph

christophhebeisen, 2023-11-04:
Sorry, closed accidentally. Please see comment above.

christophhebeisen, 2023-11-11:
Hello @hmalpani ,

I was now able to properly test the code - in addition to the patch you provided above, three additional patches were required. I am attaching those patches to this comment:

1. As explained in my previous comment, quoting in the client's `Authorization:` header did not agree with RFC. The `algorithm` parameter value is not supposed to be quoted, while the `cnonce` value is supposed to be quoted. The first patch fixes this.
2. The parsing of the server's `WWW-Authenticate:` header relied on `algorithm` not to be the last parameter i.e. it would only work if the value was followed by a comma instead of the end of the line. The second patch addresses this problem by  accepting the rest of the line as the parameter value if no comma is found in the string anywhere after `algorithm=`.
3. The `uri` parameter in the client's `Authorization:` header needs to contain the entire URI, including any query parameters. It was only set to the path and as such, authentication would only work if there were no query parameters. The third patch sets the `uri` parameter to only the path if there are no query parameters but adds in the query if present.

The final code with all four patches (i.e. yours plus the three above) applied was tested against [http://httpbin.org/digest-auth/auth/user/passwd/SHA-256/never](http://httpbin.org/digest-auth/auth/user/passwd/SHA-256/never) and other HTTP servers.

Please let me know if you have any questions.

Cheers,
Christoph

[1-fixHeaderParameterQuoting_patch.txt](https://github.com/espressif/esp-idf/files/13324262/1-fixHeaderParameterQuoting_patch.txt)
[2-fixAlgorithmAsLastParameter_patch.txt](https://github.com/espressif/esp-idf/files/13324266/2-fixAlgorithmAsLastParameter_patch.txt)
[3-includeQueryInUri_patch.txt](https://github.com/espressif/esp-idf/files/13324267/3-includeQueryInUri_patch.txt)

christophhebeisen, 2023-11-17:
@hmalpani please let me know if you have seen my comments above - since I accidentally closed (but then reopened) this issue, I am not sure if you are still getting notified or if I am sending messages nowhere. If I don't hear back I'll need to open a new issue - sorry...

hmalpani, 2023-11-17:
Hello @christophhebeisen 
Thanks for your response. I have already fixed some of the issues you addressed above. I will verify the remaining fixes and let you know of any updates.
Thanks!

closed 2023-12-18 by espressif-bot