espressif/esp-idf #12385: HTTP Client doesn't honor esp_http_client_config_t.if_name

rbev, 2023-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.1.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

LILYGO T-INTERNET-COM

### Power Supply used.

USB

### What is the expected behavior?

Network interfaces configured:

* Ethernet
* PPP (over LTE)

When these are set up I expect to be able to create a http request over the PPP connection by providing the interface name

Below is a simplified example of the call to the http client: 

```c++

    struct ifreq ifr;
    esp_netif_get_netif_impl_name(item.netif, ifr.ifr_name);
   
    esp_http_client_config_t config = {
        .url = "https://a-temporary-funtion-app.azurewebsites.net/api/my-ip",
        .method = item.verb,
        .disable_auto_redirect = true,        
        // event handler code remove for brevity
        //.event_handler = 
        .crt_bundle_attach = esp_crt_bundle_attach,
        .if_name = &ifr
    };
   
    ESP_LOGI(TAG, "Using IF: %s", config.if_name->ifr_name);

    auto response = do_http_request(config);
   
    ESP_LOGI(TAG, "%s IP: %s", config.if_name->ifr_name, response);
```

I would expect this code to output the external internet ip addresses of the two interfaces (that's what the endpoint returns)

### What is the actual behavior?

However if i call that http code from my main task for my two network devices i get this:

```
I (62976) MyAPP: Using IF: en1
I (78803) MyAPP: GET,en1 (200) https://a-temporary-funtion-app.azurewebsites.net/api/my-ip
I (78806) MyAPP: en1 IP: 49.255.51.242
I (78866) MyAPP: Using IF: pp2
I (80938) MyAPP: GET,pp2 (200) https://a-temporary-funtion-app.azurewebsites.net/api/my-ip
I (80940) MyAPP: pp2 IP: 49.255.51.242
```

If i unplug the ethernet cable iand call it just for the PPP interface i get:

```
I (78866) MyAPP: Using IF: pp2
I (80938) MyAPP: GET,pp2 (200) https://a-temporary-funtion-app.azurewebsites.net/api/my-ip
I (80940) MyAPP: pp2 IP: 1.126.105.33
```

### Steps to reproduce.

Make a http request using the non-default adapter when using multiple network devices

### More Information.

An additional note is that specifying the `if_name` appears to affect DNS requests, when providing the `if_name` all dns requests begin failing unless i also configure the interfaces with fallback DNS servers.

rbev, 2023-10-30:
has this issue been looked into?
there appears to be no triage on it after 3 weeks

david-cermak, 2023-10-30:
Hi @rbev 

There's been a bug related to interface binding, which caused the interface name parameter to be ignored on SSL transports, but worked correctly on TCP. It's been fixed in `v5.1.2`.

Since that you're on `v5.1.1` and using `https`, you might be experiencing this issue. You can cherry-pick https://github.com/espressif/esp-idf/commit/1363566b73c5b9d69c213eb9f6e7453ed2dd4e5b to quickly check if it's the case

david-cermak, 2023-11-28:
1363566b73 was merged to release branch and included in `v5.1.2`

closed 2023-11-28 by david-cermak