espressif/esp-idf #12387: Issue with MDNS v1.2.1

krupis, 2023-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect the mDNS 1.2.1 to work exactly the same as 1.2.0. I did not have any issues with the 1.2.0 as I can reach my webserver by using mDNS instance name.

### What is the actual behavior?

mDNS not working when updated to the latest version (1.2.1) due to ERR_NAME_NOT_RESOLVED error:
[image]

It loads for about 15 seconds and then returns an error:
[image]

I am able to capture mDNS packets using wire shark but cannot figure out why any browser fails to connect to it:
[image]

### Steps to reproduce.

Change the version of the mdns component in idf_component.yml from 1.2.0 to 1.2.1 and rebuild the project.

### Debug Logs.

```plain
N/A
```

### More Information.

We use the following function to setup the mDNS:
```
static void wifi_mdns_init(void) {
    esp_err_t err = ESP_OK;
    err           = mdns_init();
    if (err != ESP_OK) {
        TraceErrWIFI("mDNS INIT: FAILED (%s)", esp_err_to_name(err));
        return;
    }
    err = mdns_hostname_set(CONFIG_LWIP_LOCAL_HOSTNAME);
    if (err != ESP_OK) {
        TraceErrWIFI("HOSTNAME: FAILED (%s)", esp_err_to_name(err));
        return;
    }
    err = mdns_instance_name_set(CONFIG_LWIP_LOCAL_HOSTNAME);
    if (err != ESP_OK) {
        TraceErrWIFI("INSTANCE: FAILED (%s)", esp_err_to_name(err));
        return;
    }
    err = mdns_service_add("XXXXXXX-XXXXX", "_http", "_tcp", 80, NULL, 0);
    if (err != ESP_OK) {
        TraceErrWIFI("SERVICE: FAILED (%s)", esp_err_to_name(err));
        return;
    }
}
```

[image]

For now, I am using mDNS version 1.2.0 but I am just really curious what could be an issue with the 1.2.1 and what could cause this failure. Any ideas are appreciated!

gabsuren, 2023-10-18:
@krupis thank you for reporting. 
As I understand, you want to configure mdns over the WIFI interface, right?
The problem, as you noted, is the absence of `CONFIG_ESP_WIFI_ENABLED` in your `sdkconfig.h`. This is needed to register `mdns` event handlers.
Could you please check if `CONFIG_ESP_WIFI_ENABLED` is enabled in the `sdkconfig` (this file used to generate `sdkconfig.h`) file? If it's not, it should be enabled so that mdns can register events over the `WIFI` interface.

krupis, 2023-10-18:
@gabsuren 
Thanks for response.

```CONFIG_ESP_WIFI_ENABLED``` also does not exist in my sdkconfig. I am using esp-idf version 5.0.1

gabsuren, 2023-10-18:
After createing `sdkconfig.defaults`, please executed `idf.py reconfigure` or remove `skconfig` and call `idf.py build/menuconfig` again

gabsuren, 2023-10-18:
Updated: The issues is coming that `CONFIG_ESP_WIFI_ENABLED` is not exists in `esp-idf 5.0.1`. I will fix it soon. 
For quick fix there are a few options. 
1. Revert this commit `24f7031`  : fix(mdns): fix build issue if CONFIG_ESP_WIFI_ENABLED disabled
2. Add new config in `Kconfig` 
[image]
3. Use latest `ESP-IDF`

Meanwhile I will fix the issue for official branch

krupis, 2023-10-18:
I can confirm that adding ```ESP_WIFI_ENABLED ``` in Kconfig fixed the issue.

closed 2023-11-01 by gabsuren