espressif/esp-idf #12388: HCI initilization for esp32c6

pavan-coding, 2023-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

// NimBLE Client - Scan

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_nimble_hci.h"
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "services/gap/ble_svc_gap.h"
#include "sdkconfig.h"

char *TAG = "BLE Client Scan";
uint8_t ble_addr_type;
void ble_app_scan(void);

// BLE event handling
static int ble_gap_event(struct ble_gap_event *event, void *arg)
{
    struct ble_hs_adv_fields fields;

    switch (event->type)
    {
    // NimBLE event discovery
    case BLE_GAP_EVENT_DISC:
        ESP_LOGI("GAP", "GAP EVENT DISCOVERY");
        ble_hs_adv_parse_fields(&fields, event->disc.data, event->disc.length_data);
        if (fields.name_len > 0)
        {
            printf("Name: %.*s\n", fields.name_len, fields.name);
        }
        break;
    default:
        break;
    }
    return 0;
}

void ble_app_scan(void)
{
    printf("Start scanning ...\n");

    struct ble_gap_disc_params disc_params;
    disc_params.filter_duplicates = 1;
    disc_params.passive = 0;
    disc_params.itvl = 0;
    disc_params.window = 0;
    disc_params.filter_policy = 0;
    disc_params.limited = 0;

    ble_gap_disc(ble_addr_type, BLE_HS_FOREVER, &disc_params, ble_gap_event, NULL);
}

// The application
void ble_app_on_sync(void)
{
    ble_hs_id_infer_auto(0, &ble_addr_type); // Determines the best address type automatically
    ble_app_scan();                          
}

// The infinite task
void host_task(void *param)
{
    nimble_port_run(); // This function will return only when nimble_port_stop() is executed
}

void app_main()
{
    nvs_flash_init();                               // 1 - Initialize NVS flash using
    esp_nimble_hci_and_controller_init();           // 2 - Initialize ESP controller
    nimble_port_init();                             // 3 - Initialize the controller stack
    ble_svc_gap_device_name_set("BLE-Scan-Client"); // 4 - Set device name characteristic
    ble_svc_gap_init();                             // 4 - Initialize GAP service
    ble_hs_cfg.sync_cb = ble_app_on_sync;           // 5 - Set application
    nimble_port_freertos_init(host_task);           // 6 - Set infinite task
}
this code work fine for other esp devices other than esp32c6 do esp32c6 don't need any hci intilization even the document provide the initilization of hci [link](https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32c6/api-reference/bluetooth/nimble/index.html) but i am unable to import it can any one tell the reason

rahult-github, 2023-10-11:
Hi @pavan-coding 

> esp_nimble_hci_and_controller_init(); // 2 - Initialize ESP controller 

This step is no more needed, starting 5.x , the mentioned API is removed. Details for this can be found [here](https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32c6/migration-guides/release-5.x/5.0/bluetooth-low-energy.html?highlight=esp_nimble_hci_and_controller_init).

You can refer to existing blecent example [here](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blecent/main/main.c#L903) to see a working reference main function.

rahult-github, 2023-11-20:
Closing this issue. Feel free to reopen in case of further queries.

closed 2023-11-20 by rahult-github