espressif/esp-idf #12389: Wrong SDK version read, when building binaries in CI pipeline

bhuvanchandra, 2023-10-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

when using esp_get_idf_version expected version is "v5.1.1"

### What is the actual behavior?

when using esp_get_idf_version expected version is "HEAD-HASH-NOTFOUND"

### Steps to reproduce.

use esp-idf docker image for building the firmware binaries and call the esp_get_idf_version method.

### More Information.

When building the firmware binaries through the CI pipeline, the ESP-IDF version read via the API is incorrect. Instead of the expected "v5.1.1", it reads "HEAD-HASH-NOTFOUND". Upon verifying, the Docker image for the esp-idf doesn't explicitly check out to the release tag, which is necessary for the API to retrieve the correct version from the git tag.

I'm using [this](https://github.com/espressif/esp-idf-ci-action) action, v5.1.1 docker tag.

fhrbata, 2023-10-16:
Hello @bhuvanchandra , 

I've tried using the v5.1.1 docker image locally and also with the `esp-idf-ci-action`, but I cannot reproduce the behavior you are describing. My testing repository for this can be found [here](https://github.com/fhrbata/hello_world). It's the `hello_world` example from `esp-idf`  with dispatch action using `esp-idf-ci-action`.  I've downloaded the [artifacts](https://github.com/fhrbata/hello_world/actions/runs/6532320139) from the dispatch workflow and flash them into `esp32s3`.

1. Download the artifacts from https://github.com/fhrbata/hello_world/actions/runs/6532320139
2. Flashed with `esptool.py  -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 hello_world.bin`
3. check with `screen` that the output contains expected version `screen /dev/ttyUSB0 115200`

The output is contains
```
> esp_idf_version: v5.1.1
Restarting in 6 seconds...
```

TBH I'm quite not sure how the ``HEAD-HASH-NOTFOUND can get there. IIUC [esp_get_idf_version](https://github.com/espressif/esp-idf/blob/v5.1.1/components/esp_system/port/esp_system_chip.c#L77) returns just `IDF_VER`, which is defined in [cmake](https://github.com/espressif/esp-idf/blob/v5.1.1/tools/cmake/build.cmake#L79) and added to each compilation unit. This can be seen in the `build/compile_commands.json` where you should see something like `-DIDF_VER=\\\"v5.1.1\\\"`.

If I'm reading the cmake's [git_describe](https://github.com/espressif/esp-idf/blob/v5.1.1/tools/cmake/third_party/GetGitRevisionDescription.cmake#L89) function right, the only case where the `HEAD-HASH-NOTFOUND` can end up in `IDF_VER` is
1. The `git describe` fails [here](https://github.com/espressif/esp-idf/blob/v5.1.1/tools/cmake/third_party/GetGitRevisionDescription.cmake#L60)
2. the `HEAD` does not exists in `.git` dir [here](https://github.com/espressif/esp-idf/blob/v5.1.1/tools/cmake/third_party/GetGitRevisionDescription.cmake#L73)

Are you using the stock docker image provided by espressif `espressif/idf:v5.1.1` or you have some customizations? Maybe try to double check that the `IDF_PATH` is actually a git working tree. Maybe you are using just a copy without `.git` directory in your CI . I can reproduce this if I delete the `.git` directory, in which case the `git-describe` of course fails and `IDF_VER` contains `HEAD-HASH-NOTFOUND`.

Note the `v5.1.1` tag does not have to checked out for `git describe` to work and IMHO this is not related to your problem. 

HTH, but this doesn't seem like a  bug in `esp-idf`, `v5.1.1` docker image or the `esp-idf-ci-action`.

Thank you

bhuvanchandra, 2023-10-17:
Hello @fhrbata, thanks for looking into the issue.

I tried again, and the problem appears to be related to permissions. With the default root user, I can see the IDF version correctly. However, when running the docker image as a non-root user, the git describe command fails with the error: "fatal: detected dubious ownership in the repository at /opt/esp/idf/."

In my workflow, I need to run the following command to access the binary artifacts.

```
docker run --rm -u "$(id -u):$(id -g)" -e CCACHE_DIR=/tmp/.ccache -v $PWD:/oks_robot_connect -w /fw_app/ espressif/idf:v5.1.1 idf.py build
  cd fw_app/build
  cat version.txt
  zip fw_app.zip \
  bootloader/bootloader.bin \
  partition_table/partition-table.bin \
    ota_data_initial.bin \
    fw_app.bin \
    fw_app.elf \
    flasher_args.json
```

 This is necessary for committing and subsequently creating a PR. It's important for me to access these binaries as a standard user, not as root. I'll review and address the permission issues in my workflow.

fhrbata, 2023-10-17:
Hello @bhuvanchandra , maybe you can look into [safe.directory](https://git-scm.com/docs/git-config#Documentation/git-config.txt-safedirectory) git config option. I think setting it for your repo might help here. Thank you

closed 2023-11-07 by espressif-bot