espressif/esp-idf #12393: OTA non-app partition

law-ko, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

What is the best recommended way to update a partition that is non app partition? For example, the RESTful server `www` partition via OTA.

Thank you

hmalpani, 2023-10-18:
Hello @law-ko 
You can use esp_http_client to download the data which is to be written in a partition. You can use esp_partition APIs to write the data in a non-app partition. Please refer to https://github.com/espressif/esp-idf/tree/master/examples/storage/partition_api/partition_ops example.

hmalpani, 2023-11-06:
Hello @law-ko 
I hope the above suggestion helped you. I am closing this issue. Please feel free to reopen in case of any queries.
Thanks!

closed 2023-11-06 by hmalpani

law-ko, 2023-11-14:
Hi @hmalpani ,

The `esp_partition_write` function requires a malloc pointer to store the downloaded OTA BIN file first instead of directly writing into the partition. Since the OTA BIN file can be as large as 1MB and mallocing 1MB does not seem feasible, can we directly write into the paritition via `esp_partition_write`? Or should we write it chunk by chunk?