espressif/esp-idf #12394: input/output GPIO pins swapped in example mcpwm_capture_hc_sr04.c

morgana2313, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-393-g7698a413bc

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Wemos Lolin S3

### Power Supply used.

USB

### What is the expected behavior?

The example `examples/peripherals/mcpwm/mcpwm_capture_hc_sr04` should work. 

### What is the actual behavior?

The example `examples/peripherals/mcpwm/mcpwm_capture_hc_sr04` doesn't work, because the GPIO pins in the source are swapped in `examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c`:

* The (input) Tigger capture channel (`#define HC_SR04_TRIG_GPIO  0`) is configured as an output: `GPIO[0]| InputEn: 0| OutputEn: 1`.
* The (output) echo channel (`#define HC_SR04_ECHO_GPIO  2`) is configured as an input: `GPIO[2]| InputEn: 1| OutputEn: 0`.

```
I (321) example: Install capture timer
I (331) example: Install capture channel
I (331) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (341) example: Register capture callback
I (351) example: Enable capture channel
I (351) example: Configure Trig pin
I (361) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (361) example: Enable and start capture timer
```

This diff fixes this mistake and then the example works.
 
```diff
diff --git a/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c b/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
index 10e8a55eef..830ce8f656 100644
--- a/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
+++ b/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
@@ -63,10 +63,10 @@ void app_main(void)
     };
     ESP_ERROR_CHECK(mcpwm_new_capture_timer(&cap_conf, &cap_timer));

-    ESP_LOGI(TAG, "Install capture channel");
+    ESP_LOGI(TAG, "Install capture input trigger channel");
     mcpwm_cap_channel_handle_t cap_chan = NULL;
     mcpwm_capture_channel_config_t cap_ch_conf = {
-        .gpio_num = HC_SR04_ECHO_GPIO,
+        .gpio_num = HC_SR04_TRIG_GPIO,
         .prescale = 1,
         // capture on both edge
         .flags.neg_edge = true,
@@ -86,14 +86,14 @@ void app_main(void)
     ESP_LOGI(TAG, "Enable capture channel");
     ESP_ERROR_CHECK(mcpwm_capture_channel_enable(cap_chan));

-    ESP_LOGI(TAG, "Configure Trig pin");
+    ESP_LOGI(TAG, "Configure Echo output pin");
     gpio_config_t io_conf = {
         .mode = GPIO_MODE_OUTPUT,
-        .pin_bit_mask = 1ULL << HC_SR04_TRIG_GPIO,
+        .pin_bit_mask = 1ULL << HC_SR04_ECHO_GPIO,
     };
     ESP_ERROR_CHECK(gpio_config(&io_conf));
     // drive low by default
-    ESP_ERROR_CHECK(gpio_set_level(HC_SR04_TRIG_GPIO, 0));
+    ESP_ERROR_CHECK(gpio_set_level(HC_SR04_ECHO_GPIO, 0));

     ESP_LOGI(TAG, "Enable and start capture timer");
     ESP_ERROR_CHECK(mcpwm_capture_timer_enable(cap_timer));
```

### Steps to reproduce.

1. Compile and run the `peripherals/mcpwm/mcpwm_capture_hc_sr04` example.

### Debug Logs.

```plain
diff --git a/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c b/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
index 10e8a55eef..bd21a982e7 100644
--- a/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
+++ b/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04/main/mcpwm_capture_hc_sr04.c
@@ -63,10 +63,10 @@ void app_main(void)
     };
     ESP_ERROR_CHECK(mcpwm_new_capture_timer(&cap_conf, &cap_timer));

-    ESP_LOGI(TAG, "Install capture channel");
+    ESP_LOGI(TAG, "Install capture trigger channel");
     mcpwm_cap_channel_handle_t cap_chan = NULL;
     mcpwm_capture_channel_config_t cap_ch_conf = {
-        .gpio_num = HC_SR04_ECHO_GPIO,
+        .gpio_num = HC_SR04_TRIG_GPIO,
         .prescale = 1,
         // capture on both edge
         .flags.neg_edge = true,
@@ -86,14 +86,14 @@ void app_main(void)
     ESP_LOGI(TAG, "Enable capture channel");
     ESP_ERROR_CHECK(mcpwm_capture_channel_enable(cap_chan));

-    ESP_LOGI(TAG, "Configure Trig pin");
+    ESP_LOGI(TAG, "Configure Echo pin");
     gpio_config_t io_conf = {
         .mode = GPIO_MODE_OUTPUT,
-        .pin_bit_mask = 1ULL << HC_SR04_TRIG_GPIO,
+        .pin_bit_mask = 1ULL << HC_SR04_ECHO_GPIO,
     };
     ESP_ERROR_CHECK(gpio_config(&io_conf));
     // drive low by default
-    ESP_ERROR_CHECK(gpio_set_level(HC_SR04_TRIG_GPIO, 0));
+    ESP_ERROR_CHECK(gpio_set_level(HC_SR04_ECHO_GPIO, 0));

     ESP_LOGI(TAG, "Enable and start capture timer");
     ESP_ERROR_CHECK(mcpwm_capture_timer_enable(cap_timer));
```
```

suda-morris, 2023-10-13:
@morgana2313 The "Trig" means the esp32 sends out the trigger signal, so esp32 configures it as "output". The "echo" means the **sensor** responses a result, and esp32 receives it, so esp32 configures it as "input".

morgana2313, 2023-10-13:
> @morgana2313 The "Trig" means the esp32 sends out the trigger signal, so esp32 configures it as "output". The "echo" means the **sensor** responses a result, and esp32 receives it, so esp32 configures it as "input".
> 
> https://cdn.sparkfun.com/datasheets/Sensors/Proximity/HCSR04.pdf
> 
> What you need is to swap your hardware connection.

I think you're right. Thank you for setting me straight!

closed 2023-10-13 by morgana2313