espressif/esp-idf #12396: adc continuous read feature that interrupts when threshold is hit

saillingaway, 2023-10-12:
### Is your feature request related to a problem?

I want to be able to continuously track the voltage level on the adc channel. I know this can be done using the ADC continuous mode driver, but in order to have it work like an interrupt I have to continuously read the voltage on the channel, which isn't ideal for my project.

### Describe the solution you'd like.

I'd like an ADC threshold mode that triggers an interrupt when the specified threshold is hit.

saillingaway, 2023-10-12:
Okay so after looking around more on the documentation I see there's an adc [monitor](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/adc_continuous.html?highlight=adc#monitor) functionality on the **master (latest)** which is what I want. Any estimate on when this feature will be available on a stable release?

wanckl, 2023-10-13:
@saillingaway Yeah there is an ADC hardware monitor feature who will trigger interrupts once continue convert value outside the threshold you set, If no problem, this feature will in stable version which will released end this year

saillingaway, 2023-10-13:
@wanckl I see, that's a little late for us unfortunately. Is there any possibility of getting access to a beta build so we could use the feature sooner? We would be using it extensively and would report any bugs we find.

wanckl, 2023-10-16:
@saillingaway     
emmm,, It is available on Master, and is for the **New version of ADC** (not legacy) driver, If you can't working on Master, which release version you use? can try if it is possible to pick the [Commit](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/commit/87848110c135ed0a83a32e9ec636fa06ac95234c) to your work tree.

saillingaway, 2023-10-18:
that gitlab link seems like it's internal, so I can't actually see the commit. But from the hash on the link you gave, it looks like this is the commit on GitHub, is this right? commit https://github.com/espressif/esp-idf/commit/87848110c135ed0a83a32e9ec636fa06ac95234c

wanckl, 2023-10-19:
@saillingaway  Oh :rofl: yeah, sorry, commit id should be same, your commit is right.      
So can you work on master or pick this commit to your branch, otherwise, It will released end this year.       
If need help to pick this commit, feel free to poke me.    
Best wishes!

saillingaway, 2023-10-19:
Thanks! We'll let you know if we notice anything out of the ordinary : )

closed 2025-01-14 by Alvin1Zhang