espressif/esp-idf #12397: Volume command is doing nothing in esp_hid_device on esp-idf V4.4

josh-gc, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-249-gcc71308e2f (also tried on v4.4.6-35-gcdd72ac1d4)

### Espressif SoC revision.

ESP32-PICO-V3-02 (revision v3.0)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom PCB

### Power Supply used.

External 3.3V

### What is the expected behavior?

When the code sends the volume up and down commands this would be reflected on the connected device.

### What is the actual behavior?

Nothing... The volume was not being adjusted by the esp32.

### Steps to reproduce.

1. Flash the ESP_HID_DEVICE example from the IDF to an ESP32 board
2. Connect to iPhone*
3. Code should connect fine 
4. Console will show "HID_DEV_DEMO: Send the volume"
5. No volume adjustment happens on the phone

* If using a Pixel7 it will connect using BT Classic which has a different demo code (which worked). However when I modified the code to send the volume commands it did not work. Also connected to a MacBook Pro and that also did not change the volume

josh-gc, 2023-10-12:
Update:
Sorry if this is not the best way to document but I made some major changes to the code so the volume up/down was from user input and not the timed xTask. This now works on the iPhone13 (on ble) but still nothing else.

Also, I am getting an error on the `esp_hidd_dev_input_set` command when connected as a BT Classic device (Pixel 7).
### Code:
```
if(hid_connection_status == HID_IS_BLE)
{
	esp_hidd_dev_input_set(s_ble_hid_param.hid_dev, 1, HID_RPT_ID_CC_IN, buffer, HID_CC_IN_RPT_LEN);
}
else if(hid_connection_status == HID_IS_BT)
{
	esp_hidd_dev_input_set(s_bt_hid_param.hid_dev, 1, HID_RPT_ID_CC_IN, buffer, HID_CC_IN_RPT_LEN);
}
else
{
	ESP_LOGE(TAG, "HID not connected!");
}
```
### Error:
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (566) app_main: app_main
I (606) esp_hid_gap_init: esp_hid_gap_init
I (606) init_low_level: init_low_level
I (606) BTDM_INIT: BT controller compile version [946b762]
I (606) system_api: Base MAC address is not set
I (616) system_api: read default base MAC address from EFUSE
I (616) BTDM_INIT: Bluetooth MAC: 4c:75:25:f1:94:4a
I (626) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1346) init_bt_gap: init_bt_gap
I (1356) init_ble_gap: init_ble_gap
I (1356) esp_hid_ble_gap_adv_init: esp_hid_ble_gap_adv_init
W (1366) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (1386) ble_hidd_event_callback: Setting to connectable, discoverable
I (2396) bt_hidd_event_callback: Setting to connectable, discoverable
I (2396) esp_hid_device_main: run_example
##############################
ESP HID DEVICE DEMO COMMANDS:
You can input these values
s -- Start pairing
e -- End pairing/disconnect
u -- Send volume up
d -- Send volume down
##############################

I (68116) esp_hid_device_main: send_volume_up
E (68116) esp_hid_device_main: HID not connected!
I (72296) esp_hid_device_main: send_volume_down
E (72296) esp_hid_device_main: HID not connected!
W (524716) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (524766) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
W (524766) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
W (526536) BT_APPL: new conn_srvc id:20, app_id:1
W (526546) BT_APPL: bta_dm_pm_ssr conn_srvc id:20, app_id:1
W (526546) BT_APPL: bta_dm_pm_ssr:3, lat:360
I (526546) BT_HIDD: Connected to c8:2a:dd:82:0f:0d
I (526556) bt_hidd_event_callback: CONNECTED
W (531746) BT_HCI: hci cmd send: sniff: hdl 0x81, intv(10 18)
W (531756) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 18, status 0x0
I (531756) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
W (531816) BT_HCI: hcif ssr evt: st 0x0, hdl 0x81, tx_lat 360 rx_lat 720
I (580116) esp_hid_device_main: send_volume_up
E (580116) BT_HIDD: index out of range[0-0]
E (580116) BT_HIDD: HID device not connected!
E (580216) BT_HIDD: index out of range[0-0]
E (580216) BT_HIDD: HID device not connected!
```

I made a repo for the full code here: https://github.com/josh-gc/esp_selfie
Any help would be greatly appreciated!

josh-gc, 2023-10-12:
Update again:
The iPhone works on the first connection after reboot but not if the device is disconnected and then reconnected...

zhp0406, 2023-10-17:
@josh-gc 
Hello, please apply the following patch and test with iPhone 13 again.
(esp-idf/examples/bluetooth/esp_hid_device)

[0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char.patch](https://github.com/espressif/esp-idf/files/12924136/0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char.patch)

If you continue to encounter issues, please let me know.

thanks.

zhp0406, 2023-10-17:
@josh-gc 
I tested it using an iPhone 13（IOS 17.0.3）, and it appears to be working correctly. Therefore, I need your help in reproducing the issue. Here is a summary of the my testing results:

After connecting to the esp32 (esp_hid_device), the volume on the iPhone 13 starts updating. When I disconnect the connection and open Bluetooth again, it automatically reconnects, and the volume continues to update.

https://github.com/espressif/esp-idf/assets/131946088/874c8b1d-97d9-48c4-9555-8ef555d46b82

josh-gc, 2023-10-17:
@zhp0406 Thank you for responding. Sorry my bug was not clear, here is the process to reproduce:
1. "Forget" the connection and reset the ESP32
2. Connect to phone -> Volume works
3. "Forget" the device
4. Reconnect the device -> Volume does not work
5. Reset the ESP32
6. Reconnect the device -> Volume works again

I had the same experience with my code ([esp_selfie](https://github.com/josh-gc/esp_selfie)) and the esp_hid_device example. The .patch file did not make a difference. I made another thread in the espressif forum to track the issue with bt/Android commands in my code.

josh-gc, 2023-10-18:
Other thread here: https://www.esp32.com/viewtopic.php?f=13&t=36330

zhp0406, 2023-10-18:
@josh-gc 
I was able to reproduce the issue without applying any patches. Here is the environment I used:

iPhone 13
esp-idf/examples/bluetooth/esp_hid_device
IDF commit: cc71308e2fdce1d6d27fc52d39943f5d0fb83f35
I observed a phenomenon: when the iOS pairing window pops up, if I quickly click on pairing, the subsequent operations work fine. However, if I don't click quickly (e.g., waiting for 5 seconds before clicking on pairing), an issue occurs.

Is this observation similar to what you've experienced?

josh-gc, 2023-10-18:
@zhp0406 I don't think that the timing is the issue. Here is a screen capture from the iPhone where I:
1. Open BLE
2. Connect to device
3. Volume commands **fails**
4. Reset the ESP32
5. Reconnect
6. Volume commands **work**
7. Turn off / on bluetooth
8. Reconnects
9. Volume commands **work**
10. Forget device
11. Reconnect device
12. Volume commands **fails**
13. Reset device
14. Reconnect
15. Volume commands **work**

Also, when I tried to do the screen capture over a cable from my computer I could not get it to work at all.

Also, after about a minute I started getting these error messages:
E (56718) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (56718) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (58328) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (58328) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (63398) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (63398) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (74978) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (74978) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (120128) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (120128) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (209268) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (209268) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (218378) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (218378) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT

(This is on my code, not the esp_hid_device but I saw the same issues on both)

zhp0406, 2023-10-19:
@josh-gc 
Based on the current analysis, this should be a compatibility issue. If encryption is not successful, Apple devices will not recognize the ESP32 as an HID device. To verify my theory, please kindly test with the following patch. Thank you.
[0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs.patch](https://github.com/espressif/esp-idf/files/13042787/0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs.patch)

zhp0406, 2023-10-19:
E (63398) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
The above log indicates a failed pairing. Please make sure you have successfully paired it.If pairing is not successful,the HID example cannot be used.

josh-gc, 2023-10-19:
@zhp0406
` add_db_record(_last_db, index++, (uint8_t *)&s_character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE_ENCRYPTED, 2, 0, NULL);`
The above change you made looks to have solved the re-connection after "forget"!

I am still getting the 'E (94802) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT' error though. It only happens when the iPhone goes to sleep so I had to modify the code to pause long enough to let it sleep. Here is my modified `ble_hid_demo_task`:

```
void ble_hid_demo_task(void *pvParameters)
{
	static bool send_volum_up = false;
	static int	count = 0;
	while (1) {
		ESP_LOGI(TAG, "Send the volume");
		if (send_volum_up) {
			esp_hidd_send_consumer_value(HID_CONSUMER_VOLUME_UP, true);
			vTaskDelay(100 / portTICK_PERIOD_MS);
			esp_hidd_send_consumer_value(HID_CONSUMER_VOLUME_UP, false);
		} else {
			esp_hidd_send_consumer_value(HID_CONSUMER_VOLUME_DOWN, true);
			vTaskDelay(100 / portTICK_PERIOD_MS);
			esp_hidd_send_consumer_value(HID_CONSUMER_VOLUME_DOWN, false);
		}
		send_volum_up = !send_volum_up;
		vTaskDelay(2000 / portTICK_PERIOD_MS);

		count ++;	// Increment count

		// Delay for 10 seconds to let the phone sleep
		if(count > 4)
		{
			printf("pause \n");
			vTaskDelay(60000 / portTICK_PERIOD_MS);
			count = 0;
		}
	}
}
```
and here is the error logs:

```
I (2402) HID_DEV_BT: Setting to connectable, discoverable
I (5522) HID_DEV_BLE: CONNECT
E (6072) BT_BTM: btm_ble_remove_resolving_list_entry_complete remove resolving list error 0x2
E (6332) BT_SMP: Value for numeric comparison = 995800
I (6332) ESP_HID_GAP: BLE GAP NC_REQ passkey:995800
E (7592) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
W (10292) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (10412) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_LENC
I (10412) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_PENC
I (10412) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_LID
I (10542) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_PID
I (10692) ESP_HID_GAP: BLE GAP AUTH SUCCESS
I (10692) HID_DEV_DEMO: Send the volume
I (12792) HID_DEV_DEMO: Send the volume
I (14892) HID_DEV_DEMO: Send the volume
I (16992) HID_DEV_DEMO: Send the volume
I (19092) HID_DEV_DEMO: Send the volume
pause 
E (59612) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (59612) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (61392) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (61402) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (69632) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (69632) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
I (81192) HID_DEV_DEMO: Send the volume
E (81592) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
E (81592) BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT
I (83292) HID_DEV_DEMO: Send the volume
I (85392) HID_DEV_DEMO: Send the volume
```
When I wake up the phone it works fine but the volume command does not wake it up on it's own.

zhp0406, 2023-10-20:
@josh-gc 
I've fixed the issue in the latest patch. Please test it, and thank you.
If there are any problems, please report it to me.

[0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs_2.patch](https://github.com/espressif/esp-idf/files/13054251/0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs_2.patch)

zhp0406, 2023-10-23:
@josh-gc 
The first occurrence of "BT_GATT: gatts_write_attr_perm_check - GATT_WRITE_NOT_PERMIT" is normal because the connection is not yet encrypted at this point. However, there were several subsequent occurrences, which indicate a bug. I have fixed it in the patch provided below.You can directly apply this patch.
[0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs_2.patch](https://github.com/espressif/esp-idf/files/13067611/0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs_2.patch)

Furthermore, I have also addressed another compatibility issue with newer Android devices. If your product is intended for use on higher versions of Android devices, it is recommended to apply this patch as well.

[0002-fix-bt-bluedroid-Optimize-compatibility-with-Android.patch](https://github.com/espressif/esp-idf/files/13067618/0002-fix-bt-bluedroid-Optimize-compatibility-with-Android.patch)

If you encounter any other issues during usage, I will promptly resolve them for you.

zhp0406, 2023-10-23:
@josh-gc 
Please test it with the patch, and I look forward to your response. Thank you!

josh-gc, 2023-10-24:
@zhp0406 I was having issues recreating the GATT_WRITE_NOT_PERMIT issue so I reset everything and went through all the steps. Here are my results:

Update the IDF/submodules to: v4.4.6-98-g5f257494c5

1. Tried to apply patch 0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char but this was already there.
2. Confirmed the volume commands do not work on iPhone after "Forget Device"
3. Applied **0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs**
4. Confirmed that this patch solved the above (#2) issue
5. I added the delay to test if the phone sleeping caused an issue and was not able to recreate this bug. Sorry for the false report. I skipped **0001-fix-bt-bluedroid-Fixed-BLE-hid-example-bugs_2** since this was no longer needed.
6. I updated the config to be BLE only:
  a. Component config → Bluetooth → Controller Options → Bluetooth controller mode (BR/EDR/BLE/DUALMODE) (BLE Only)
  b. Component config → Bluetooth → Bluedroid Options → Classic Bluetooth(Disabled)
7. Tested on Android (Pixel 7) and the volume buttons did not work
8. Applied patch 0002-fix-bt-bluedroid-Optimize-compatibility-with-Android
9. This now works great on my phone!

All of my issues are solved. Thank you for your help!
Will this task stay open until these two patches are in the main branch or is there another way I can keep up with this?

zhp0406, 2023-10-25:
@josh-gc 
All are okay. I suggest closing the current issue for now. If you encounter other issues, then create a new issue.
At the same time, I will merge the patch I gave you into the master branch(and release branch) after testing it.
Thank you very much for reporting the issue.

zhp0406, 2023-10-25:
@josh-gc 
I will report back to you after merging into the master branch. However, this might take some time as there are many tests to run. In the meantime, please go ahead and use the patch I gave you.

josh-gc, 2023-10-25:
Closing issue (still waiting on merge to main branch)

closed 2023-10-25 by josh-gc

zhp0406, 2023-11-09:
@josh-gc 
We have fixed the issue with the BLE HID device. Please use the latest branch.

Alternatively, you can also use the following commits:

Commit c73d7c0b600517221e11b5fe54fe9e3c160a0963 improves the compatibility of the HID device on the Windows platform.
Commit 5727802497b0504167216932af6d08a392845c6e enhances compatibility with iOS and macOS.
Commit b90dc6a515f69d9b9a2a9401f873186167f5562f adds support for the HID device control point feature.
Commit f7e174e9487cc218b92cffee9dc5953e962e0ff4 improves compatibility with Android versions after 10.