espressif/esp-idf #12398: tusb_msc example missing tusb.h

wmarchewka, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Using platformio, esp-idf NOT arduino .

I am trying to run the [TinyUSB Mass Storage Device Example] "tusb_msc"  example here:
[https://github.com/espressif/esp-idf/tree/8fc8f3f4799/examples/peripherals/usb/device/tusb_msc](url)

Im using platfromio and have copied the TinyUSB library from here into my lib folder:
[https://github.com/espressif/idf-extra-components/tree/master/usb/esp_tinyusb](url)

However, tinyusb.h is missing "tusb.h":
[https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/include/tinyusb.h](url)

### What is the actual behavior?

lib/espressif_esp_tinyusb_1.4.2/include/tinyusb.h:11:10: fatal error: tusb.h: No such file or directory

### Steps to reproduce.

I am trying to run the [TinyUSB Mass Storage Device Example] "tusb_msc"  example here:
[https://github.com/espressif/esp-idf/tree/8fc8f3f4799/examples/peripherals/usb/device/tusb_msc](url)

Im using platfromio and have copied the TinyUSB library from here into my lib folder:
[https://github.com/espressif/idf-extra-components/tree/master/usb/esp_tinyusb](url)

However, tinyusb.h is missing "tusb.h":
[https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/include/tinyusb.h](url)

Where is "tusb.h" sourced from?

### Build or installation Logs.

```plain
Executing task: platformio run 

Processing lilygo-t-display-s3 (platform: espressif32; board: lilygo-t-display-s3; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lilygo-t-display-s3.html
PLATFORM: Espressif 32 (6.4.0) > LilyGo T-Display-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- espressif_esp_tinyusb_1.4.2
Building in release mode
Compiling .pio/build/lilygo-t-display-s3/src/main.o
Generating LD script .pio/build/lilygo-t-display-s3/memory.ld
Compiling .pio/build/lilygo-t-display-s3/app_trace/app_trace.o
Compiling .pio/build/lilygo-t-display-s3/app_trace/app_trace_util.o
In file included from src/main.c:19:
lib/espressif_esp_tinyusb_1.4.2/include/tinyusb.h:11:10: fatal error: tusb.h: No such file or directory
```

nopnop2002, 2023-10-13:
PlatformIO uses PlatformIO's own library, not the ESP-IDF library.

This location is more appropriate.
https://github.com/platformio/platform-espressif32/issues

wmarchewka, 2023-10-13:
Thank you. I will investigate there.

closed 2023-10-13 by wmarchewka