espressif/esp-idf #12399: esp_http_server: httpd_resp_send_err - not all HTTP status codes supported

valkuc, 2023-10-12:
### Is your feature request related to a problem?

Sometimes it is necessary to throw specific HTTP status code with TCP_NODELAY socket option turned on. Unfortunately, `httpd_resp_send_err` does not support all HTTP status codes as they are defined as internal enum.

### Describe the solution you'd like.

I see 2 options:
1) Add another method which will be not restricted by `httpd_err_code_t` enum and will accept `httpd_req_t *req, const char *status` as parameters - this will give possibility for user to send any http status.
2) Extend existing `httpd_err_code_t` enum with all existing HTTP status codes.

### Describe alternatives you've considered.

Tried to create my own `httpd_resp_send_err` method but because internal structures (httpd_req_aux) are hidden from the end user (located in private header esp_httpd_priv.h) this becomes not possible.

hmalpani, 2023-10-30:
Hello @valkuc 

You can use [httpd_resp_set_status()](https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1213) API for this. Please refer to [file_serving example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/file_serving/main/file_server.c#L54)

Please let me know if this serves your purpose.

Thanks!

valkuc, 2023-10-30:
> Hello @valkuc
>
> You can use [httpd_resp_set_status()](https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1213) API for this. Please refer to [...]

Yes, but no :-) The `httpd_resp_send_err` method has a bit different behavior - it can disable Nagle's algorithm and send data to socket without delay. The `httpd_resp_set_status` is mostly suitable for 2xx and 3xx status codes.

hmalpani, 2023-12-05:
Hello @valkuc 

I have created an internal MR for adding the support. I have introduced a new API for serving this purpose. I am sharing a patch here. 
[IDFGH_11235.txt](https://github.com/espressif/esp-idf/files/13556813/IDFGH_11235.txt)

closed 2023-12-23 by espressif-bot