espressif/esp-idf #12400: esp_hid_host's BT classic part does not work

ImproperCatGirl, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.4

### Espressif SoC revision.

ESP32-S3 (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Third party custom board

### Power Supply used.

USB

### What is the expected behavior?

I expected the esp_hid_host demo to  successfully find and pair with a BT classic HID device such as Logitech K380 used for testing.

### What is the actual behavior?

The esp_hid_host demo fails to find the BT classic devices in pairing mode.

### Steps to reproduce.

1.
Build, flash and monitor esp_hid_host example
2.
Put the BT Classic HID device into pairing mode (in my case it was a Logitech K380)
3.
Reset the ESP32-S3
4.
The demo fails to find anything

### Debug Logs.

```plain
I (24) boot: ESP-IDF v5.0.4 2nd stage bootloader
I (24) boot: compile time 07:17:04
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=236b4h (145076) map
I (122) esp_image: segment 1: paddr=000336dc vaddr=3fc97000 size=03638h ( 13880) load
I (125) esp_image: segment 2: paddr=00036d1c vaddr=40374000 size=092fch ( 37628) load
I (136) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=7fd40h (523584) map
I (231) esp_image: segment 4: paddr=000bfd68 vaddr=4037d2fc size=09c28h ( 39976) load
I (247) boot: Loaded app from partition at offset 0x10000
I (248) boot: Disabling RNG early entropy source...
I (259) cpu_start: Multicore app
I (259) cpu_start: Pro cpu up.
I (259) cpu_start: Starting app cpu, entry point is 0x4037535c
0x4037535c: call_start_cpu1 at /home/meow/esp/esp-idf_v5/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (278) cpu_start: Pro cpu start user code
I (278) cpu_start: cpu freq: 160000000 Hz
I (278) cpu_start: Application information:
I (281) cpu_start: Project name:     esp_hid_host
I (286) cpu_start: App version:      1
I (291) cpu_start: Compile time:     Oct 13 2023 07:16:59
I (297) cpu_start: ELF file SHA256:  d14f1a81ac64e854...
I (303) cpu_start: ESP-IDF:          v5.0.4
I (307) cpu_start: Min chip rev:     v0.0
I (312) cpu_start: Max chip rev:     v0.99 
I (317) cpu_start: Chip rev:         v0.2
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (329) heap_init: At 3FC9E5E0 len 0004B130 (300 KiB): DRAM
I (335) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (342) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (348) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (355) spi_flash: detected chip: gd
I (359) spi_flash: flash io: dio
W (363) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (376) coexist: coexist rom version e7ae62f
I (381) app_start: Starting scheduler on CPU0
I (386) app_start: Starting scheduler on CPU1
I (386) main_task: Started on CPU0
I (396) main_task: Calling app_main()
I (406) ESP_HIDH_DEMO: setting hid gap, mode:1
I (406) BLE_INIT: BT controller compile version [963cad4]
I (416) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (456) system_api: Base MAC address is not set
I (456) system_api: read default base MAC address from EFUSE
I (456) BLE_INIT: Bluetooth MAC: 

I (496) ESP_HIDH_DEMO: SCAN...
I (496) main_task: Returned from app_main()
BLE: aa:bb:cc:dd:ee:ff, RSSI: -91, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: aa:bb:cc:dd:ee:ff, RSSI: -93, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: aa:bb:cc:dd:ee:ff, RSSI: -62, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: aa:bb:cc:dd:ee:ff, RSSI: -88, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'PUBLIC'
BLE: aa:bb:cc:dd:ee:ff, RSSI: -90, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'PUBLIC', NAME: 'U-AC7BC6'
I (5496) ESP_HIDH_DEMO: SCAN: 0 results
```

boblane1, 2023-10-18:
Hi @ImproperCatGirl, 

Thanks for your reporting. ESP32-S3 does not support Classic Bluetooth, you can think about using ESP32.

closed 2024-02-27 by Alvin1Zhang

Slion, 2024-06-09:
I could not find any Classic Bluetooth devices either using an Arduino Nano ESP32 which is also based on ESP32-S3 and that would explain why. That's really too bad cause I bought those thinking they did support Classic Bluetooth (BR/EDR). 🤦‍♂️