espressif/esp-idf #12401: esp_hid_host example fails to respond with HID reports from an actual BLE HID keyboard

ImproperCatGirl, 2023-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.4

### Espressif SoC revision.

ESP32-S3 (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Third party custom board

### Power Supply used.

USB

### What is the expected behavior?

After scanning and pairing, the example prints the HID report data to serial console when a key is pressed.

### What is the actual behavior?

Nothing happens after a key is pressed on the keyboard.

### Steps to reproduce.

1.Build, flash and monitor esp_hid_host example
2.Pair with an actual keyboard, not the esp_hid_device example. (The keyboard used is a Logitech K390 here)
3.Press any key on the keyboard.
4.After the first VENDOR type report, crickets.

### Debug Logs.

```plain
I (24) boot: ESP-IDF v5.0.4 2nd stage bootloader
I (24) boot: compile time 07:17:04
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=236b4h (145076) map
I (122) esp_image: segment 1: paddr=000336dc vaddr=3fc97000 size=03638h ( 13880) load
I (125) esp_image: segment 2: paddr=00036d1c vaddr=40374000 size=092fch ( 37628) load
I (136) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=7fd40h (523584) map
I (230) esp_image: segment 4: paddr=000bfd68 vaddr=4037d2fc size=09c28h ( 39976) load
I (247) boot: Loaded app from partition at offset 0x10000
I (248) boot: Disabling RNG early entropy source...
I (259) cpu_start: Multicore app
I (259) cpu_start: Pro cpu up.
I (259) cpu_start: Starting app cpu, entry point is 0x4037535c
0x4037535c: call_start_cpu1 at /home/meow/esp/esp-idf_v5/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (278) cpu_start: Pro cpu start user code
I (278) cpu_start: cpu freq: 160000000 Hz
I (278) cpu_start: Application information:
I (281) cpu_start: Project name:     esp_hid_host
I (286) cpu_start: App version:      1
I (290) cpu_start: Compile time:     Oct 13 2023 07:16:59
I (296) cpu_start: ELF file SHA256:  d14f1a81ac64e854...
I (302) cpu_start: ESP-IDF:          v5.0.4
I (307) cpu_start: Min chip rev:     v0.0
I (312) cpu_start: Max chip rev:     v0.99 
I (317) cpu_start: Chip rev:         v0.2
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (329) heap_init: At 3FC9E5E0 len 0004B130 (300 KiB): DRAM
I (335) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (342) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (348) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (355) spi_flash: detected chip: gd
I (359) spi_flash: flash io: dio
W (362) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (376) coexist: coexist rom version e7ae62f
I (381) app_start: Starting scheduler on CPU0
I (386) app_start: Starting scheduler on CPU1
I (386) main_task: Started on CPU0
I (396) main_task: Calling app_main()
I (406) ESP_HIDH_DEMO: setting hid gap, mode:1
I (406) BLE_INIT: BT controller compile version [963cad4]
I (416) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (466) system_api: Base MAC address is not set
I (466) system_api: read default base MAC address from EFUSE
I (466) BLE_INIT: Bluetooth MAC: [redacted]

I (506) ESP_HIDH_DEMO: SCAN...
I (506) main_task: Returned from app_main()
[unrelated scan results redacted]
I (5506) ESP_HIDH_DEMO: SCAN: 1 results
  BLE: aa:bb:cc:dd:ee:ff, RSSI: -19, USAGE: GENERIC, APPEARANCE: 0x03c1, ADDR_TYPE: 'RANDOM', NAME: Pebble K390 
I (8336) ESP_HID_GAP: BLE GAP PASSKEY_NOTIF passkey:930885
W (17186) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (17396) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_PID
I (17396) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_LENC
I (17396) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_PENC
I (17396) ESP_HID_GAP: BLE GAP KEY type = ESP_LE_KEY_LID
I (17596) ESP_HID_GAP: BLE GAP AUTH SUCCESS
I (18836) ESP_HIDH_DEMO: aa:bb:cc:dd:ee:ff OPEN: Pebble K390
BDA:aa:bb:cc:dd:ee:ff, Appearance: 0x03c0, Connection ID: 0
Name: Pebble K390, Manufacturer: Logitech, Serial Number:
E (18846) BT_APPL: Max Notification Reached, registration failed.
PID: 0xb377, VID: 0x046d, VERSION: 0x0011
Battery: Handle: 29, CCC Handle: 30
Report Maps: 1
  Report Map Length: 286
E (18866) BT_APPL: Max Notification Reached, registration failed.
      VENDOR  OUTPUT REPORT, ID: 17, Length:  19, Permissions: 0x0e, Handle:  68, CCC Handle:   0
      VENDOR   INPUT REPORT, ID: 17, Length:  19, Permissions: 0x12, Handle:  64, CCC Handle:  65
    CCONTROL   INPUT REPORT, ID:  5, Length:   4, Permissions: 0x12, Handle:  60, CCC Handle:  61
     GENERIC   INPUT REPORT, ID:  4, Length:   1, Permissions: 0x12, Handle:  56, CCC Handle:  57
      VENDOR   INPUT REPORT, ID:  3, Length:   4, Permissions: 0x12, Handle:  52, CCC Handle:  53
      VENDOR FEATURE REPORT, ID:  9, Length:  48, Permissions: 0x0a, Handle:  49, CCC Handle:   0
    KEYBOARD  OUTPUT REPORT, ID:  1, Length:   1, Permissions: 0x0e, Handle:  46, CCC Handle:   0
    KEYBOARD   INPUT REPORT, ID:  1, Length:   7, Permissions: 0x12, Handle:  42, CCC Handle:  43
    KEYBOARD  OUTPUT   BOOT, ID:  1, Length:   1, Permissions: 0x0e, Handle:  38, CCC Handle:   0
    KEYBOARD   INPUT   BOOT, ID:  1, Length:   8, Permissions: 0x12, Handle:  35, CCC Handle:  36
I (18986) ESP_HIDH_DEMO: aa:bb:cc:dd:ee:ff INPUT:   VENDOR, MAP:  0, ID:  17, Len: 19, Data:
I (18986) ESP_HIDH_DEMO: ff 04 00 01 01 01 00 00 00 00 00 00 00 00 00 00 
I (18996) ESP_HIDH_DEMO: 00 00 00
```

zhp0406, 2023-10-18:
@ImproperCatGirl 
Could you please use the following patch and check if the issue still exists?
If the problem persists, please report it to me.
[0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char.patch](https://github.com/espressif/esp-idf/files/12981647/0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char.patch)

ImproperCatGirl, 2023-10-23:
> @ImproperCatGirl Could you please use the following patch and check if the issue still exists? If the problem persists, please report it to me. [...]

Is this the patch for esp-idf v5.0.4? When I try to apply it, it fails.

git apply --verbose  '~/Downloads/0003-fix-bt-bluedroid-Fix-bugs-in-ble-service-change-char.patch' 
Checking patch components/bt/host/bluedroid/bta/gatt/bta_gatts_act.c...
Checking patch components/bt/host/bluedroid/stack/gatt/gatt_api.c...
Hunk #1 succeeded at 763 (offset -18 lines).
Hunk #2 succeeded at 789 (offset -18 lines).
Checking patch components/bt/host/bluedroid/stack/gatt/gatt_attr.c...
error: while searching for:
#define BLE_GATT_CL_SUPP_FEAT_BITMASK 0x07

#define GATTP_MAX_NUM_INC_SVR       0
#define GATTP_MAX_CHAR_NUM          4
#define GATTP_MAX_ATTR_NUM          (GATTP_MAX_CHAR_NUM * 2 + GATTP_MAX_NUM_INC_SVR + 1)
#define GATTP_MAX_CHAR_VALUE_SIZE   50

error: patch failed: components/bt/host/bluedroid/stack/gatt/gatt_attr.c:40
error: components/bt/host/bluedroid/stack/gatt/gatt_attr.c: patch does not apply
Checking patch components/bt/host/bluedroid/stack/gatt/gatt_db.c...
error: while searching for:
** Returns          Status of the operation.
**
*******************************************************************************/
static tGATT_STATUS gatts_get_attr_value_internal(UINT16 attr_handle, UINT16 *length, UINT8 **value)
{
    UINT8 i;
    tGATT_READ_REQ read_req;

error: patch failed: components/bt/host/bluedroid/stack/gatt/gatt_db.c:780
error: components/bt/host/bluedroid/stack/gatt/gatt_db.c: patch does not apply
Checking patch components/bt/host/bluedroid/stack/gatt/include/gatt_int.h...
Hunk #1 succeeded at 737 (offset -17 lines).
Checking patch components/bt/host/bluedroid/stack/include/stack/gatt_api.h...
Hunk #1 succeeded at 930 (offset -13 lines).

zhp0406, 2023-10-23:
@ImproperCatGirl 
Hello, I have made a new discovery. 
By default, Bluedroid can register up to 5 notifications, but I think this number is too low, and I suspect it's causing the issue. However, I don't have the Pebble K390 device here, so I need you to give it another try to verify. Please do the following:

In `esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_api.c`, change the line:

```
#define BTA_GATTC_NOTIF_REG_MAX     BTA_GATTC_CONN_MAX
```

to

```
#define BTA_GATTC_NOTIF_REG_MAX     64
```

This change should increase the maximum number of notifications that can be registered to 64.

zhp0406, 2023-10-23:
@ImproperCatGirl 
Please test it, and I look forward to your response. Thank you!

zhp0406, 2023-10-31:
@ImproperCatGirl 
Hi,Have the issue solved？

Indastri, 2023-11-21:
Feel free to reopen this issue if it still persists. Closing for now. Thanks for your contribution!

closed 2023-11-21 by Indastri

K7MDL2, 2025-10-29:
I have the same issue as above with a K380s BLE keyboard.    I have a Rii i8+ (aka K08) that does work.  

I increased the connections to 64 per above but it did not work.  It appears the Vendor event is the first report and that is being acted on as a keyboard input report (I assume) so no keystrokes processed. However, after some time passed, occasionally I get keystrokes that were queued up minutes ago suddenly blurt out.  No idea where they were gathered and what triggered the queue to release.  Debug did not show anything in my test app.

The K08 mini keyboard only shows 4 reports, the first one is the Keyboard.  No VENDOR reports.  I have been trying to track this down for the last couple of days, still looking into it.  The fact that I can eventually get delayed keystrokes offers some hope.

Additionally the BLE keyboard is not recognized after a disconnect, I have to enter a new passkey each connection.  This is with esp-idf 5.5.  A Classic BT keyboard (K380) reconnects fine and keystrokes work proper.

Another possibly related oddity is the same program compiled under Arduino IDE fails connection to a Classic BT keyboard (K380).  It works OK under esp-idf.  The only significant changes are the extern 'C' app_main replaces with loop(); and iniit_Arduino() is called.  I am guessing the esp-idf components library is precompiled and different than what is used in the esp-idf env.

Here is some data related to the K380s BLE keyboard not working.

      BLE: d5:64:cb:77:41:4e, RSSI: -51, USAGE: GENERIC, APPEARANCE: 0x03c1, ADDR_TYPE: 'RANDOM', NAME: Pebble K380s
    Please enter the following pairing code followed with ENTER on your keyboard: 740608
    BDA:d5:64:cb:77:41:4e, Appearance: 0x03c0, Connection ID: 0
    Name: Pebble K380s, Manufacturer: Logitech, Serial Number: 0D503908
    PID: 0xb377, VID: 0x046d, VERSION: 0x0011
    Battery: Handle: 29, CCC Handle: 30
    Report Maps: 1
      Report Map Length: 286
          VENDOR  OUTPUT REPORT, ID: 17, Length:  19, Permissions: 0x0e, Handle:  68, CCC Handle:   0
          VENDOR   INPUT REPORT, ID: 17, Length:  19, Permissions: 0x12, Handle:  64, CCC Handle:  65
        CCONTROL   INPUT REPORT, ID:  5, Length:   4, Permissions: 0x12, Handle:  60, CCC Handle:  61
    E (24727) BT_APPL: Max Notification Reached, registration failed,see CONFIG_BT_GATTC_NOTIF_REG_MAX in menuconfig
         GENERIC   INPUT REPORT, ID:  4, Length:   1, Permissions: 0x12, Handle:  56, CCC Handle:  57
          VENDOR   INPUT REPORT, ID:  3, Length:   4, Permissions: 0x12, Handle:  52, CCC Handle:  53
          VENDOR FEATURE REPORT, ID:  9, Length:  48, Permissions: 0x0a, Handle:  49, CCC Handle:   0
        KEYBOARD  OUTPUT REPORT, ID:  1, Length:   1, Permissions: 0x0e, Handle:  46, CCC Handle:   0
        KEYBOARD   INPUT REPORT, ID:  1, Length:   7, Permissions: 0x12, Handle:  42, CCC Handle:  43
        KEYBOARD  OUTPUT   BOOT, ID:  1, Length:   1, Permissions: 0x0e, Handle:  38, CCC Handle:   0
        KEYBOARD   INPUT   BOOT, ID:  1, Length:   8, Permissions: 0x12, Handle:  35, CCC Handle:  36
    ----> Connected to keyboard <----

Note the error that popped up:
E (24727) BT_APPL: Max Notification Reached, registration failed,see CONFIG_BT_GATTC_NOTIF_REG_MAX in menuconfig.   In sdkconfig it was set to 5.  I changed it to 64.   So it is now changed in both esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_api.c and and my project sdkconfig here.

For reference, here is a good working BLE keyboard, the Rii i8+ mini keyboard.  It does not advertise a UUID == 0x1812, it is 0x0000.  I bypass that check and just use appearance == 0x03C1 and all works OK.

[image]

Only 4 reports, first are Keyboard reports.  A coincidence it works, and Keyboard reports are the first reports?

K7MDL2, 2025-10-29:
I finally got more debug enabled.  I see there are keyboard events without delay but not getting propagated up.  The length is 7 bytes versus the standaed 8 bytes.  Here is a Key down event followed by key up event.  

    D (311761) BT_GATT: gatt_get_ch_state: ch_state=4
    D (311762) BT_GATT: gatt_process_notification
    D (311762) BT_GATT: gatt_get_link_encrypt_status status=0x0
    D (311765) BT_GATT: gatt_cl_op_cmpl_cback
    D (311769) BT_APPL: bta_gattc_cmpl_cback: conn_id = 3 op = 6 status = 0
    I (311776) BT_GATT: GATT_GetConnectionInfor conn_id=3
    D (311782) BT_APPL: bta_gattc_cmpl_cback: conn_id = 4 op = 6 status = 0
    I (311788) BT_GATT: GATT_GetConnectionInfor conn_id=4
    D (311794) BT_APPL: Notification registered!
    D (311798) BT_APPL: bta_gattc_proc_other_indication check p_data->att_value.handle=42 p_data->handle=0
    D (311808) BT_APPL: is_notify 128
    D (311811) event: running post ESP_HIDH_EVENTS:2 with handler 0x400ebb80 and context 0x3ffd69f0 on loop 0x3ffd6e38
    --- 0x400ebb80: esp_hidh_event_handler_wrapper at D:/esp-idf/v5.5/esp-idf/components/esp_hid/src/esp_hidh.c:65
    D (311821) BTKeyboard: d5:64:cb:77:41:50 INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 7, Data:
    D (311830) BTKeyboard: 00 09 00 00 00 00 00
    D (311886) BT_GATT: gatt_get_ch_state: ch_state=4
    D (311887) BT_GATT: gatt_process_notification
    D (311887) BT_GATT: gatt_get_link_encrypt_status status=0x0
    D (311890) BT_GATT: gatt_cl_op_cmpl_cback
    D (311894) BT_APPL: bta_gattc_cmpl_cback: conn_id = 3 op = 6 status = 0
    I (311901) BT_GATT: GATT_GetConnectionInfor conn_id=3
    D (311907) BT_APPL: bta_gattc_cmpl_cback: conn_id = 4 op = 6 status = 0
    I (311913) BT_GATT: GATT_GetConnectionInfor conn_id=4
    D (311919) BT_APPL: Notification registered!
    D (311923) BT_APPL: bta_gattc_proc_other_indication check p_data->att_value.handle=42 p_data->handle=0
    D (311933) BT_APPL: is_notify 128
    D (311936) event: running post ESP_HIDH_EVENTS:2 with handler 0x400ebb80 and context 0x3ffd69f0 on loop 0x3ffd6e38
    --- 0x400ebb80: esp_hidh_event_handler_wrapper at D:/esp-idf/v5.5/esp-idf/components/esp_hid/src/esp_hidh.c:65
    D (311946) BTKeyboard: d5:64:cb:77:41:50 INPUT: KEYBOARD, MAP:  0, ID:   1, Len: 7, Data:
    D (311954) BTKeyboard: 00 00 00 00 00 00 00

I also noted the bonded count is up to 14 and when IO connect it is removing a bonded device each time.  Inbluedroid_user_config.h the limit == 15 (CONFIG_BT_SMP_MAX_BONDS = 15).  I bumped up the number in sdkconfig to 40.  Still 14.  Changed CONFIG_BT_SMP_MAX_BONDS = 40 in the file.

So it seems the keyboard events are there, but not getting passed up to the app.   I am not a BLE stack expert so still trying to walk through the steps to track down the problem.