espressif/esp-idf #12406: Warning about old I2C driver despite having changed to the new one

webcompas, 2023-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-POE-ISO

### Power Supply used.

USB

### What is the expected behavior?

I'm using a little OLED display which is controlled via I2C. With ESP-IDF v5.2 a new I2C driver implementation has been added. Therefore the following warning is shown on the monitor console if the old driver is used:
``
i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
``

So I changed my code to the new driver and hence expected this warning not to be shown anymore.

### What is the actual behavior?

Despite having changed to the new I2C driver the warning is still shown.

### Steps to reproduce.

Set up an I2C config and create a bus handle with it. Afterwards use it for the displayas show in the following code excerpt:
```c
const i2c_master_bus_config_t i2c_bus_config = {
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .i2c_port = PORT_NUMBER,
    .scl_io_num = CONFIG_DISPLAY_SSD1306_GPIO_SCL,
    .sda_io_num = CONFIG_DISPLAY_SSD1306_GPIO_SDA,
    .flags.enable_internal_pullup = 1,
};

i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_bus_config, &bus_handle));

esp_lcd_panel_io_handle_t io_handle = NULL;

esp_lcd_panel_io_i2c_config_t io_config = {
    .dev_addr = CONFIG_DISPLAY_SSD1306_I2C_HW_ADDR,
    .control_phase_bytes = 1,
    .lcd_cmd_bits = DISPLAY_CMD_BITS,
    .lcd_param_bits = DISPLAY_PARAM_BITS,
    .dc_bit_offset = 6,
};

ESP_ERROR_CHECK(esp_lcd_new_panel_io_i2c(bus_handle, &io_config, &io_handle));
```

suda-morris, 2023-10-16:
@webcompas Please check if there are any places in your project is include the `#include "driver/i2c.h"` header file? For using the new I2C Master driver, you need only the `#include "driver/i2c_master.h"`

webcompas, 2023-10-24:
My apologies. For some reason the source file using the old driver has unexpectedly been added to the CMakeLists.txt and hence has been compiled. After removing it, the warning doesn't appear anymore.

closed 2023-10-24 by webcompas