espressif/esp-idf #12407: New version of QSPI example error

2339212597, 2023-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

commit e088c3766ba440e72268b458a68f27b6e7d63986 (HEAD, tag: v5.1.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Download from official website [iot-solution](https://github.com/espressif/esp-iot-solution)  
commit 834dd6cec70772a1a62b828b97982eeb56240dd5 (HEAD -> master)
Use the **examples\display\lcd\qspi_with_ram** example,use VSCODE build is compile error
 error: 'LCD_RGB_ELEMENT_ORDER_BGR' undeclared (first use in this function)
   77 |     case LCD_RGB_ELEMENT_ORDER_BGR:
No changes have been made to the project,It looks like some file definitions are missing,I found 
definitions in the file **components\hal\include\hal\lcd_types.h** in my **idf component**, but the 
example does not include this header file directly, nor does it define the properties in this file

### What is the actual behavior?

build it

### Steps to reproduce.

1. open **examples\display\lcd\qspi_with_ram**
2. build it

### Build or installation Logs.

```plain
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
Processing 9 dependencies:
[1/9] esp_lcd_gc9b71 (0.0.2)
[2/9] esp_lcd_sh8601 (0.0.2)
[3/9] esp_lcd_spd2010 (0.0.2)
[4/9] espressif/cmake_utilities (0.5.2)
[5/9] espressif/esp_lcd_touch (1.1.0)        
[6/9] espressif/esp_lcd_touch_cst816s (1.0.3)
[7/9] espressif/esp_lcd_touch_spd2010 (0.0.1)
[8/9] idf (5.1.1)
[9/9] lvgl/lvgl (8.3.10)
-- Project sdkconfig file C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf

-- App "example_qspi_with_ram" version: 834dd6ce-dirty
-- Adding linker script C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
[... 50 lines cut ...]
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
[14/462] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\AMD\Desktop\esp_iot\examples\display\lcd\qspi_with_ram\build\bootloader\esp-idf\esptool_py && d:\idf_path\5.2tool\python_env\idf5.1_py3.11_env\Scripts\python.exe D:/idf_path/idf/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build/bootloader/bootloader.bin"
Bootloader binary size 0x51f0 bytes. 0x2e10 bytes (36%) free.
[16/460] Building C object esp-idf/esp_lcd_sh8601/CMakeFiles/__idf_esp_lcd_sh8601.dir/esp_lcd_sh8601.c.obj
FAILED: esp-idf/esp_lcd_sh8601/CMakeFiles/__idf_esp_lcd_sh8601.dir/esp_lcd_sh8601.c.obj
ccache D:\idf_path\5.2tool\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build/config -IC:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_sh8601/include -ID:/idf_path/idf/esp-idf/components/newlib/platform_include -ID:/idf_path/idf/esp-idf/components/freertos/FreeRTOS-Kernel/include -ID:/idf_path/idf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/include/freertos -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/include -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/arch/xtensa/include -ID:/idf_path/idf/esp-idf/components/esp_hw_support/include -ID:/idf_path/idf/esp-idf/components/esp_hw_support/include/soc -ID:/idf_path/idf/esp-idf/components/esp_hw_support/include/soc/esp32s3 -ID:/idf_path/idf/esp-idf/components/esp_hw_support/port/esp32s3/. -ID:/idf_path/idf/esp-idf/components/esp_hw_support/port/esp32s3/private_include -ID:/idf_path/idf/esp-idf/components/heap/include -ID:/idf_path/idf/esp-idf/components/log/include -ID:/idf_path/idf/esp-idf/components/soc/include -ID:/idf_path/idf/esp-idf/components/soc/esp32s3 -ID:/idf_path/idf/esp-idf/components/soc/esp32s3/include -ID:/idf_path/idf/esp-idf/components/hal/esp32s3/include -ID:/idf_path/idf/esp-idf/components/hal/include -ID:/idf_path/idf/esp-idf/components/hal/platform_port/include -ID:/idf_path/idf/esp-idf/components/esp_rom/include -ID:/idf_path/idf/esp-idf/components/esp_rom/include/esp32s3 -ID:/idf_path/idf/esp-idf/components/esp_rom/esp32s3 -ID:/idf_path/idf/esp-idf/components/esp_common/include -ID:/idf_path/idf/esp-idf/components/esp_system/include -ID:/idf_path/idf/esp-idf/components/esp_system/port/soc -ID:/idf_path/idf/esp-idf/components/esp_system/port/include/private -ID:/idf_path/idf/esp-idf/components/xtensa/include -ID:/idf_path/idf/esp-idf/components/xtensa/esp32s3/include -ID:/idf_path/idf/esp-idf/components/lwip/include -ID:/idf_path/idf/esp-idf/components/lwip/include/apps -ID:/idf_path/idf/esp-idf/components/lwip/include/apps/sntp -ID:/idf_path/idf/esp-idf/components/lwip/lwip/src/include -ID:/idf_path/idf/esp-idf/components/lwip/port/include -ID:/idf_path/idf/esp-idf/components/lwip/port/freertos/include -ID:/idf_path/idf/esp-idf/components/lwip/port/esp32xx/include -ID:/idf_path/idf/esp-idf/components/lwip/port/esp32xx/include/arch -ID:/idf_path/idf/esp-idf/components/esp_lcd/include -ID:/idf_path/idf/esp-idf/components/esp_lcd/interface -ID:/idf_path/idf/esp-idf/components/driver/include -ID:/idf_path/idf/esp-idf/components/driver/deprecated -ID:/idf_path/idf/esp-idf/components/driver/analog_comparator/include -ID:/idf_path/idf/esp-idf/components/driver/dac/include -ID:/idf_path/idf/esp-idf/components/driver/gpio/include -ID:/idf_path/idf/esp-idf/components/driver/gptimer/include -ID:/idf_path/idf/esp-idf/components/driver/i2c/include -ID:/idf_path/idf/esp-idf/components/driver/i2s/include -ID:/idf_path/idf/esp-idf/components/driver/ledc/include -ID:/idf_path/idf/esp-idf/components/driver/mcpwm/include -ID:/idf_path/idf/esp-idf/components/driver/parlio/include -ID:/idf_path/idf/esp-idf/components/driver/pcnt/include -ID:/idf_path/idf/esp-idf/components/driver/rmt/include -ID:/idf_path/idf/esp-idf/components/driver/sdio_slave/include -ID:/idf_path/idf/esp-idf/components/driver/sdmmc/include -ID:/idf_path/idf/esp-idf/components/driver/sigma_delta/include -ID:/idf_path/idf/esp-idf/components/driver/spi/include -ID:/idf_path/idf/esp-idf/components/driver/temperature_sensor/include -ID:/idf_path/idf/esp-idf/components/driver/touch_sensor/include -ID:/idf_path/idf/esp-idf/components/driver/twai/include -ID:/idf_path/idf/esp-idf/components/driver/uart/include -ID:/idf_path/idf/esp-idf/components/driver/usb_serial_jtag/include -ID:/idf_path/idf/esp-idf/components/driver/touch_sensor/esp32s3/include -ID:/idf_path/idf/esp-idf/components/esp_pm/include -ID:/idf_path/idf/esp-idf/components/esp_ringbuf/include -mlongcalls  -fdiagnostics-color=always -Wno-format -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram=. -fmacro-prefix-map=D:/idf_path/idf/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversi [... cut ...]
C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_sh8601/esp_lcd_sh8601.c:77:10: error: 'LCD_RGB_ELEMENT_ORDER_BGR' undeclared (first use in this function)
   77 |     case LCD_RGB_ELEMENT_ORDER_BGR:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
[22/460] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sdl/lv_draw_sdl_composite.c.obj
ninja: build stopped: subcommand failed.
```

### More Information.

The missing parts of the project are there, but the direct compilation error

suda-morris, 2023-10-16:
it's because the gc9b71 driver is using a type that doesn't exist on the tag/5.1.1 .  The type was backported to release/v5.1 branch in https://github.com/espressif/esp-idf/commit/fefa61a7c754b9ea75c5506de027ab751a1f4350

So a quick solution for you is to switch to the release/v5.1 branch.

Or update the gc9b71 driver to use the legacy type: LCD_RGB_ELEMENT_ORDER_RGB --> LCD_RGB_ENDIAN_RGB @Lzw655

Lzw655, 2023-10-16:
@suda-morris The QSPI LCD feature and `LCD_RGB_ELEMENT_ORDER_RGB ` were merged meanwhile. It means only release/v5.1 or release/v5.0 can support this feature.  So It's not useful even to use the legacy type.

BTW, because you have opened an [issue in esp-iot-solution](https://github.com/espressif/esp-iot-solution/issues/303), please close this one. Thank you! @skyoffcx

skyoffcx, 2023-10-16:
> 这是因为 GC9B71 驱动程序使用的类型在 tag/5.1.1 上不存在。该类型已向后移植到 [fefa5a](https://github.com/espressif/esp-idf/commit/fefa61a7c754b9ea75c5506de027ab751a1f4350) 中的 release/v1.61 分支
> 
> 因此，一个快速的解决方案是切换到 release/v5.1 分支。
> 
> 或者更新 gc9b71 驱动程序以使用旧类型：LCD_RGB_ELEMENT_ORDER_RGB --> LCD_RGB_ENDIAN_RGB
  thanks for your reply，Line 12 in my compilation log is  mine idf version ，Maybe my current ability may not be able to get any closer to solving the problem，Can you give some suggestions on configuration files or configuration methods？

skyoffcx, 2023-10-16:
thanks for your reply，Line 12 in my compilation log is mine idf version ，Maybe my current ability may not be able to get any closer to solving the problem，Can you give some suggestions on configuration files or configuration methods？

skyoffcx, 2023-10-16:
> @suda-morris The QSPI LCD feature and `LCD_RGB_ELEMENT_ORDER_RGB ` were merged meanwhile. It means only release/v5.1 or release/v5.0 can support this feature. So It's not useful even to use the [...]

Can you turn it off for me

closed 2023-10-17 by suda-morris