espressif/esp-idf #12408: BLE MTU issue with android phone

DixitChinmay, 2023-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am using **bleprph_throughput** example from nimble. The function **ble_att_set_preferred_mtu(PREFERRED_MTU_VALUE)** works fine with ios system but for android, i am not getting any response for the same like I get for ios in **BLE_GAP_EVENT_MTU** event. I tried to reduce **PREFERRED_MTU_VALUE** as well but no luck. The accepted MTU size is important for me to decide the packet size while notifying to the mobile app.
Can anyone tell me what and where I am doing wrong?

I am attaching my logs. In that, the first two connections are with ios and the last one with andriod with connection params update request but no response for ble_att_set_preferred_mtu.
Thanks in advance.
Chinmay.

[image]

rahult-github, 2023-10-15:
Hi @DixitChinmay ,

ble_att_set_preferred_mtu is used to set the local MTU . Once configured, this MTU is the value that would be presented to remote in case of exchange MTU handshake.  

MTU update procedure has to be explicitly initiated. In general, when a MTU is set, during exchange MTU handshake, both sides would share their MTU values and the lowest of the two would be selected. The exchange MTU sequence has to be initiated for this. In case of android phone, may be the exchange MTU is not getting initiated in the device application and hence no MTU update is happening.  

You can use a third party device like nrfConnect which presents a option to exchange MTU from client side.  You should then be able to observe the MTU update events on peripheral side.  Below is one example where using "Request MTU " on client side for possible values, would show up the events on peripheral side. 

[image]

DixitChinmay, 2023-10-15:
Hi Rahul,
Thank you so much for the reply. I was not aware about the fact that we have to turn on the setting to have the MTU request from the client(mobile app). So by turning it on, we are able to get MTU size as 512 bytes on android as well. For turning on the MTU setting in nrfConnect, Go to settings-> Connectivity and you will be able to see that option to auto send maximum MTU request from mobile app.
Closing this issue. Thanks again!!
Chinmay

DixitChinmay, 2023-10-15:
Hi Rahul,
Thank you so much for the reply. I was not aware about the fact that we have to turn on the setting to have the MTU request from the client(mobile app). So by turning it on, we are able to get MTU size as 512 bytes on android as well. For turning on the MTU setting in nrfConnect, Go to settings-> Connectivity and you will be able to see that option to auto send maximum MTU request from mobile app.
Closing this issue. Thanks again!!
Chinmay.

closed 2023-10-15 by DixitChinmay