espressif/esp-idf #12409: Oneshot ADC log output can't be controlled separately

cyberhuman, 2023-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

When testing ADC on my board, I noticed calling `adc_oneshot_read` produces two lines on every invocation:
```
D (379929) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (379929) adc_hal: delay: 3
```
I have `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y` but I expect the log level of adc_hal to be controllable separately from the default log level.

I noticed there exists a `CONFIG_HAL_LOG_LEVEL` setting but it's not configurable as soon as `LOG_DEFAULT_LEVEL` is defined.

Another issue is that `adc_oneshot_hal` uses `HAL_EARLY_LOGD` instead of `HAL_LOGD`; the former doesn't check `CONFIG_HAL_LOG_LEVEL` and calls `ESP_EARLY_LOGD` instead, which checks the default log level (and `LOG_LOCAL_LEVEL`, which isn't set by `adc_oneshot_hal` and thus defaults to `LOG_MAXIMUM_LEVEL`).

### What is the actual behavior?

when `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`, it's not possible to suppress the output of `adc_oneshot_hal`.

### Steps to reproduce.

1. get the ADC `oneshot_read` example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/adc/oneshot_read
2. run `idf.py menuconfig`
3. set the "Default log verbosity" setting to Debug 
4. notice the "HAL layer log verbosity" setting is not available
5. run the example and observe the debug output from `adc_hal`.

### Debug Logs.

```plain
...
D (400250) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (400250) adc_hal: delay: 3
D (400289) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (400289) adc_hal: delay: 3
D (401268) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (401268) adc_hal: delay: 3
D (401307) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (401307) adc_hal: delay: 3
D (402286) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (402286) adc_hal: delay: 3
D (402325) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (402325) adc_hal: delay: 3
D (403304) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (403304) adc_hal: delay: 3
D (403343) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (403343) adc_hal: delay: 3
D (404322) adc_hal: clk_src_freq_hz: 80000000, digi_clk: 5000000, delay: 3
D (404322) adc_hal: delay: 3
...
```

Icarus113, 2023-10-31:
Hi again @cyberhuman ,

- If you use the ESP-IDF project, the `log` component is always included, so the hal layer log will also use the `log` implementation.
- If you only use the ESP-IDF `hal` component, then `CONFIG_HAL_LOG_LEVEL` is configurable.

> Another issue is that adc_oneshot_hal uses HAL_EARLY_LOGD instead of HAL_LOGD

Yes, this is intended as this hal function is in critical section. And the “EARLY” log macros do not support per module setting of log verbosity. These macros will always log at the “default” verbosity level. 

So in a word, we will not make this hal layer function log to be controlled separately.

On the other hand, we can provide a workaround by controlling the `adc_oneshot` driver logs independently. Then you can see `adc_oneshot` driver layer debug logs, without seeing the hal layer debug logs.

cyberhuman, 2023-10-31:
Hi @Icarus113!
Thank you for the explanation.

> On the other hand, we can provide a workaround by controlling the `adc_oneshot` driver logs independently.

This would be nice!

Bruce297, 2023-11-02:
Now there is a new commit that add a config `ADC_ENABLE_DEBUG_LOG` allow you to control  the `adc_oneshot` driver logs independently. If you enable it, you can see adc_oneshot` driver debug logs without `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`

Icarus113, 2023-11-08:
It's done in this commit https://github.com/espressif/esp-idf/commit/c14d7fbb17b00a28b21ee8e19e451e6ab7f6801e

We'll close this issue, feel free to reopen if there's any further issue. @cyberhuman

closed 2023-11-08 by Icarus113