espressif/esp-idf #12410: MQTT broker cannot connect

SFNFIH, 2023-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My ESP-IDF version is v5.1.1-dirty
I used the mqtt case program in examples, but when connecting to the mqtt broker I built, I was prompted with this paragraph:
```
E (4047) mqtt_client: No scheme found
E (4047) mqtt_client: Failed to create transport list
```
Below is my configuration code.
```
static void mqtt_app_start(void)
{
     esp_mqtt_client_config_t mqtt_cfg = {
         .broker.address.hostname = "mqtt://192.168.239.20:1883",
     };
     esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
     /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
     esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
     esp_mqtt_client_start(client);
}
```

nopnop2002, 2023-10-14:
```
     esp_mqtt_client_config_t mqtt_cfg = {
         .broker.address.uri = "mqtt://192.168.239.20",
        .broker.address.port = 1883
     };
```

SFNFIH, 2023-10-15:
Thank you for your suggestion,i already tried，But still can't connect. @nopnop2002

nopnop2002, 2023-10-15:
Is your MQTT broker valid?

$ mosquitto_pub -h 192.168.239.20 -p 1883 -t "/test/test" -m "test"

SFNFIH, 2023-10-15:
I have tested it with mqttx and can connect to the broker.

nopnop2002, 2023-10-15:
In mosquitto V1.7 and later, allow_anonymous is set to false by default, and external connections are rejected.

SFNFIH, 2023-10-16:
My MQTT agent is built with EMQX. I didn't configure its connection strategy, and I didn't even set a user name and password.[This is its official website.](https://www.emqx.io/)

nopnop2002, 2023-10-16:
EMQX mqtt broker allows external connections by default.

There may be a problem with your Proxy

SFNFIH, 2023-10-16:
But I can connect to the mqtt agent with any mqtt test software on my computer or mobile phone.

nopnop2002, 2023-10-16:
Can you connect to "mqtt://broker.emqx.io" ?

SFNFIH, 2023-10-16:
Any proxy that is not in the form of IP can be connected, and a single proxy in the form of IP cannot be connected.

nopnop2002, 2023-10-16:
Can you connec here.
mqtt://54.244.173.190

SFNFIH, 2023-10-16:
no，Still can't connect

nopnop2002, 2023-10-16:
It is your network problem.

Becase broker.emqx.io is 54.244.173.190.

```
ping broker.emqx.io
PING emqx.public-broker.com (54.244.173.190) 56(84) bytes of data.
```

SFNFIH, 2023-10-16:
I don’t know the reason, but my mqtt broker is built locally, and the connection is also a LAN connection. I tried to connect to mqtt://broker.emqx.io, but the connection was successful.

nopnop2002, 2023-10-16:
Connect to mqtt://broker.emqx.io ---> Success
Connect to mqtt://54.244.173.190 ---> Fail

I don't know why, but your WiFi router can't connect to the server by IP address.

SFNFIH, 2023-10-16:
I use the hotspot on my mobile phone to access the Internet，I don't know if it's a problem with the phone

nopnop2002, 2023-10-16:
Maybe your hotspot doesn't support connections using IP addresses.

SFNFIH, 2023-10-16:
Why can I connect to the agent with another computer or mobile phone on the same network?

SFNFIH, 2023-10-16:
Have you tried to connect the proxy with IP?

nopnop2002, 2023-10-16:
>Why can I connect to the agent with another computer or mobile phone on the same network?

This place is inappropriate for your question.
This place is for ESP-IDF issues.

>Have you tried to connect the proxy with IP?

No.

SFNFIH, 2023-10-16:
Yes, the computer and mobile phone can be connected normally.

SFNFIH, 2023-10-16:
```
I (6057) esp_netif_handlers: sta ip: 192.168.38.31, mask: 255.255.255.0, gw: 192.168.38.222
I (6057) MQTT_EXAMPLE: got ip:192.168.38.31
I (6057) WIFI: WIFI连接成功! 马上启动MQTT连接！
```
this is my record.

nopnop2002, 2023-10-16:
>I (6057) esp_netif_handlers: sta ip: 192.168.38.31, mask: 255.255.255.0, gw: 192.168.38.222

Since the mqtt server (192.168.239.20) is different from Segumen, it is not possible to connect to the mqtt server.

SFNFIH, 2023-10-17:
So how should I solve this problem? Does ESP-IDF have an API for setting IP?

nopnop2002, 2023-10-17:
You can't change network segments.

SFNFIH, 2023-10-17:
Why is the connected hotspot the same, but it is assigned different network segments? This is strange!

nopnop2002, 2023-10-17:
I don’ｔ know.

closed 2023-10-17 by SFNFIH