espressif/esp-idf #12412: "Root port reset failed" on slow starting USB device

Crazor, 2023-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3 (revision v0.2)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-devkitc

### Power Supply used.

USB

### What is the expected behavior?

I'm trying to connect to an ATMEGA32U4 based USB device (to which I don't have the firmware sources) via the S3's USB host mode. The USB device itself has two modes, let's call them bootloader and application. If I force the device to enter the bootloader mode, it enumerates on the S3 just fine. If I let it boot it's application mode (the one I'm interested in talking to), all I get on the S3 is `HUB: Root port reset failed`.

### What is the actual behavior?

I looked at the D+/D- lines with an oscilloscope and noticed a timing difference between the two modes when I hook it up to a PC. In bootloader mode, the device starts talking to the host immediately, i.e. after a couple of milliseconds. In application mode, it takes the device approximately 3 seconds to start talking. If I hook the device up to the S3, I see similar behaviour when in bootloader mode, but no bus activity at all when trying to start the application mode.

### Steps to reproduce.

I tried increasing the configurable timeouts, i.e. `CONFIG_USB_HOST_DEBOUNCE_DELAY_MS`, `CONFIG_USB_HOST_RESET_HOLD_MS` and `CONFIG_USB_HOST_RESET_RECOVERY_MS`, but that does not seem to make any difference.

tore-espressif, 2023-10-23:
Hello @Crazor 

> I tried increasing the configurable timeouts, i.e. CONFIG_USB_HOST_DEBOUNCE_DELAY_MS, CONFIG_USB_HOST_RESET_HOLD_MS and CONFIG_USB_HOST_RESET_RECOVERY_MS, but that does not seem to make any difference.

How much did you increase those timeouts? Did you try setting `CONFIG_USB_HOST_DEBOUNCE_DELAY_MS` > 3500 ?

This should give your device enough time to boot

Crazor, 2023-10-23:
Hi @tore-espressif 

I did try to increase `CONFIG_USB_HOST_DEBOUNCE_DELAY_MS` to 5000, but it didn't work. I will try this again. Could you provide some pointers on what to watch out for with either the debugger or with a scope on the USB lines to make sure this is the correct "knob" to turn?

roma-jam, 2023-10-23:
Hi everyone,

The main device logic part started after first device reset. So literally, if there is a chance to fix this with increasing the configurable timeouts, it will be either `CONFIG_USB_HOST_RESET_RECOVERY_MS` or `CONFIG_USB_HOST_SET_ADDR_RECOVERY_MS`.

But, if there is a chance to distinguish the specific moment, after which logic the delay is happened on the device it will help a lot.
Mainly, to answer the question "does that delay happened before USB setting the device address or after"?.

Crazor, 2023-10-23:
I don't think the bootloader is even using the USB port in any way, except when I set the boot jumper. With that kind of delay before the actual firmware connects, I would assume that I should see it appearing for a second or two under Windows, or hear the USB chimes showing connection-disconnection-connection sequence on normal boot. I think I need to test this under Linux and look at the syslog. 
So far I don't have a good way to capture USB traffic in hardware. Wireshark should be no problem to set up. Is there a recommended setup for hardware capture on the line that does not break the bank? I suppose a Saleae is not fast enough, but I got to look into that.

roma-jam, 2023-10-23:
@Crazor,

I see. 

I though that the problem is when device changes the work mode from bootloader to application and both share USB.
Thus, the decreasing delays won’t help, obviously. 

Yes, to start, Wireshark trace could help. 
For both bootloader and application attach process.

roma-jam, 2023-10-23:
@Crazor 

But before, probably to increase the `CONFIG_USB_HOST_RESET_RECOVERY_MS` value is worth trying. 
I am still not able to reproduce the problem quickly, but here is a timings diagram that can help to understand the delays and how they could affect the enumeration process. Increasing the `CONFIG_USB_HOST_RESET_RECOVERY_MS` doesn't affect the enumeration completion success, just increase the time of it. At least I have tried it with the value `5000`.

[image]

And feel free to share the Wireshark tracelog, it may help us to move forward.

roma-jam, 2023-11-07:
Hi @Crazor, 

Is there any updates on the problem?

Crazor, 2023-11-08:
Haven't made an inch of progress so far, still working on it. Hopefully I can provide more detailed info on what's going on during next weekend.

closed 2024-01-09 by roma-jam

roma-jam, 2024-01-09:
Closing the issue as there not enough information to reproduce. 

Please, feel free to reopen in case there will anything else.

antunesls, 2024-08-05:
I've had the same issue. After days looking for a solution.

I solved it link the pad USB- OTG.

MillerRen, 2025-10-03:
same issue when device delay start or switch mode.  u disk works well but some video player not.