espressif/esp-idf #12414: "BLE_INIT Malloc failed" error when WIFI and BLE work at the same time

serdartoren, 2023-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1-N8R8

### Power Supply used.

USB

### What is the expected behavior?

Avoid getting "BLE_INIT Malloc failed" error when a new BLE communication starts.

### What is the actual behavior?

When Wifi and BLE are working at the same time and when paired with a phone i get the "BLE_INIT Malloc failed" error.

It seems like there is insufficient internal RAM but there is RAM.

//// *esp/esp-idf/components/bt/controller/esp32c3/bt.c* ////
static void *malloc_internal_wrapper(size_t size)
{
    //void *p = heap_caps_malloc(size, MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA); 
    void *p = heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA); 
    if(p == NULL) {
        ESP_LOGE(BT_LOG_TAG, "Malloc failed");
 }
////

The problem is solved by not using the MALLOC_CAP_DEFAULT macro in the heap_caps_malloc() function.
Do you think this will cause a problem? What does this macro do?

> heap_1 = heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);

> heap_2 = heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);

I measured the remaining RAM around the times the error occurred and found that heap_2 was always larger than heap_1.
Why is the size of heap_2 larger than heap_1?
So I changed the original ESP-IDF code a bit, the problem seems to be solved but I'm wondering why and if it will have side effects.

### Steps to reproduce.

1. Connect to an AP(hotspot)
2. Pair with a phone with your BLE code set as in this example
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_client
3. If this is not enough, pair with 1 more phone or turn off and on the current pairing.
4.I started having this problem at the end of a large project.
To reproduce this issue, allocating large blocks of RAM with malloc may help.

### More Information.

To begin with, I just wanted to explain this situation and I can create a sample project based on your answer.

Have a nice day.

Shaco-Ma, 2023-11-13:
I Get the same issues, how to fixed it?

yanmartins, 2023-11-21:
Any news about this?

zhp0406, 2023-11-22:
@yanmartins 
Have you also encountered this issue? I am currently trying to reproduce the problem.

yanmartins, 2023-11-22:
> @yanmartins 
> Have you also encountered this issue? I am currently trying to reproduce the problem.

Yes. I'm using the WiFi Mesh network and I use Bluetooth on the same device.

And this error appears sometimes (not always).

zhp0406, 2023-11-22:
@yanmartins 
Is the issue you mentioned occurring frequently on your end? I would greatly appreciate it if you could provide me with a reproducible example. 
Additionally, please specify the exact chip and IDF version you are using (currently unsure if it is related to the chip and IDF version).

yanmartins, 2023-11-22:
> @yanmartins 
> Is the issue you mentioned occurring frequently on your end? I would greatly appreciate it if you could provide me with a reproducible example. 
> Additionally, please specify the exact chip and IDF version you are using (currently unsure if it is related to the chip and IDF version).

I'm using the ip_internal_network example to generate the Mesh Network. Then I make connections with Bluetooth devices. I can't provide my code.

Chip: ESP32-S3
IDF: 5.0.4

zhp0406, 2023-11-29:
In the previous approach, the memory used by BLE needs to satisfy the capabilities of MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA. If we eliminate the MALLOC_CAP_DEFAULT capability, more memory becomes available. 
Thank you for your feedback; we will evaluate whether to disable the MALLOC_CAP_DEFAULT capability.
Please disable MALLOC_CAP_DEFAULT in your current usage. If you encounter any issues with memory allocation again, please let me know.

serdartoren, 2024-01-03:
Inside the bt.c file I made this 
`    void *p = heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);`
instead
`    void *p = heap_caps_malloc(size, MALLOC_CAP_DEFAULT|MALLOC_CAP_INTERNAL|MALLOC_CAP_DMA);`
I have been using it for 2 months without any problems.

I also started using some tasks in PSRAM.
This ones solved my problem.

I hope this discussion will be useful to other people, I think the issues can be closed now.

closed 2024-01-03 by serdartoren

brentpicasso, 2024-01-05:
Hi, will this patch make its way into the next version of the ESP IDF?

zhp0406, 2024-01-05:
@brentpicasso 
yes

MuhammadTaseen, 2024-05-14:
I have also resolve the same issue while using PSRAM for some tasks using heap_caps_mallos just like that

 xStack = (StackType_t *)heap_caps_malloc((2 * 1024 * 1024), MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT);

xTaskCreateStatic(XYZ_Task, "XYZ_TASK", 2 * 1024 * 1024, NULL, 10, xStack, &xTaskBuffer);

zhp0406, 2024-05-15:
fixed:
master 0f29b3cef97d406c5887d79da8592308540cdda0
v5.2 62b84e3169a7d053c1865a2018eb1d4b3accba11
v5.1 d6d6963fea8571b0b470101058aff53d2f01e1a5
v5.0 e37dcaa92ffba4ee0af1e1858ae5ff4861b51dce
v4.4 b13dfdceed79ea864b1721afeb9ee65e552b8571