espressif/esp-idf #12415: Issue in Local Firmware upgrade from Android App.

Rbhagwat62, 2023-10-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

We are using ESP 32 as a web server in local (direct ) mode, Android App is used in client mode.  When we are trying to do the firmware upgrade from the Android app, after a few POST requests, the device is receiving data 0 and at the lwip level it is returning an error -15 (NETCONN_FIN_RX_PENDING). ESP32 is acting as AP for the Android Mobile app.

ESP-IDF version is 4.4.5

Enabled HTTP server mode.

We are experiencing this only on few devices.

Following is the snapshot of log where it is sending error,
------------------------------------------------------
D (284428) lwip: lwip_recvfrom(59, 0x3ffd6984, 6145, 0x0, ..)[0m

D (284434) lwip: lwip_recv_tcp: top while sock->lastdata=0x0[0m

D (284440) lwip: netconn_recv_data: received 0x3ffd33d8, len=31[0m

D (284446) lwip: lwip_recv_tcp: netconn_recv err=0, pbuf=0x3ffd33d8[0m

D (284453) lwip: lwip_recv_tcp: buflen=31 recv_left=6145 off=0[0m

D (284459) lwip: lwip_recv_tcp: deleting pbuf=0x3ffd33d8[0m

D (284464) lwip: pbuf_free(0x3ffd33d8)[0m

D (284469) lwip: pbuf_free: deallocating 0x3ffd33d8[0m

D (284474) lwip: lwip_recv_tcp: top while sock->lastdata=0x0[0m

Debug ERR_OK CLOSED

Debug : buff NULL

Debug : buff NULL -- ERR_WOULDBLOCK - NETCONN_FIN_RX_PENDING

D (284489) lwip: lwip_recv_tcp: netconn_recv err=-7, pbuf=0x0[0m

D (284495) lwip: tcp_recved: received 31 bytes, wnd 5744 (0).[0m

D (284501) lwip: lwip_recvfrom(59):  addr=[0m

D (284505) lwip: 192.168.4.2[0m

D (284508) lwip:  port=60342 len=31[0m

D (284512) lwip: lwip_recvfrom(59, 0x3ffd69a3, 6114, 0x0, ..)[0m

D (284518) lwip: lwip_recv_tcp: top while sock->lastdata=0x0[0m

Debug : NETCONN_FIN_RX_PENDING

D (284527) lwip: lwip_recv_tcp: netconn_recv err=-15, pbuf=0x3f4ec6d8[0m

D (284533) lwip: lwip_recv_tcp: p == NULL, error is "Connection closed."![0m

D (284540) lwip: lwip_recvfrom(59):  addr=[0mD (284544) lwip: 192.168.4.2[0m

D (284547) lwip:  port=60342 len=0[0m

D (284551) httpd_txrx: httpd_recv_with_opt: received length = 0[0m

D (284557) httpd_txrx: httpd_req_recv: received length = 0[0m

************ FIRMWARE UPDATING ******************* length 0 and count 673

hmalpani, 2023-10-26:
Hello @Rbhagwat62 
I tried to reproduce this issue. I am testing using the project: https://github.com/hmalpani/HTTP_Server_OTA

I am using cURL instead of Android App to push the binary to the device (acting as server). I was able to do OTA successfully. Can you please try doing OTA using cURL as a client. If that works, there might be some issue in the Android App. It using cURL still gives you error, can you share the minimal application code to reproduce this issue?

Rbhagwat62, 2023-10-31:
Hello @hmalpani,

Thank you for your reply.

We have tested given sample example. It is working as expected with firmware size 2MB. 

In our project we are using secure connection for HTTP using secure element and we have enabled HTTP server component with LRU PURGE is true and SO_LINGER enabled. Our firmware size is of 4MB. We have not tested our firmware with curl because of secured HTTP connection. We are facing issue in **POST API in local Mode**, after few requests it is receiving 0 expected length is 1024.

Can you tell us when below log condition can occur?

Debug ERR_OK CLOSED

Debug : buff NULL

Debug : buff NULL -- **ERR_WOULDBLOCK - NETCONN_FIN_RX_PENDING**

D (284489) lwip: lwip_recv_tcp: netconn_recv err=-7, pbuf=0x0�[0m

D (284495) lwip: tcp_recved: received 31 bytes, wnd 5744 (0).�[0m

D (284501) lwip: lwip_recvfrom(59): addr=�[0m

D (284505) lwip: 192.168.4.2�[0m

D (284508) lwip: port=60342 len=31�[0m

D (284512) lwip: lwip_recvfrom(59, 0x3ffd69a3, 6114, 0x0, ..)�[0m

D (284518) lwip: lwip_recv_tcp: top while sock->lastdata=0x0�[0m

Debug : **NETCONN_FIN_RX_PENDING**

Rbhagwat62, 2023-11-30:
The issue has been identified in flash erase cycle during firmware upgrade causing delay. Increasing App connection connection timeout resolve the issue.

closed 2023-11-30 by Rbhagwat62