espressif/esp-idf #12416: gatt_security_server working on ESP32 but not functional ESP32-S3

firmware-lyev, 2023-10-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

gatt_security_server working seamless in ESP32 but not in ESP32-S3. Quite stuck, any resolution will be helpful.

zhp0406, 2023-10-18:
@firmware-lyev 
Thank you for reporting the issue.
Could you please provide me with your complete log?

firmware-lyev, 2023-10-23:
I have resolved the issue, was messed-up with configuration in gatts_db