espressif/esp-idf #12421: ESP32 HTTP OTA esp_tls_conn_read error

Focalcrest-Madi, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Espressif SoC revision.

ESP32C3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I am using ESP32 for HTTP OTA upgrade. The OTA server uses AWS. There is a small probability that the upgrade fails during the OTA upgrade of the ESP device. The OTA failure log content is "TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11". What is the reason for this?

### What is the actual behavior?

I am performing repeated OTA upgrades on the ESP32C3 device.

### Steps to reproduce.

I am performing repeated OTA upgrades on the ESP32C3 device

### Debug Logs.

```plain
(242378) HTTP_CLIENT: http_on_body 289
D (242388) esp_https_ota: Written image length 908630
D (242388) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242398) TRANSPORT_BASE: remain data in cache, need to read again
D (242398) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (242418) HTTP_CLIENT: http_on_body 289
D (242418) esp_https_ota: Written image length 908919
D (242428) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242428) TRANSPORT_BASE: remain data in cache, need to read again
D (242438) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (242438) HTTP_CLIENT: http_on_body 289
D (242448) esp_https_ota: Written image length 909208
D (242448) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242458) TRANSPORT_BASE: remain data in cache, need to read again
D (242468) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (242478) HTTP_CLIENT: http_on_body 289
D (242478) esp_https_ota: Written image length 909497
D (242478) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242488) TRANSPORT_BASE: remain data in cache, need to read again
D (242498) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (242498) HTTP_CLIENT: http_on_body 289
D (242548) esp_https_ota: Written image length 909786
D (242548) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242548) TRANSPORT_BASE: remain data in cache, need to read again
D (242568) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=289, ridx=0
D (242578) HTTP_CLIENT: http_on_body 289
D (242578) esp_https_ota: Written image length 910075
D (242578) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (242588) TRANSPORT_BASE: remain data in cache, need to read again
D (242598) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=7, ridx=0
D (242598) HTTP_CLIENT: http_on_body 7
D (242598) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (244058) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (244158) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
I (245698) HLW8112: read A channel IA_Reg[206] IAC_Reg[caf4]
W (245698) HLW8112: Real-time I[0][0.003200]
D (248058) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (248158) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
I (250798) HLW8112: read A channel IA_Reg[1f4] IAC_Reg[caf4]
W (250798) HLW8112: Real-time I[0][0.003089]
D (253658) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (253758) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
E (255218) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
D (255218) HTTP_CLIENT: need_read=282, byte_to_read=282, rlen=-26880, ridx=7
W (255228) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11 
D (255228) HTTP_CLIENT: Received EAGAIN! rlen = -26880, errno 11
D (255238) esp_https_ota: Written image length 910082
D (255238) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
I (255898) HLW8112: read A channel IA_Reg[1ed] IAC_Reg[caf4]
W (255898) HLW8112: Real-time I[0][0.003046]
D (260248) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=0, ridx=0
D (260248) HTTP_CLIENT: Data processed 909793 != Data specified in content length 1918528
D (260258) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
I (260998) HLW8112: read A channel IA_Reg[1ed] IAC_Reg[caf4]
W (260998) HLW8112: Real-time I[0][0.003046]
D (260998) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (261058) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (261158) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (261258) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (262658) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (262758) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (265258) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=0, ridx=0
D (265258) HTTP_CLIENT: Data processed 909793 != Data specified in content length 1918528
D (265258) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (265358) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (265458) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
I (266098) HLW8112: read A channel IA_Reg[1ee] IAC_Reg[caf4]
W (266098) HLW8112: Real-time I[0][0.003052]
D (266858) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (266958) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (267458) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (267558) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
I (271198) HLW8112: read A channel IA_Reg[1e1] IAC_Reg[caf4]
W (271198) HLW8112: Real-time I[0][0.002971]
E (271588) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
D (271588) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=-26880, ridx=0
W (271598) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11 
D (271598) HTTP_CLIENT: Received EAGAIN! rlen = -26880, errno 11
D (271608) HTTP_CLIENT: Data processed 909793 != Data specified in content length 1918528
D (271618) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1918528
D (272058) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (272158) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (272758) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (272858) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (273658) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
D (273758) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcb3e80
```

hmalpani, 2023-10-18:
Hello @Focalcrest-Madi 
Can you please share the minimal application to reproduce this issue. I tried simple_ota example on v4.4.2, but was unable to reproduce this issue. Please also share the sdkconfig with which you are able to reproduce this issue.

hmalpani, 2023-11-21:
Hello @Focalcrest-Madi 
Closing this issue. Please feel free to reopen.

closed 2023-11-21 by hmalpani