espressif/esp-idf #12422: Can I understand the interaction between Thread devices by using two ESP32 H2 devices, such as controlling RGB lights with each other? Are there any relevant examples

shuiYe-704265085, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If there are no relevant suggestions, can you give me some suggestions

chshu, 2023-11-10:
Thread is a network layer protocol, just like Wi-Fi. An application is required to build Thread products (light, switch, ...).

Matter is currently the most popular application runs over Thread, please check the light example in esp-matter here: https://github.com/espressif/esp-matter/tree/main/examples/light

If you want to implement some other applicaitons, you can also use the TCP/UDP APIs directly, refer to the udp example: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_udp_socket.c

KEEPHEADU, 2023-11-23:
> Thread is a network layer protocol, just like Wi-Fi. An application is required to build Thread products (light, switch, ...).
>
> Matter is currently the most popular application runs over Thread, [...]

@chshu Hello, I have used the above code to achieve a demo of communicating through TCP on openthread, but the premise of communication is to know the information of the other thread network. Is there any way for the device to automatically discover the surrounding thread network and join it?

I saw in menuconfig that there are options for joiner and commissioner of openthread, is it possible to use these two options?

chshu, 2024-02-23:
> Is there any way for the device to automatically discover the surrounding thread network and join it?

Thread is a secure network, the device should know the Thread network credentials to join it. The application should support a mechanism to provision the network credentials to device.

> I saw in menuconfig that there are options for joiner and commissioner of openthread, is it possible to use these two options?

Yes, it's doable with these options, just consider it as a simple applicaion behavior that hardcode the credentials in firmware, and the Thread network should be formed by the same credentials.

chshu, 2024-02-23:
@shuiYe-704265085 @KEEPHEADU Please reopen if any follow up questions.

closed 2024-02-23 by chshu