espressif/esp-idf #12425: ESP-Prov: add option to config BLE connection parameters

HiFiPhile, 2023-10-18:
### Is your feature request related to a problem?

Our product is powered with energy harvesting, a supercapacitor is used to support RF peak consumption.

During BLE provisioning, once the device is connected the power consumption is too high, if the procedure is too long (user input password slowly) then the supercapacitor is drained and device is powered off.

During BLE advertising the the default interval `adv_params.itvl_min = 0x100;
    adv_params.itvl_max = 0x100;` is OK to maintetaint the power level.

I had to patch the SDK to increase the interval:
```
diff --git a/components/protocomm/src/transports/protocomm_nimble.c b/components/protocomm/src/transports/protocomm_nimble.c
index 9cb524b30b..874c4cbe5e 100644
--- a/components/protocomm/src/transports/protocomm_nimble.c
+++ b/components/protocomm/src/transports/protocomm_nimble.c
@@ -134,6 +134,21 @@ typedef struct {

 static simple_ble_cfg_t *ble_cfg_p;

+static struct ble_gap_upd_params conn_params = {
+    /** Minimum value for connection interval in 1.25ms units */
+    .itvl_min = 100,
+    /** Maximum value for connection interval in 1.25ms units */
+    .itvl_max = 200,
+    /** Connection latency */
+    .latency = 0,
+    /** Supervision timeout in 10ms units */
+    .supervision_timeout = 0x0100,
+    /** Minimum length of connection event in 0.625ms units */
+    .min_ce_len = 0x0100,
+    /** Maximum length of connection event in 0.625ms units */
+    .max_ce_len = 0x0300,
+};
+
 /************************************************************
 * Functions to set and get attr value based on attr Handle  *
 ************************************************************/
@@ -232,6 +247,10 @@ simple_ble_gap_event(struct ble_gap_event *event, void *arg)
                 ESP_LOGE(TAG, "No open connection with the specified handle");
                 return rc;
             }
+            rc = ble_gap_update_params(event->connect.conn_handle, &conn_params);
+            if (rc != 0) {
+                ESP_LOGE(TAG, "Failed to update params; rc = %d", rc);
+            }
             s_cached_conn_handle = event->connect.conn_handle;
         } else {
             /* Connection failed; resume advertising. */

```

### Describe the solution you'd like.

`itvl_min` and `itvl_max` are very useful for energy harvesting devices, add a config option to make this interval configurable.

### Describe alternatives you've considered.

Add SDK patching to the build workflow.

rahult-github, 2023-10-19:
Hi @HiFiPhile 

I feel this approach may lead to some issues.  

1. Hardcoding of these parameters might be working and satisfying your use case, but it may not be applicable for someone else. So hardcoding these values for connection update can risk causing regression for other users / applications.

2.  If we refer to spec  , then for LE Connection update HCI command,  it is mentioned :

 
[image]

 So, even if we pass some values, the link layer can still override those values and use it as per the resources that are available. This becomes more easier for the link layer to handle multiple connection scenario and choose connection parameters accordingly. 

3. The ble_gap_update_params is a nimble API that is exposed to user. So application can choose to invoke this API with needed parameters after connection is completed as per its use case. This is just one possible solution that am suggesting, actual implementation can be application / upper layer specific.

HiFiPhile, 2023-10-19:
Hi @rahult-github 

1. I'm not suggesting hard code these values, instead make them configurable with menuconfig. It should be generic for nimble and bluedroid.
2. In our office we have tested 2 iPhone with iOS 15&16, 4 Android phones from 11 to 13,  so this option is well implemented.
3. In the esp-prov user case nimble api are hiding under protocomm layer and not accessible.

darshan7patel, 2023-10-27:
Thanks. The  third point you mentioned is basically the bottleneck. So, to solve this, I have made a patch that ensures the necessary information is included in the event posted to the application.
 
Now, users or applications can use the connection handle and should be able to further use it as per their use case. In your case, on PROTOCOMM_TRANSPORT_BLE_CONNECTED event, you can then invoke the ble_gap_update_params API from the Nimble stack and pass the desired parameters you want. Kindly check the attached patch. This should help in your case.
[p1.zip](https://github.com/espressif/esp-idf/files/13187016/p1.zip)

HiFiPhile, 2023-11-02:
Hi @darshan7patel,
Thanks for the patch, it works well.

darshan7patel, 2024-01-18:
Thanks, I hope the problem has been fixed.

Indastri, 2024-01-18:
If you have any further problems or issues. Feel free to reopen or create a new issue.

Cheers.

closed 2024-01-18 by Indastri