espressif/esp-idf #12426: EPS32 ADC readings during light sleep mode not possible

borjanbozhinovski, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32­-WROVER­-E module

### Power Supply used.

External 3.3V

### What is the expected behavior?

Running ADC measurement using ULP and enabled power management for tickles idle light sleep mode.

Expected behavior is to be able to get ADC readings in light sleep and normal mode trough the ULP. Using idf v4.4.5 code works as expected. 

We are toggling a GPIO pin each time ULP is woken up. The expected behavior of the ULP cycle, which is triggered each 10ms, can be seen on oscillogram below:
[image]

### What is the actual behavior?

Switching to v4.4.6, a change in the idf code which switches off the SAR ADC peripheral power once the device enters in light sleep mode is introduced: _components/esp_hw_support/sleep_modes.c_ found in the diff: https://github.com/espressif/esp-idf/compare/v4.4.5...v4.4.6?diff=unified

[image]

Because the power is disabled trough `sar_periph_ctrl_power_disable();`, ULP will wait on `adc` instruction until the power is on, which only happens when ESP32 exits from light sleep mode.

Using the same code, but building with idf 4.4.6, changes the behavior and ULP execution time. The actual behavior of the ULP cycle can be seen on oscillogram below:
[image]

Removing this fix from IDF or force enabling/disabling the SAR ADC power inside ULP code before and after reading from ADC:
`WRITE_RTC_REG(SENS_SAR_MEAS_WAIT2_REG, SENS_FORCE_XPD_SAR_S, 2, SENS_FORCE_XPD_SAR_PU) /* Force enable SAR ADC power*/`
`WRITE_RTC_REG(SENS_SAR_MEAS_WAIT2_REG, SENS_FORCE_XPD_SAR_S, 2, SENS_FORCE_XPD_SAR_PD) /* Force disable SAR ADC power`
, will fix it but will cause additional issue, explained in the "More information" section.

### Steps to reproduce.

1. Enable power management for tickles idle light sleep mode
2. Enable ULP and try to read from ADC
3. Build the project with idf 4.4.6
4. ADC readings should be done at the same time intervals(each time ULP is woken up, in our case 10ms)

### More Information.

Also, there is one related issue to this. Removing the SAR ADC power toggling when entering/exiting from light sleep mode, causes **some of the devices** to randomly reset with different reset causes at different time:

1. rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2. Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
3. Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
4. Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

So, only disabling/enabling SAR ADC power fix from the IDF is not a solution, due to the reset issue mentioned above. So the question is, is it allowed to use ADC from ULP when tickles idle light sleep modes is enabled?

CaptivatingCat, 2023-10-23:
I can confirm that the bug is also present in the 5.1.1 release. We are not using configUSE_TICKLESS_IDLE in our case.

I have (for now) reenabled the SAR peripheral before entering sleep by wrapping the `rtc_sleep_init` function. Sample code:

CMakeLists:
```
target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=rtc_sleep_init")
```
The actual wrapping code:
```
#include <esp_private/sar_periph_ctrl.h>
void __real_rtc_sleep_init(rtc_sleep_config_t cfg);
void  __wrap_rtc_sleep_init(rtc_sleep_config_t cfg) {
    sar_periph_ctrl_power_enable();
    __real_rtc_sleep_init(cfg);
}
```
.h file:
```
#include <soc/rtc.h>
void  __wrap_rtc_sleep_init(rtc_sleep_config_t cfg);
```
I have have not had any new crashes, but I have not tested this extensively.

PocropivniiA, 2023-10-27:
Is there a known limitation/reason of why SAR is disabled during sleep or there is none and its intended to be toggled by user for power consumption improvement? 
[image]

kutukvpavel, 2023-12-10:
Can confirm this affects ESP32S3 too, and in a strange way. You can check out #12765 for more info on my experience with this bug.

Short version is: any optimization level higher than `O0` somehow triggers the bug, sometimes in not-so-obvious ways (`Og`). If I set `O0` optimization level (for the app, not the bootloader, of course) in menuconfig code runs as expected even without any workarounds presented above.

patrickmagee, 2024-01-01:
Hi, I'm having a very similar issue and can confirm 5.2.0 does not solve it. I will try the workaround later, but has this been solved in the master branch?

cdollar393, 2024-01-01:
I recently refactored my code to use the newer ADC one-shot driver instead of the legacy ADC driver, and am no longer seeing any issues reading the ADC during deep-sleep from a ULP program. I was able to get rid of the workaround mentioned above, along with another workaround I'd been using that dealt with wifi causing occasional problems with the ADC. This is all with IDF v5.1.2.

If you're in a position to give the newer driver a try I'd suggest giving that a go to see if it helps the situation.

patrickmagee, 2024-01-03:
Many thanks to @CaptivatingCat for the fix.
I was able to get it working but only by modding the espidf files directly. Obviously not a long term solution. I've been trying in vain to get the wrapper correctly implemented.

I get an error :

> 
> Successfully created esp32s3 image.
> Generated C:/Projects/redact/NM52-Device/build/bootloader/bootloader.bin
> [112/112] cmd.exe /C "cd /D C:\Projects\redacr\NM52-Device\build\bootloader\esp-idf\esptool_py && c:\esp\to...eck_sizes.py --offset 0x8000 bootloader 0x0 C:/Projects/redact/NM52-Device/build/bootloader/bootloader.bin" 
> Bootloader binary size 0x5000 bytes. 0x3000 bytes (38%) free.
> [1418/1420] Linking CXX executable NM52-Device.elf
> FAILED: NM52-Device.elf
> cmd.exe /C "cd . && C:\esp\tools\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Projects/redeact/NM52-Device/build/NM52-Device.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld -T ulp_main.ld @CMakeFiles\NM52-Device.elf.rsp -o NM52-Device.elf  && cd ."
> C:/esp/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj):(.iram1.18.literal+0x34): undefined reference to `__wrap_rtc_sleep_init'
> C:/esp/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj): in function `esp_sleep_start':
> C:/esp/esp-idf/components/esp_hw_support/sleep_modes.c:770:(.iram1.18+0x23a): undefined reference to `__wrap_rtc_sleep_init'
> collect2.exe: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1, output of the command is in the C:\Projects\redact\NM52-Device\build\log\idf_py_stderr_output_4004 and C:\Projects\redact\NM52-Device\build\log\idf_py_stdout_output_4004
> PS C:\Projects\redact\NM52-Device>

I've modded the cmakelists in the main folder, perhaps this is incorrect?

```
idf_component_register(SRCS "NM52-Device_main.cpp"
"ULP_Init.cpp"
    INCLUDE_DIRS "."
    REQUIRES soc
    ulp
    esp_adc
    rtc_sleep_fix    <----- wrapper c and h here
)

target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=rtc_sleep_init")

#
# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_${COMPONENT_NAME})

#
# 2. Specify all C and Assembly source files.
# Files should be placed into a separate directory (in this case, ulp/),
# which should not be added to COMPONENT_SRCS.
set(ulp_riscv_sources "ulp/main.c")

#
# 3. List all the component source files which include automatically
# generated ULP export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "NM52-Device_main.cpp" "ULP_Init.cpp")

#
# 4. Call function to build ULP binary and embed in project using the argument
# values above.
ulp_embed_binary(${ulp_app_name} "${ulp_riscv_sources}" "${ulp_exp_dep_srcs}")
```

Can anyone share where to put this command?

patrickmagee, 2024-01-03:
There are quite a few reports of this bug, for which espressif have not yet implemented a fix. To assist, I'll share my experience

Chip details 

```
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
Crystal is 40MHz
```

ESPIDF versions 5.2.0 and v5.2-beta1-263-ge49823f10c both exhit the bug.

Expected behavior:
ulp_riscv_adc_read_channel  works as it does in both deep sleep and with the main processor running.

Actual behavior
ulp_riscv_adc_read_channel returns the same value during all of light sleep, the value appears to be the value that was read when first entering light sleep.

Other similar issues mentioned here have other fixes that did not work for me
- Removing compiler optimization (no effect)
- Use oneshot adc (this was already the case)
- RTC Clock Source (Using external 32KHz Crystal, changing to internal 136KHz or 17.5MHz has no effect)
- Roll back to 5.0.1 (not possible, as it has other ADC bugs related to deep sleep, which we also use on this project)

This is a pretty nasty bug, is there a timeline for a proper espidf fix?

patrickmagee, 2024-01-08:
@espressif-bot could we get an update on this rather glaring bug ?

closed 2025-04-18 by espressif-bot