espressif/esp-idf #12427: DHCP server enabled on ETH even though started on AP

ClaesIvarsson, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6

### Espressif SoC revision.

ESP32-D0WD-V3 (revision 3)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I am using a modified software based on ip_internal_network example.
I have a dedicated root node with Ethernet connection.

When I start up the root node as normal everything is working normally, nodes in the mesh network gets an IP address and can communicate with external resources.

I disconnect the ethernet connection and reconnect again after a few seconds.

Communication continues as normal on both mesh network and on local network

### What is the actual behavior?

After the disconnection/re-connection of ethernet cable, the root node start to act as DCHP server on the ethernet interface.
Confirmed by debug output from root with matching DHCP server lease for a DHCP request on a different computer on LAN.

I have confirmed that this behavior happens only after a re connection of the ethernet (link down/link up/got ip) and DHCP server is not present on ethernet interface directly after boot.

### Steps to reproduce.

1. start all nodes in the network
2. disconnect network cable from ethernet port
3. wait 10s
4. connect network cable on ethernet port
5. trigger new dhcp request on another computer on LAN
6. computer gets a new IP from root node

### Debug Logs.

```plain
I (555941) [ethernet, 81]: Ethernet Link Down
I (555945) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:0
I (555948) [bridge_main, 113]: MESH_EVENT_TODS_STATE toDS=0
I (561941) [ethernet, 63]: Ethernet Link Up
I (561942) [ethernet, 65]: Ethernet HW Addr 40:f5:20:49:e8:5b
I (562944) [ethernet, 132]: <IP_EVENT_ETH_GOT_IP>IP:192.168.44.60
I (562945) [mesh_netif, 445]: Already wifi station, no need to do anything
I (562946) [ethernet, 109]: Ethernet Got IP Address
I (562946) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:1
I (562949) [ethernet, 110]: ETHIP:   192.168.44.60
I (562954) [ethernet, 111]: ETHMASK: 255.255.255.0
I (562957) [ethernet, 112]: ETHGW:   192.168.44.1
I (562995) esp_netif_handlers: eth ip: 192.168.44.60, mask: 255.255.255.0, gw: 192.168.44.1
I (562996) [bridge_main, 107]: Root obtains the IP address. It is posted by LwIP stack automatically
I (562999) [bridge_main, 113]: MESH_EVENT_TODS_STATE toDS=1
I (568673) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.10
```

ClaesIvarsson, 2023-10-18:
```
/** Event handler for Ethernet events */
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *reinterpret_cast<esp_eth_handle_t *>(event_data);

    switch (event_id) {
        case ETHERNET_EVENT_CONNECTED:
            esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
            P_LOGI("Ethernet Link Up");
            P_LOGI("Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                     mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
            s_ethernet_is_connected = true;
            break;

        case ETHERNET_EVENT_DISCONNECTED:
            P_LOGI("Ethernet Link Down");
            ip_status.connected = false;
            ip_status.ip_address[0] = '\0';
            ip_status.ip_mask[0] = '\0';
            ip_status.gateway[0] = '\0';
            s_ethernet_is_connected = false;
            s_got_ip = false;
            mesh_netifs_stop();
            esp_mesh_post_toDS_state(false);
            break;

        case ETHERNET_EVENT_START:
            P_LOGI("Ethernet Started");
            break;

        case ETHERNET_EVENT_STOP:
            P_LOGI("Ethernet Stopped");
            break;

        default:
            break;
    }
}

/** Event handler for IP_EVENT_ETH_GOT_IP */
static void ip_event_handler(void *arg, esp_event_base_t event_base,
                             int32_t event_id, void *event_data)
{
    switch (event_id) {
        case IP_EVENT_ETH_LOST_IP: {
            mesh_netifs_stop();
            esp_mesh_post_toDS_state(false);
        }
            break;
        case IP_EVENT_ETH_GOT_IP: {
            ip_event_got_ip_t *event = reinterpret_cast<ip_event_got_ip_t *>(event_data);
            P_LOGI("<IP_EVENT_ETH_GOT_IP>IP:" IPSTR, IP2STR(&event->ip_info.ip));
            s_got_ip = true;

            mesh_netifs_start(true);
            esp_mesh_post_toDS_state(true);
            got_ip_address(event_data);

            s_current_ip.addr = event->ip_info.ip.addr;
#if !CONFIG_MESH_USE_GLOBAL_DNS_IP
            esp_netif_t *netif = event->esp_netif;
            ESP_ERROR_CHECK(esp_netif_get_dns_info(netif, ESP_NETIF_DNS_MAIN, &s_dns));
            mesh_netif_start_root_ap(esp_mesh_is_root(), s_dns.ip.u_addr.ip4.addr);
#endif
            mdf_event_loop_send(MDF_EVENT_MWIFI_ROOT_GOT_IP, NULL);
            fct_socket_got_ip_event();
            break;
        }
        default:
            P_LOGI("ip_event_handler: %d (unhandled)", event_id);
            break;
    }
}

```

zhangyanjiaoesp, 2023-10-27:
I noticed that you called the function of `mdf_event_loop_send(MDF_EVENT_MWIFI_ROOT_GOT_IP, NULL);`, are you using [esp-mdf](https://github.com/espressif/esp-mdf) repository code ? This repository only support up to IDF v4.3, and we no longer maintain this repository, recommend to use https://github.com/espressif/esp-mesh-lite.

ClaesIvarsson, 2023-10-27:
I used to use the esp-mdf repo, but I no longer do that, but there is still some traces left in my code.

zhangyanjiaoesp, 2023-10-30:
Have you set the `nonmesh_max_connection` to non-zero?

ClaesIvarsson, 2023-10-30:
> Have you set the `nonmesh_max_connection` to non-zero?

No, cfg.mesh_ap.nonmesh_max_connection = 0

xueyunfei998, 2023-10-31:
hi @ClaesIvarsson 

Is the esp32 Ethernet port connected to the router?

xueyunfei998, 2023-10-31:
hi @ClaesIvarsson 
Your scenario is that the esp32 Ethernet port is connected to the router at first, and then the esp32 Ethernet port is connected to the pc after disconnection?
Can you describe the scene?

xueyunfei998, 2023-10-31:
hi @ClaesIvarsson 
Can you post your Ethernet initialization code?

ClaesIvarsson, 2023-10-31:
```
static void eth_event_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *reinterpret_cast<esp_eth_handle_t *>(event_data);

    switch (event_id) {
        case ETHERNET_EVENT_CONNECTED:
            esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
            P_LOGI("Ethernet Link Up");
            P_LOGI("Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                     mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
            s_ethernet_is_connected = true;
            break;

        case ETHERNET_EVENT_DISCONNECTED:
            P_LOGI("Ethernet Link Down");
            ip_status.connected = false;
            ip_status.ip_address[0] = '\0';
            ip_status.ip_mask[0] = '\0';
            ip_status.gateway[0] = '\0';
            s_ethernet_is_connected = false;
            s_got_ip = false;
            mesh_netifs_stop();
            esp_mesh_post_toDS_state(false);
            break;

        case ETHERNET_EVENT_START:
            P_LOGI("Ethernet Started");
            break;

        case ETHERNET_EVENT_STOP:
            P_LOGI("Ethernet Stopped");
            break;

        default:
            break;
    }
}

static void ip_event_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    switch (event_id) {
        case IP_EVENT_ETH_LOST_IP: {
            mesh_netifs_stop();
            esp_mesh_post_toDS_state(false);
        }
            break;
        case IP_EVENT_ETH_GOT_IP: {
            ip_event_got_ip_t *event = reinterpret_cast<ip_event_got_ip_t *>(event_data);
            P_LOGI("<IP_EVENT_ETH_GOT_IP>IP:" IPSTR, IP2STR(&event->ip_info.ip));
            s_got_ip = true;

            mesh_netifs_start(true);
            esp_mesh_post_toDS_state(true);
            got_ip_address(event_data);

            s_current_ip.addr = event->ip_info.ip.addr;
            esp_netif_t *netif = event->esp_netif;
            ESP_ERROR_CHECK(esp_netif_get_dns_info(netif, ESP_NETIF_DNS_MAIN, &s_dns));
            mesh_netif_start_root_ap(esp_mesh_is_root(), s_dns.ip.u_addr.ip4.addr);
            mdf_event_loop_send(MDF_EVENT_MWIFI_ROOT_GOT_IP, NULL);
            fct_socket_got_ip_event();
            break;
        }
        case IP_EVENT_AP_STAIPASSIGNED: {
            // ip_event_ap_staipassigned_t *event = (ip_event_ap_staipassigned_t *) event_data;
            // P_LOGI("<IP_EVENT_AP_STAIPASSIGNED>IP:" IPSTR, IP2STR(&event->ip));
            break;
        }
        default:
            P_LOGI("ip_event_handler: %d (unhandled)", event_id);
            break;
    }
}

esp_err_t eth_init()
{
    P_LOGI("eth_init()");

    esp_netif_config_t eth_cfg = ESP_NETIF_DEFAULT_ETH();
    netif_eth = esp_netif_new(&eth_cfg);
    // ESP_ERROR_CHECK(esp_netif_set_default_netif(netif_eth));

    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &ip_event_handler, NULL));

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();

    const BoardEthSetup_t* init_data = Hal_GetEthInitData();

    phy_config.phy_addr = init_data->phy_addr;                      // 1
    phy_config.reset_gpio_num = init_data->reset_gpio_num;          // CONFIG_ETH_PHY_RST_GPIO = 5
    mac_config.smi_mdc_gpio_num = init_data->smi_mdc_gpio_num;      // CONFIG_ETH_MDC_GPIO = 23
    mac_config.smi_mdio_gpio_num = init_data->smi_mdio_gpio_num;    // CONFIG_ETH_MDIO_GPIO = 18
    esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
    esp_eth_phy_t *phy = NULL;

    phy = init_data->phy_init_func(&phy_config);                    // esp_eth_phy_new_dp83848
    if(phy == NULL) {
        P_LOGE("Failed to start ethernet phy, is this a fake root node ?");
        return ESP_FAIL;
    }

    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
    s_eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &s_eth_handle));
    ESP_ERROR_CHECK(esp_netif_attach(netif_eth, esp_eth_new_netif_glue(s_eth_handle)));
    ESP_ERROR_CHECK(esp_eth_start(s_eth_handle));

    if (Hal_GetBoardId() == HWB_BRIDGE_REV_C) {
        /* hardware pullup is too week to force RMII mode,
        * hence writing to the register to ensure correct mode */
        mac->write_phy_reg(mac, phy_config.phy_addr, 0x17, 0x21);
    }

    P_LOGI("eth_init() completed ok");
    return ESP_OK;
}
```

ClaesIvarsson, 2023-10-31:
> hi @ClaesIvarsson Your scenario is that the esp32 Ethernet port is connected to the router at first, and then the esp32 Ethernet port is connected to the pc after disconnection? Can you describe the scene?

my root node is connected to the router using ethernet, router assign IP address fo ethernet
root assign IP addresses to all nodes in the mesh network.

When I remove the ethernet cable, the root looses ip/Link
When attaching ethernet cable again, the root gets IP from router but starts acting as DHCP server on LAN.
Start a computer attached to the same LAN will cause the laptop to get IP address from the root node instead of the router.
Laptop get IP from the range used in mesh network, I can also see in the root log that IP is given by the root.

If I restart the root again and let the Laptop renew its address on LAN, I can see that the IP is given from the router (as expected).

This happens every time I repeat the procedure.

xueyunfei998, 2023-11-03:
HI  @ClaesIvarsson 
Can you provide a reproduction code of demo?

Sherry616, 2024-04-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-23 by Sherry616