espressif/esp-idf #12429: cmake error

epicdenver1, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

i have a fresh install of 5.2-dev and this code has worked before on a windows based visual studio machine. i am using a tinly ml conversion of a TensorFlow lite model to try and get it to work on my esp32-s3.
bellow is my cmake file
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/13030203/CMakeLists.txt)
i am aware that in 5.0 updating to 5.1 they changed the way idf_component_register changed but this was not an issue on 5.11 on windows. the other files such as the project file and model will be attached bellow. i have been working on this for 8 hours and have been racking my brain; any expertise even if it makes me look dump is appreciated.
[app_main.txt](https://github.com/espressif/esp-idf/files/13030257/app_main.txt)
[handrecognition_coefficient.txt](https://github.com/espressif/esp-idf/files/13030263/handrecognition_coefficient.txt)
i converted the c++ format of the above two files

thanks in advance -ryan

0xjakob, 2023-10-19:
Hi @epicdenver1, would you mind sharing the actual error message? "cmake error" can mean a lot to us, we need more details to trace down the error.

Thank you!

epicdenver1, 2023-10-19:
this is the error, i get the same one with both idf v 4.4.4 , 5.2 dev , and 5.1.1
/home/ryan/esp-idf-v4.4.4/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
CMake Error at CMakeLists.txt:13 (register_component):
  Unknown CMake command "register_component".

-- Configuring incomplete, errors occurred!
See also "/home/ryan/model_deployment/main/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
/usr/bin/cmake --regenerate-during-build -S/home/ryan/model_deployment/main -B/home/ryan/model_deployment/main/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
i have ninja installed and double checked before i made this post

igrr, 2023-10-19:
Looks like you are trying to run `idf.py build` in the `main` subdirectory, not in the project directory?

epicdenver1, 2023-10-19:
[idf_py_stdout_output_1924.txt](https://github.com/espressif/esp-idf/files/13042798/idf_py_stdout_output_1924.txt)
i grabbed the wrong file my bad

igrr, 2023-10-19:
Over here:
```cmake
if(${IDF_TARGET} STREQUAL "esp32")
    set(links "-L ${CMAKE_CURRENT_SOURCE_DIR}/../../../lib/esp32")
elseif(${IDF_TARGET} STREQUAL "esp32s2")
    set(links "-L ${CMAKE_CURRENT_SOURCE_DIR}/../../../lib/esp32s2")
elseif(${IDF_TARGET} STREQUAL "esp32s3")
    set(links "-L ${CMAKE_CURRENT_SOURCE_DIR}/../../../lib/esp32s3")
elseif(${IDF_TARGET} STREQUAL "esp32c3")
    set(links "-L ${CMAKE_CURRENT_SOURCE_DIR}/../../../lib/esp32c3")
endif()

target_link_libraries(${COMPONENT_TARGET} ${links})

```
you are only specifying the directory where to search for the pre-compiled libraries, but you aren't actually linking any specific library. Hence,
```
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.cpp.obj):(.literal._ZN2dl6TensorIsE12copy_elementERS1_b[_ZN2dl6TensorIsE12copy_elementERS1_b]+0xc): undefined reference to `dl::tool::copy_memory(void*, void*, int)'

c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.cpp.obj):(.literal._ZN2dl5layer7SoftmaxIsfLi1ELb0EED5Ev[_ZN2dl5layer7SoftmaxIsfLi1ELb0EED5Ev]+0x0): undefined reference to `dl::layer::Layer::~Layer()'
```
some of the references from `main` component to the pre-compiled library don't get resolved.

epicdenver1, 2023-10-19:
thanks for that, how do i fix that, those are from the guide i was using to develop this model but if you have an idea i'm willing to give it a shot. thanks

igrr, 2023-10-19:
Could you please link to the guide you are reading?

igrr, 2023-10-19:
Seems like you are missing this line:
https://github.com/alibukharai/Blogs/blob/a39db60cf8080797190349147231c534944c860b/ESP-DL/model_deployment/main/CMakeLists.txt#L11
and this line:
https://github.com/alibukharai/Blogs/blob/a39db60cf8080797190349147231c534944c860b/ESP-DL/model_deployment/main/CMakeLists.txt#L29

epicdenver1, 2023-10-19:
i made the changes, still get deploy.elf issues

 Build files have been written to: C:/Espressif/Blogs/ESP-DL/model_deployment/build
[1/1] cmd.exe /C "cd /D C:\Espressif\Blogs\ESP-DL\model_de...s/ESP-DL/model_deployment/build/bootloader/bootloader.bin"Bootloader binary size 0x56d0 bytes. 0x2930 bytes (32%) free.
[2/4] Linking CXX executable deploy.elfFAILED: deploy.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Espressif/Blogs/ESP-DL/model_deployment/build/deploy.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\deploy.elf.rsp -o deploy.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -ldl: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Espressif\Blogs\ESP-DL\model_deployment\build\log\idf_py_stderr_output_5320 and C:\Espressif\Blogs\ESP-DL\model_deployment\build\log\idf_py_stdout_output_5320
PS C:\Espressif\Blogs\ESP-DL\model_deployment>

igrr, 2023-10-19:
Can you check if there is `libdl.a` in the directory you have specified in CMakeLists.txt, i.e. `<project_dir>\main\..\..\..\lib\esp32s3\libdl.a`?

epicdenver1, 2023-10-19:
if i had that library in my /main or in the directory how would i edit the cmake file

0xjakob, 2023-10-24:
@epicdenver1 I checked with the creator of the tutorial. Unfortunately, the example is only made for IDF v4.4. It won't work out-of-the-box on IDF v5.2.