espressif/esp-idf #12430: Ethernet Iperf watchdog issue

Sotung96, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, I use the example to test the iperf performance on my project but I enable the task watchdog. When doing the iperf performance test. it shows up 
```
E (220792) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (220792) task_wdt:  - IDLE (CPU 0)
E (220792) task_wdt: Tasks currently running:
E (220792) task_wdt: CPU 0: w5500_tsk
E (220792) task_wdt: Print CPU 0 (current core) backtrace
``` 
Does anyone any how to fix the problem?

KaeLL, 2023-10-18:
Likely the task reading data from the chip doesn't yield in anticipation of data being available at every check for a time longer than the watchdog window.

0xjakob, 2023-10-19:
@Sotung96 Do you mean the [IDF Ethernet Iperf example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/iperf) with "Ethernet Iperf"? I'm asking because the task watchdog is disabled in that example. Maybe it has been enabled inadvertently.

Sotung96, 2023-10-19:
@0xjakob Yes, I know that is disable in the example. Is it mean I can't enable the task watch dog when I doing the iperf test? Also, how can I know what is cpu idle task doing?

0xjakob, 2023-10-19:
@Sotung96 It seems like enabling the task watchdog with this iperf implementation is not possible. But since it is used for testing, it should not be a big deal. The assumption here is that someone is attending the device when running the test. 

When you have applications on devices in the field, however, you should always enable the watchdogs to prevent a device from getting stuck.

If you want to see what certain tasks are doing on the CPU, you could try to enable the app trace library and [use SystemView to analyze the behavior](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html?highlight=sysview#system-behavior-analysis-with-segger-systemview).

Sotung96, 2023-10-19:
@0xjakob I got some error when I enable the `CONFIG_APPTRACE_SV_ENABLE`
```c
C:/Espressif/frameworks/esp-idf/components/app_trace/sys_view/esp/SEGGER_RTT_esp.c: In function 'SEGGER_RTT_WriteSkipNoLock':
C:/Espressif/frameworks/esp-idf/components/app_trace/sys_view/esp/SEGGER_RTT_esp.c:170:20: error: 'SYSVIEW_EVTID_USER_START' undeclared (first use in this function); did you mean 'SYSVIEW_EVTID_MARK_START'?
  170 |       (event_id == SYSVIEW_EVTID_USER_START) ||
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
      |                    SYSVIEW_EVTID_MARK_START
C:/Espressif/frameworks/esp-idf/components/app_trace/sys_view/esp/SEGGER_RTT_esp.c:170:20: note: each undeclared identifier is reported only once for each function it appears in
C:/Espressif/frameworks/esp-idf/components/app_trace/sys_view/esp/SEGGER_RTT_esp.c:171:20: error: 'SYSVIEW_EVTID_USER_STOP' undeclared (first use in this function); did you mean 'SYSVIEW_EVTID_MARK_STOP'?
  171 |       (event_id == SYSVIEW_EVTID_USER_STOP) ||
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
      |                    SYSVIEW_EVTID_MARK_STOP
``` 
 Then, I checked there are a bug in `SEGGER_RTT_esp.c` 
```c
unsigned SEGGER_RTT_WriteSkipNoLock(unsigned BufferIndex, const void* pBuffer, unsigned NumBytes) {
  uint8_t *pbuf = (uint8_t *)pBuffer;
  uint8_t event_id = *pbuf;
#if CONFIG_APPTRACE_SV_DEST_UART
  if (
    (APPTRACE_SV_DEST_CPU != esp_cpu_get_core_id()) &&
    (
      (event_id == SYSVIEW_EVTID_ISR_ENTER) ||
      (event_id == SYSVIEW_EVTID_ISR_EXIT) ||
      (event_id == SYSVIEW_EVTID_TASK_START_EXEC) ||
      (event_id == SYSVIEW_EVTID_TASK_STOP_EXEC) ||
      (event_id == SYSVIEW_EVTID_TASK_START_READY) ||
      (event_id == SYSVIEW_EVTID_TASK_STOP_READY) ||
      (event_id == SYSVIEW_EVTID_USER_START) ||
      (event_id == SYSVIEW_EVTID_USER_STOP) ||
      (event_id == SYSVIEW_EVTID_TIMER_ENTER) ||
      (event_id == SYSVIEW_EVTID_TIMER_EXIT) ||
      (event_id == SYSVIEW_EVTID_STACK_INFO) ||
      (event_id == SYSVIEW_EVTID_MODULEDESC)
    )
  ){
    return NumBytes;
  }
``` 
Do you know which one should replace the `SYSVIEW_EVTID_USER_START` and `SYSVIEW_EVTID_USER_STOP`?

0xjakob, 2023-10-20:
@Sotung96 I was able to build and run the iperf application. I used the plain example application and configured it, according following the READMEs and the sysview documentation page (which I posted above). I'm using IDF master branch.

Could you try the following configuration file and see if it works?
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13050665/sdkconfig.zip)

Sotung96, 2023-10-23:
@0xjakob I checked the file. It haven't enable task watchdog. I have try disable task watchdog in my config file. It is fine but I want to enable the task watchdog when doing iperf test.

Sotung96, 2023-10-24:
@0xjakob Yes, the iperf test without watchdog is fine. Do you know how to use USB CDC to do  Application Level Tracing? I have enable it and select USB_CDC in Data Destination 2 but it show me  `E (3099) esp_apptrace: Trace destination 2 not supported!` when I'm running the code.

Sotung96, 2023-11-02:
@0xjakob Will the SPI3 be efficed by the cpu speed? because it will be fine when i chagne the cpu speed to 240MHz. The watchdog got triggered is happened when running spi device pollingh transmit.
```c
E (1998974) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (1998974) task_wdt:  - IDLE (CPU 0)
E (1998974) task_wdt: Tasks currently running:
E (1998974) task_wdt: CPU 0: tiT
E (1998974) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420142DA:0x3FC98B90 0x42014456:0x3FC98BB0 0x40376F92:0x3FC98BD0 0x4037DB17:0x3FCA30A0 0x40379EF5:0x3FCA30C0 0x40379F5D:0x3FCA30E0 0x4202607D:0x3FCA3100 0x4202611F:0x3FCA3160 0x42026DB8:0x3FCA3190 0x42025246:0x3FCA31D0 0x420448BF:0x3FCA3200 0x4037B481:0x3FCA3220 0x4038365D:0x3FCA3240 0x403834F6:0x3FCA3260 0x403833C1:0x3FCA3280 0x40383C95:0x3FCA32B0 0x40383A21:0x3FCA32F0 0x403839E9:0x3FCA3320 0x4201CD29:0x3FCA3350 0x4201D921:0x3FCA3380 0x4201C715:0x3FCA33B0 0x40383955:0x3FCA33E0 0x403835AE:0x3FCA3400 0x403851A5:0x3FCA3420 0x40385154:0x3FCA3440 0x4037EC39:0x3FCA3470
0x420142da: task_wdt_timeout_handling at C:/espressif/esp-idf/components/esp_system/task_wdt/task_wdt.c:461 (discriminator 3)

0x42014456: task_wdt_isr at C:/espressif/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40376f92: _xt_lowint1 at C:/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4037db17: xTaskGetTickCount at C:/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:2701

0x40379ef5: spi_device_polling_end at C:/espressif/esp-idf/components/driver/spi/gpspi/spi_master.c:1083

0x40379f5d: spi_device_polling_transmit at C:/espressif/esp-idf/components/driver/spi/gpspi/spi_master.c:1113

0x4202607d: w5500_write at C:/espressif/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:74

0x4202611f: w5500_write_buffer at C:/espressif/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:174

0x42026db8: emac_w5500_transmit at C:/espressif/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:490 (discriminator 2)

0x42025246: esp_eth_transmit at C:/espressif/esp-idf/components/esp_eth/src/esp_eth.c:362 (discriminator 2)

0x420448bf: esp_netif_transmit at C:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1211

0x4037b481: ethernet_low_level_output at C:/espressif/esp-idf/components/esp_netif/lwip/netif/ethernetif.c:84

0x4038365d: ethernet_output at C:/espressif/esp-idf/components/lwip/lwip/src/netif/ethernet.c:312 (discriminator 2)

0x403834f6: etharp_output_to_arp_index at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:769

0x403833c1: etharp_output at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:868

0x40383c95: ip4_output_if_opt_src at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:1054

0x40383a21: ip4_output_if_opt at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:855

0x403839e9: ip4_output_if at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:832

0x4201cd29: tcp_output_segment at C:/espressif/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1608 (discriminator 4)

0x4201d921: tcp_output at C:/espressif/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1361

0x4201c715: tcp_input at C:/espressif/esp-idf/components/lwip/lwip/src/core/tcp_in.c:566

0x40383955: ip4_input at C:/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:749

0x403835ae: ethernet_input at C:/espressif/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186

0x403851a5: tcpip_thread_handle_msg at C:/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:174

0x40385154: tcpip_thread at C:/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

0x4037ec39: vPortTaskWrapper at C:/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

closed 2024-01-11 by Sotung96