espressif/esp-idf #12431: WIFI-MESH: Node lost connection after Root reboot

yanmartins, 2023-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I'm working in a project using ESP32-S3.
I run the [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example with this init:

```C
    //! initialize the wifi stack in "mode" with config in RAM
    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_LOGI(TAG, "srv_wifi_init [WIFI_INIT_CONFIG_DEFAULT]");

    ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));
    ESP_LOGI(TAG, "srv_wifi_init [esp_wifi_init]");

    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_LOGI(TAG, "srv_wifi_init [esp_wifi_set_storage]");

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_LOGI(TAG, "srv_wifi_init [esp_wifi_set_mode]");

    //! Create a network interfaces for mesh (only station instance saved for further manipulation, soft AP instance ignored
    ESP_ERROR_CHECK(mesh_netifs_init(recv_cb));

    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_sta_event_handler, NULL));

    ESP_ERROR_CHECK(esp_wifi_start());
    /*  mesh initialization */
    ESP_ERROR_CHECK(esp_mesh_init());

    ESP_ERROR_CHECK(esp_mesh_set_xon_qsize(128));
    ESP_ERROR_CHECK(esp_mesh_allow_root_conflicts(false));

    esp_mesh_fix_root(true);
    if(wifi_mesh_config->is_root) esp_mesh_set_type(MESH_ROOT);
    
    ESP_ERROR_CHECK(esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL));
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(6));

    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
    /* mesh ID */
    memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, 6);

    /* router */
    cfg.channel         = 0;
    cfg.router.ssid_len = strlen("xx");
    memcpy((uint8_t *) &cfg.router.ssid, "xx", cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, "yy", strlen("yy"));
    /* mesh softAP */
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(WIFI_AUTH_WPA2_PSK));
    cfg.mesh_ap.max_connection         = 6;
    cfg.mesh_ap.nonmesh_max_connection = 0;
    memcpy((uint8_t *) &cfg.mesh_ap.password, "MAP_PASSWD", strlen("MAP_PASSWD"));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));

    /* mesh start */
    ESP_ERROR_CHECK(esp_mesh_start());
    ESP_LOGI(TAG, "mesh starts successfully, heap: %lu, %s\n", esp_get_free_heap_size(), esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed");
```

Following is the log output of the program on a NODE.

ROOT is turned off, after the MESH_EVENT_PARENT_CONNECTED event on the NODE. But when ROOT is turned on again, NODE never gets a connection. Remaining infinitely in this loop. What could be causing this?

[... 37 lines cut ...]
[2023-10-18 15:04:47.596] I (6052908) app_main: Heap size -> 8217500
[2023-10-18 15:04:47.660] W (6052958) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:7
[2023-10-18 15:04:48.593] I (6053908) app_main: Heap size -> 8217500
[2023-10-18 15:04:49.458] W (6054758) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:8
[2023-10-18 15:04:49.600] I (6054908) app_main: Heap size -> 8217500
[2023-10-18 15:04:50.603] I (6055908) app_main: Heap size -> 8217500
[2023-10-18 15:04:51.260] W (6056558) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:9
[2023-10-18 15:04:51.606] I (6056908) app_main: Heap size -> 8217500
[2023-10-18 15:04:52.597] I (6057908) app_main: Heap size -> 8217500
[2023-10-18 15:04:53.067] W (6058358) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:10
[2023-10-18 15:04:53.590] I (6058908) app_main: Heap size -> 8217500
[2023-10-18 15:04:54.601] I (6059908) app_main: Heap size -> 8217500
[2023-10-18 15:04:54.871] W (6060158) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:11
[2023-10-18 15:04:55.604] I (6060908) app_main: Heap size -> 8217500
[2023-10-18 15:04:56.598] I (6061908) app_main: Heap size -> 8217500
[2023-10-18 15:04:56.664] W (6061958) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:12
[2023-10-18 15:04:57.601] I (6062908) app_main: Heap size -> 8217500
[2023-10-18 15:04:58.465] W (6063758) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:13
[2023-10-18 15:04:58.597] I (6063908) app_main: Heap size -> 8217500
[2023-10-18 15:04:59.603] I (6064908) app_main: Heap size -> 8217500
[2023-10-18 15:05:00.262] W (6065558) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:14
[2023-10-18 15:05:00.606] I (6065908) app_main: Heap size -> 8217500
[2023-10-18 15:05:01.610] I (6066908) app_main: Heap size -> 8217500
[2023-10-18 15:05:02.063] W (6067358) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:1492, no_wnd_count:0, timeout_count:15
```

zhangyanjiaoesp, 2023-10-24:
@yanmartins 
Can you update your IDF branch to 2048d1e0c0e3d33926a75160bb8d4db8102c4b2b to see whether the issue has been resolved?

yanmartins, 2023-10-24:
> > I (940266) coexist: [940251263] Error! Should enable WiFi modem sleep when both WiFi and Bluetooth
> 
> The log means you should open power save when using coexist.

This setting was active. But I had to disable sleep via software

> @yanmartins Can you update your IDF branch to [2048d1e](https://github.com/espressif/esp-idf/commit/2048d1e0c0e3d33926a75160bb8d4db8102c4b2b) to see whether the issue has been resolved?

Thanks, I'll check. Is this commit part of a release?

zhangyanjiaoesp, 2023-10-26:
> I noticed that, in the example, `mesh_netifs_destroy` function is never called. This is necessary to delete the RX task, right?

yes, If you want to delete the RX task, you can call `mesh_netif_destroy`.

yanmartins, 2023-10-26:
@zhangyanjiaoesp in my tests, the [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example create the RX task in the NODE after each ROOT reboot. Which generated a memory leak that took a while to resolve.

It might be interesting to update this example to avoid this type of problem.

yanmartins, 2023-10-26:
> @yanmartins Can you update your IDF branch to [2048d1e](https://github.com/espressif/esp-idf/commit/2048d1e0c0e3d33926a75160bb8d4db8102c4b2b) to see whether the issue has been resolved?

And yes, updating to version `5.0.3` apparently fixed the reconnection issues

zhangyanjiaoesp, 2023-10-26:
> @zhangyanjiaoesp in my tests, the [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example create the RX task in the NODE after each ROOT [...]

thanks, we will check it.

zhangyanjiaoesp, 2023-10-27:
@yanmartins 
For the NODE device, when there is `MESH_EVENT_PARENT_CONNECTED` event, it will call `mesh_netifs_start() -> mesh_create_if_driver() ` , the `mesh_create_if_driver()` will create the RX task at the first time, then the `receive_task_is_running = true`, this will ensure the RX task won't be created again. 

I'm not clear about your memory leak issue, can you describe it in more details?

yanmartins, 2023-10-27:
Thanks, @zhangyanjiaoesp
I found my problem. I added a `receive_task_is_running = false` in `mesh_delete_if_driver` and forgot to remove it.
Sorry

zhangyanjiaoesp, 2023-10-30:
If your problem has been solved, please close this ticket, thanks.

yanmartins, 2023-10-31:
@zhangyanjiaoesp after modifying the [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example to use a fixed root using PPP data. The same error came back:

```
W (61109650) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:840
I (61110390) app_main: Heap size -> 8222364
I (61111390) app_main: Heap size -> 8222364
W (61111450) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:841
I (61112390) app_main: Heap size -> 8222364
W (61113250) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:842
I (61113390) app_main: Heap size -> 8222364
I (61114390) app_main: Heap size -> 8222364
W (61115050) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:843
I (61115390) app_main: Heap size -> 8222364
I (61116390) app_main: Heap size -> 8222364
W (61116850) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:844
I (61117390) app_main: Heap size -> 8222364
I (61118390) app_main: Heap size -> 8222364
W (61118650) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:845
I (61119390) app_main: Heap size -> 8222364
I (61120390) app_main: Heap size -> 8222364
W (61120450) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:11188, no_wnd_count:0, timeout_count:846
```

I just comment this configs:

```c
/* router */
cfg.channel         = 0;
cfg.router.ssid_len = strlen((char *) wifi_params->ssid);
memcpy((uint8_t *) &cfg.router.ssid, wifi_params->ssid, cfg.router.ssid_len);
memcpy((uint8_t *) &cfg.router.password, wifi_params->password, sizeof(wifi_params->password));
```

Add this in the init:
```c
esp_mesh_fix_root(true);
if(wifi_mesh_config->is_root) esp_mesh_set_type(MESH_ROOT);
```

And this in the PPP GOT IP event:
```c
if(!esp_mesh_is_root()) return false;

mesh_netifs_start(true);
esp_mesh_post_toDS_state(true);
return mesh_netif_start_root_ap(true, dns_addr);
```

zhangyanjiaoesp, 2023-11-01:
@yanmartins Can you provide a complete log of the Node when the reconnection works well?

By the way, I think there are two things you can optimize in your code:
1.  Call `esp_mesh_set_xon_qsize(128)` to enlarge the xon window .
2. The default mesh packet lifetime is BLOCKING(0xffffffff), you can call `esp_mesh_send_block_time()` to set the mesh packet lifetime, then the mesh node will not always show `W (32119) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:13`, it will return TX_TIMEOUT when the lifetime has reached.

zhangyanjiaoesp, 2023-11-01:
> I just comment this configs:
>
> ```c
> /* router */
> cfg.channel         = 0;
> cfg.router.ssid_len = strlen((char *) wifi_params->ssid);
> memcpy((uint8_t *) &cfg.router.ssid, wifi_params->ssid, [...]

How about you don't comment the router config lines?

yanmartins, 2023-11-01:
> @yanmartins Can you provide a complete log of the Node when the reconnection works well?

Yes! I am sending both cases.

No reconnection:
```
[2023-11-01 11:25:45.706] I (3667) pp: pp rom version: e7ae62f
[2023-11-01 11:25:45.706] I (3667) net80211: net80211 rom version: e7ae62f
[2023-11-01 11:25:45.722] I (3687) wifi:wifi driver task: 3fcc8a68, prio:23, stack:6656, core=0
[2023-11-01 11:25:45.722] I (3687) system_api: Base MAC address is not set
[2023-11-01 11:25:45.722] I (3687) system_api: read default base MAC address from EFUSE
[2023-11-01 11:25:45.736] I (3697) wifi:wifi firmware version: e2d0ad8
[2023-11-01 11:25:45.736] I (3697) wifi:wifi certification version: v7.0
[2023-11-01 11:25:45.736] I (3707) wifi:config NVS flash: enabled
[2023-11-01 11:25:45.751] I (3707) wifi:config nano formating: disabled
[2023-11-01 11:25:45.751] I (3707) wifi:Init data frame dynamic rx buffer num: 64
[2023-11-01 11:25:45.751] I (3717) wifi:Init management frame dynamic rx buffer num: 64
[2023-11-01 11:25:45.751] I (3717) wifi:Init management short buffer num: 32
[2023-11-01 11:25:45.767] I (3727) wifi:Init static tx buffer num: 6
[2023-11-01 11:25:45.767] I (3727) wifi:Init tx cache buffer num: 32
[2023-11-01 11:25:45.775] I (3737) wifi:Init static tx FG buffer num: 2
[2023-11-01 11:25:45.775] I (3737) wifi:Init static rx buffer size: 1600
[2023-11-01 11:25:45.775] I (3737) wifi:Init static rx buffer num: 6
[2023-11-01 11:25:45.775] I (3747) wifi:Init dynamic rx buffer num: 64
[2023-11-01 11:25:45.791] I (3747) wifi_init: rx ba win: 6
[2023-11-01 11:25:45.791] I (3757) wifi_init: tcpip mbox: 32
[2023-11-01 11:25:45.791] I (3757) wifi_init: udp mbox: 6
[2023-11-01 11:25:45.791] I (3757) wifi_init: tcp mbox: 6
[2023-11-01 11:25:45.806] I (3767) wifi_init: tcp tx win: 5744
[2023-11-01 11:25:45.806] I (3767) wifi_init: tcp rx win: 5744
[2023-11-01 11:25:45.806] I (3777) wifi_init: tcp mss: 624
[2023-11-01 11:25:45.822] I (3777) wifi_init: WiFi/LWIP prefer SPIRAM
[2023-11-01 11:25:45.822] I (3777) wifi_init: WiFi IRAM OP enabled
[2023-11-01 11:25:45.822] I (3787) wifi_init: WiFi RX IRAM OP enabled
[2023-11-01 11:25:45.836] I (3787) srv_wifi: srv_wifi_init [esp_wifi_init]
[2023-11-01 11:25:45.836] I (3797) srv_wifi: srv_wifi_init [esp_wifi_set_storage]
[2023-11-01 11:25:45.836] I (3807) srv_wifi: srv_wifi_init [esp_wifi_set_mode]
[2023-11-01 11:25:45.852] I (3807) srv_wifi: srv_wifi_init [OUT]
[2023-11-01 11:25:45.852] I (3817) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
[2023-11-01 11:25:45.907] I (3867) wifi:mode : sta (34:85:18:8f:4c:7c) + softAP (34:85:18:8f:4c:7d)
[2023-11-01 11:25:45.907] I (3867) wifi:enable tsf
[2023-11-01 11:25:45.907] I (3867) wifi:Total power save buffer number: 3
[2023-11-01 11:25:45.907] I (3867) wifi:Init max length of beacon: 752/752
[2023-11-01 11:25:45.907] I (3877) wifi:Init max length of beacon: 752/752
[2023-11-01 11:25:45.923] I (3877) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
[2023-11-01 11:25:46.937] I (4897) wifi:Total power save buffer number: 3
[2023-11-01 11:25:47.937] I (5897) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2023-11-01 11:25:47.937] I (5897) srv_wifi: wifi_config->router.ssid: 
[2023-11-01 11:25:47.937] I (5897) srv_wifi: wifi_config->router.password: 
[2023-11-01 11:25:47.953] I (5907) srv_wifi: wifi_config->mesh_ap.password: MAP_PASSWD
[2023-11-01 11:25:47.953] I (5917) srv_wifi: wifi_config->mesh_ap.max_connection: 6
[2023-11-01 11:25:47.953] I (5917) srv_wifi: wifi_config->mesh_id: 119:119:119:119:119:118
[2023-11-01 11:25:47.968] I (5927) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2023-11-01 11:25:47.968] I (5937) wifi:mode : sta (34:85:18:8f:4c:7c)
[2023-11-01 11:25:47.984] I (5937) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x2, need_scan_router:0x0, look_for_nwk_count:0
[2023-11-01 11:25:47.988] I (5947) srv_wifi: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
[2023-11-01 11:25:47.988] I (5947) srv_wifi: mesh starts successfully, heap: 8393248, root fixed
[2023-11-01 11:25:48.004] I (5967) app_comm_manager: interface init
[2023-11-01 11:25:51.209] I (9177) app_main: Heap size -> 8263788
[2023-11-01 11:25:52.190] I (10147) mesh: [FIND][ch:0]AP:16, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00][FIXED-ROOT]
[2023-11-01 11:25:52.190] I (10147) mesh: [FIND:1]fail to find a network, channel:0, cfg<channel:0, router:, 00:00:00:00:00:00>
[2023-11-01 11:25:52.190] 
[2023-11-01 11:25:52.205] I (10157) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x2, need_scan_router:0x0, look_for_nwk_count:0
[2023-11-01 11:25:52.205] I (10177) app_main: Heap size -> 8265088
[2023-11-01 11:25:53.204] I (11177) app_main: Heap size -> 8264788
[2023-11-01 11:25:54.209] I (12177) app_main: Heap size -> 8264688
[2023-11-01 11:25:55.211] I (13177) app_main: Heap size -> 8264688
[2023-11-01 11:25:56.214] I (14177) app_main: Heap size -> 8263988
[2023-11-01 11:25:56.415] I (14377) mesh: [FIND][ch:0]AP:11, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00][FIXED-ROOT]
[2023-11-01 11:25:56.415] I (14377) mesh: [FIND:2]fail to find a network, channel:0, cfg<channel:0, router:, 00:00:00:00:00:00>
[... 838 lines cut ...]
[2023-11-01 11:42:29.756] I (165877) mesh: 7370[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
[2023-11-01 11:42:29.756] I (165887) mesh: 716[monitor]no change, parent:34:85:18:8f:4d:29, rssi:-37
[... cut ...]
[2023-11-01 11:42:42.408] I (178527) mesh: [SCAN][ch:1]AP:1, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:2,i:2][00:00:00:00:00:00][FIXED-ROOT]<weak>
[2023-11-01 11:42:42.426] I (178537) mesh: 7370[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
[2023-11-01 11:42:42.426] I (178547) mesh: 716[monitor]no change, parent:34:85:18:8f:4d:29, rssi:-40
[2023-11-01 11:42:42.426] I (178557) mesh: 2004<arm>parent monitor, my layer:2(cap:6)(node), interval:329862ms, retries:3<>
...
[2023-11-01 11:43:39.231] I (235347) srv_mqtt: MQTT_EVENT_CONNECTED
```

yanmartins, 2023-11-01:
> By the way, I think there are two things you can optimize in your code:
> 
> 1. Call `esp_mesh_set_xon_qsize(128)` to enlarge the xon window .

Yes, I already set `esp_mesh_set_xon_qsize(128)`

> 2. The default mesh packet lifetime is BLOCKING(0xffffffff), you can call `esp_mesh_send_block_time()` to set the mesh packet lifetime, then the mesh node will not always show `W (32119) mesh: [...]

I will add this, but it seems that the error is something else, because the NODE stops looking for ROOT

> > I just comment this configs:
> > ```c
> > /* router */
> > cfg.channel         = 0;
> > cfg.router.ssid_len = strlen((char *) wifi_params->ssid);
> > memcpy((uint8_t *) &cfg.router.ssid, wifi_params->ssid, cfg.router.ssid_len);
> > memcpy((uint8_t *) &cfg.router.password, wifi_params->password, sizeof(wifi_params->password));
> > ```
> 
> How about you don't comment the router configure lines?

You tell me to just add some random value there?

zhangyanjiaoesp, 2023-11-02:
> you tell me to add some random value? If I do this, ROOT will be looking for this connection instead of connecting to PPP

Since you have set the root to fixed, then the root won't try to connect to router

zhangyanjiaoesp, 2023-11-02:
Is there any code difference between scenarios that can and cannot reconnect?
Or the code is same,  the node reconnection does not fail every time, it sometimes can success, and sometimes fail?

yanmartins, 2023-11-03:
> Is there any code difference between scenarios that can and cannot reconnect? (I see they are both fixed-root) Or the code is same.

The same code

> The node reconnection does not fail every time, it sometimes can success, and sometimes fail?

Yes, it sometimes can success, and sometimes fail with the same code.

yanmartins, 2023-11-03:
> > you tell me to add some random value? If I do this, ROOT will be looking for this connection instead of connecting to PPP
> 
> Since you have set the root to fixed, then the root won't try to connect to router

Root is fixed, but it continues searching for the wifi network

```
I (3602) wifi:wifi driver task: 3fcc89f4, prio:23, stack:6656, core=0
I (3602) system_api: Base MAC address is not set
I (3602) system_api: read default base MAC address from EFUSE
I (3612) wifi:wifi firmware version: e2d0ad8
I (3612) wifi:wifi certification version: v7.0
I (3622) wifi:config NVS flash: enabled
I (3622) wifi:config nano formating: disabled
I (3622) wifi:Init data frame dynamic rx buffer num: 64
I (3632) wifi:Init management frame dynamic rx buffer num: 64
I (3632) wifi:Init management short buffer num: 32
I (3642) wifi:Init static tx buffer num: 6
I (3642) wifi:Init tx cache buffer num: 32
I (3652) wifi:Init static tx FG buffer num: 2
I (3652) wifi:Init static rx buffer size: 1600
I (3652) wifi:Init static rx buffer num: 6
I (3662) wifi:Init dynamic rx buffer num: 64
I (3662) wifi_init: rx ba win: 6
I (3672) wifi_init: tcpip mbox: 32
I (3672) wifi_init: udp mbox: 6
I (3672) wifi_init: tcp mbox: 6
I (3682) wifi_init: tcp tx win: 5744
I (3682) wifi_init: tcp rx win: 5744
I (3692) wifi_init: tcp mss: 624
I (3692) wifi_init: WiFi/LWIP prefer SPIRAM
I (3692) wifi_init: WiFi IRAM OP enabled
I (3702) wifi_init: WiFi RX IRAM OP enabled
I (3732) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (3772) wifi:mode : sta (34:85:18:8f:4d:28) + softAP (34:85:18:8f:4d:29)
I (3772) wifi:enable tsf
I (3772) wifi:Total power save buffer number: 3
I (3772) wifi:Init max length of beacon: 752/752
I (3772) wifi:Init max length of beacon: 752/752
I (3782) mesh: [IO]disable self-organizing<reconnect>
I (4792) wifi:Total power save buffer number: 3
I (5722) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
I (5722) srv_wifi: wifi_config->router.ssid: My_SSID
I (5722) srv_wifi: wifi_config->router.password: My_Passwd
I (5732) srv_wifi: wifi_config->mesh_ap.password: MAP_PASSWD
I (5732) srv_wifi: wifi_config->mesh_ap.max_connection: 6
I (5742) srv_wifi: wifi_config->mesh_id: 119:119:119:119:119:118
I (5752) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
I (5762) mesh: [CONFIG]connect to router:My_SSID, 00:00:00:00:00:00
I (5762) srv_wifi: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (5772) srv_wifi: mesh starts successfully, heap: 8391936, root fixed
I (5792) srv_mobile_network: [IN] Init Mobile_Network
I (7202) esp-netif_lwip-ppp: User interrupt
I (7202) esp_modem_netif: PPP state changed event 5
I (8592) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,)<><>
I (8592) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201    
I (11432) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,)<><>
I (11432) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201  
I (14262) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,)<><>
I (14262) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201  
I (17102) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,)<><>
I (17102) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
```

zhangyanjiaoesp, 2023-11-06:
> > > you tell me to add some random value? If I do this, ROOT will be looking for this connection instead of connecting to PPP
> > 
> > 
> > Since you have set the root to fixed, then the root won't try to connect to router
> 
> @zhangyanjiaoesp Root is fixed, but it continues searching for the wifi network
> 
> ```
> I (3602) wifi:wifi driver task: 3fcc89f4, prio:23, stack:6656, core=0
> I (3602) system_api: Base MAC address is not set
> I (3602) system_api: read default base MAC address from EFUSE
> I (3612) wifi:wifi firmware version: e2d0ad8
> I (3612) wifi:wifi certification version: v7.0
> I (3622) wifi:config NVS flash: enabled
> I (3622) wifi:config nano formating: disabled
> I (3622) wifi:Init data frame dynamic rx buffer num: 64
> I (3632) wifi:Init management frame dynamic rx buffer num: 64
> I (3632) wifi:Init management short buffer num: 32
> I (3642) wifi:Init static tx buffer num: 6
> I (3642) wifi:Init tx cache buffer num: 32
> I (3652) wifi:Init static tx FG buffer num: 2
> I (3652) wifi:Init static rx buffer size: 1600
> I (3652) wifi:Init static rx buffer num: 6
> I (3662) wifi:Init dynamic rx buffer num: 64
> I (3662) wifi_init: rx ba win: 6
> I (3672) wifi_init: tcpip mbox: 32
> I (3672) wifi_init: udp mbox: 6
> I (3672) wifi_init: tcp mbox: 6
> I (3682) wifi_init: tcp tx win: 5744
> I (3682) wifi_init: tcp rx win: 5744
> I (3692) wifi_init: tcp mss: 624
> I (3692) wifi_init: WiFi/LWIP prefer SPIRAM
> I (3692) wifi_init: WiFi IRAM OP enabled
> I (3702) wifi_init: WiFi RX IRAM OP enabled
> I (3732) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
> I (3772) wifi:mode : sta (34:85:18:8f:4d:28) + softAP (34:85:18:8f:4d:29)
> I (3772) wifi:enable tsf
> I (3772) wifi:Total power save buffer number: 3
> I (3772) wifi:Init max length of beacon: 752/752
> I (3772) wifi:Init max length of beacon: 752/752
> I (3782) mesh: [IO]disable self-organizing<reconnect>
> I (4792) wifi:Total power save buffer number: 3
> I (5722) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> I (5722) srv_wifi: wifi_config->router.ssid: My_SSID
> I (5722) srv_wifi: wifi_config->router.password: My_Passwd
> I (5732) srv_wifi: wifi_config->mesh_ap.password: MAP_PASSWD
> I (5732) srv_wifi: wifi_config->mesh_ap.max_connection: 6
> I (5742) srv_wifi: wifi_config->mesh_id: 119:119:119:119:119:118
> I (5752) srv_wifi: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> I (5762) mesh: [CONFIG]connect to router:My_SSID, 00:00:00:00:00:00
> I (5762) srv_wifi: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
> I (5772) srv_wifi: mesh starts successfully, heap: 8391936, root fixed
> I (5792) srv_mobile_network: [IN] Init Mobile_Network
> I (7202) esp-netif_lwip-ppp: User interrupt
> I (7202) esp_modem_netif: PPP state changed event 5
> I (8592) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,)<><>
> I (8592) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201    
> I (11432) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,)<><>
> I (11432) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201  
> I (14262) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,)<><>
> I (14262) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201  
> I (17102) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,)<><>
> I (17102) srv_wifi: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
> ```

you can call APIs in this order to stop reconnection:
```
esp_mesh_fix_root(true);
esp_mesh_set_type(ROOT);
esp_mesh_set_self_organized(false,false);
```

zhangyanjiaoesp, 2023-11-06:
@yanmartins you can change the code as the above comment and test again. If the problem still exists, can you provide a demo for us to test locally? If you can't provide a demo, please provide the IDF branch commit id, we can try to add some debug log for you to test.

zhangyanjiaoesp, 2023-11-07:
Are you using ESP32?

zhangyanjiaoesp, 2023-11-07:
@yanmartins please use the attachment to replace the esp32s3 wifi libs in the `idf/components/esp_wifi/lib/esp32s3` file and retest.
[s3_wifi_lib_1107.zip](https://github.com/espressif/esp-idf/files/13275144/s3_wifi_lib_1107.zip)

If the replacement succeeds, the wifi firmware is :
[image]

And please provide the **whole log (from start to error happen)** of the node device, this will help to analyze the cause of the problem.

yanmartins, 2023-11-07:
@zhangyanjiaoesp 

Succeed:
[teraterm_node.log](https://github.com/espressif/esp-idf/files/13281437/teraterm_node.log)
[teraterm_root.log](https://github.com/espressif/esp-idf/files/13281438/teraterm_root.log)

Error:
[teraterm_node_fail.log](https://github.com/espressif/esp-idf/files/13281441/teraterm_node_fail.log)
[teraterm_root_fail.log](https://github.com/espressif/esp-idf/files/13281443/teraterm_root_fail.log)

I turn off ROOT sometimes in failure scenario to force the error in NODE. With the timestamp I believe it is easy to observe this in both logs

zhangyanjiaoesp, 2023-11-08:
@yanmartins
1. 
> > 3\. The default mesh packet lifetime is BLOCKING(0xffffffff), you can call `esp_mesh_send_block_time()` to set the mesh packet lifetime, then the mesh node will not always show `W (32119) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:13`, it will return TX_TIMEOUT when the lifetime has reached.
> 
> I will add this, but it seems that the error is something else, because the NODE stops looking for ROOT

 Have you ever tried to call the `esp_mesh_send_block_time()` api to change the lifetime of the package? I think this can solve the problem.

2. Where did you call the `esp_mesh_send()` in your application? Do you call it in the `mesh_event_handler()` ?

yanmartins, 2023-11-08:
@zhangyanjiaoesp 
> 1. Have you ever tried to call the `esp_mesh_send_block_time()` api to change the lifetime of the package? I think this can solve the problem.

Yes I tried but the error remained the same

> 2. Where did you call the `esp_mesh_send()` in your application? Do you call it in the `mesh_event_handler()` ?

I'm using the [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example. So, esp_mesh_send() is called in the netif driver like this:

```c
static esp_err_t mesh_netif_transmit_from_node_sta(void *h, void *buffer, size_t len)
{
    mesh_data_t data;
    ESP_LOGD(TAG, "Sending to root, dest addr: " MACSTR ", size: %d", MAC2STR((uint8_t *) buffer), len);
    data.data     = buffer;
    data.size     = len;
    data.proto    = MESH_PROTO_AP; //! Node's station transmits data to root's AP
    data.tos      = MESH_TOS_P2P;
    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS, NULL, 0);

    if(err != ESP_OK) ESP_LOGE(TAG, "Send with err code %d %s", err, esp_err_to_name(err));
    return err;
}
```

zhangyanjiaoesp, 2023-11-09:
sugget set the `esp_mesh_send_block_time()` at least 5s.

yanmartins, 2023-11-09:
> sugget set the `esp_mesh_send_block_time()` at least 5s.

Thanks @zhangyanjiaoesp. It looks like the reconnection problem is gone. 

Just one more question: what causes so many prints like these? Is there some configuration I may have done wrong?

```
[2023-11-09 10:04:14.977] I (58560203) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560213) mesh: <MESH_NWK_ROOT_ADDR>from assoc, layer:2, root_addr:34:85:18:8f:4d:29, root_cap:1
[2023-11-09 10:04:14.992] I (58560223) wifi:Set ps type: 1
[2023-11-09 10:04:14.992] 
[2023-11-09 10:04:14.992] I (58560223) mesh: 2004<arm>parent monitor, my layer:2(cap:6)(node), interval:8757ms, retries:1<normal connected>
[2023-11-09 10:04:14.992] I (58560233) srv_wifi: <MESH_EVENT_PARENT_CONNECTED>layer:2-->2, parent:34:85:18:8f:4d:29<layer2>, ID:01:02:03:04:05:06
[2023-11-09 10:04:14.992] I (58560233) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560253) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560253) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560263) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560263) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560273) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560273) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560283) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.992] I (58560293) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560293) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560303) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560303) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560313) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560313) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560323) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560333) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560333) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560343) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
[2023-11-09 10:04:14.995] I (58560343) mesh: <MESH_NWK_ROOT_ADDR>node, layer:2, root_addr:34:85:18:8f:4d:29, conflict_roots.num:0<>
[2023-11-09 10:04:15.037] I (58560393) srv_wifi: <MESH_EVENT_TODS_REACHABLE>state:0
[2023-11-09 10:04:15.037] I (58560393) srv_wifi: <MESH_EVENT_ROOT_ADDRESS>root address:34:85:18:8f:4d:29
[2023-11-09 10:04:15.037] I (58560403) srv_wifi: <MESH_EVENT_ROOT_ADDRESS>root address:34:85:18:8f:4d:29
```

zhangyanjiaoesp, 2023-11-10:
when the mesh ie of the parent change, there will be `[2023-11-09 10:04:14.995] I (58560333) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>` , this seems normal.

yanmartins, 2023-11-10:
Thank you very much @zhangyanjiaoesp
With all the modifications you asked for, everything is working now.

closed 2023-11-10 by yanmartins