espressif/esp-idf #12432: May gpio_set_level could use in isr?

fzboffice, 2023-10-19:
### Is your feature request related to a problem?

I want a func like this "esp_err_t IRAM_ATTR gpio_set_level"
My idf ver is 4.4.6

fzboffice, 2023-10-19:
My project use mdf ,so I can't use idf5.0

igrr, 2023-10-19:
You should be able to do this as is done here:
https://github.com/espressif/esp-idf/blob/v4.4.6/examples/peripherals/spi_master/hd_eeprom/components/eeprom/linker.lf#L14

More information: https://docs.espressif.com/projects/esp-idf/en/v4.4.6/esp32/api-guides/linker-script-generation.html

fzboffice, 2023-10-20:
@igrr Thanks,I got it.

closed 2023-10-20 by fzboffice