espressif/esp-idf #12433: VHCI returns "hardware error" code 01 event in response to basic HCI commands

OYTIS, 2023-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Espressif SoC revision.

ESP32-D0WD

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

Greetings,

I am trying to use ESP32 as a bluetooth controller with host running on a separate chip connected over SDIO interface and running mynewt nimBLE bluetooth stack. On the start up it sends HCI commands to probe bluetooth controller capabilities, like "Read local supported commands" or "Read local version information", and expects to receive responses to proceed further.

### What is the actual behavior?

Instead it gets "hardware error" responses with error code 01.

E.g. feeding `01:02:20:00` to VHCI - which as I understand should correspond to OGF 0x04 OCF 0x0002 "Read local supported commands" makes VHCI give back `04:10:01:00` which is "Hardware error" event with error code 01. Same happens with "Read local version information" (if I comment out the code that fails on the previous command) and other commands nimBLE issues on the startup. What seems to work though is "Reset command" - feeding `01:03:0C:00` into VHCI returns `04:0E:04:05:03:0C:00`.

So the stack ends up cycling between sending a command, getting no good response and resetting the controller, and only the latter seems to succeed.

Log of esp-hosted v0.4 with BT datapath logging enabled is attached.

### Steps to reproduce.

1. Build and program esp-hosted [release v0.4](https://github.com/espressif/esp-hosted/tree/release/v0.4)
2. Send a command over SDIO with ESP_HCI_IF if_type (e.g).
3. Read a response from the same ESP_HCI_IF saying command is unknown.

### Debug Logs.

```plain
I (518) cpu_start: cpu freq: 240000000 Hz
I (518) cpu_start: Application information:
I (518) cpu_start: Project name:     network_adapter
I (523) cpu_start: App version:      release/v0.4-dirty
I (528) cpu_start: Compile time:     Oct 19 2023 12:26:02
I (533) cpu_start: ELF file SHA256:  925af6fb5...
I (537) cpu_start: ESP-IDF:          v5.2-dev-3318-g8fc8f3f479
I (543) cpu_start: Min chip rev:     v0.0
I (547) cpu_start: Max chip rev:     v3.99 
I (551) cpu_start: Chip rev:         v1.0
I (555) heap_init: Initializing. RAM available for dynamic allocation:
I (561) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (566) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (571) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (576) heap_init: At 3FFD3058 len 0000CFA8 (51 KiB): DRAM
I (581) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (587) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (592) heap_init: At 4009F498 len 00000B68 (2 KiB): IRAM
I (599) spi_flash: detected chip: generic
I (601) spi_flash: flash io: dio
I (605) coexist: coex firmware version: 8770c12
I (609) main_task: Started on CPU0
I (611) main_task: Calling app_main()
I (613) NETWORK_ADAPTER: *********************************************************************
I (619) NETWORK_ADAPTER:                 ESP-Hosted Clean Firmware version :: 0.4                        
I (627) NETWORK_ADAPTER:                 Transport used :: SDIO                          
I (633) NETWORK_ADAPTER: *********************************************************************
I (639) NETWORK_ADAPTER: @Supported features are:
I (643) NETWORK_ADAPTER: - WLAN over SDIO
I (647) ESP_BT: - BT/BLE
I (649) ESP_BT:    - HCI Over SDIO
I (651) ESP_BT:    - BT/BLE dual mode
I (661) BTDM_INIT: BT controller compile version [ec4ac65]
I (663) BTDM_INIT: Bluetooth MAC: 98:cd:ac:6d:4c:8e
I (665) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (969) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 98-cd-ac-6d-4c-8e

I (971) SDIO_SLAVE: Using SDIO interface
I (973) NETWORK_ADAPTER: Start Data Path
I (1973) bt_rx: 0x3ffc4abc   01 03 0c 00                                       |....|
I (1975) bt_tx: 0x3ffd3e05   04 10 01 00                                       |....|
I (1977) bt_tx: 0x3ffd3e02   04 0e 04 05 03 0c 00                              |.......|
I (1985) bt_rx: 0x3ffc52bc   01 02 20 00                                       |.. .|
I (1991) bt_tx: 0x3ffd3e15   04 10 01 00                                       |....|
I (1995) NETWORK_ADAPTER: Initial set up done
I (1997) main_task: Returned from app_main()
I (3585) bt_rx: 0x3ffc5abc   01 03 0c 00                                       |....|
I (3587) bt_tx: 0x3ffd3e02   04 0e 04 05 03 0c 00                              |.......|
I (3591) bt_rx: 0x3ffc62bc   01 02 20 00                                       |.. .|
I (3595) bt_tx: 0x3ffd3e15   04 10 01 00                                       |....|
I (5193) bt_rx: 0x3ffc6abc   01 03 0c 00                                       |....|
I (5193) bt_tx: 0x3ffd3e02   04 0e 04 05 03 0c 00                              |.......|
I (5199) bt_rx: 0x3ffc72bc   01 02 20 00                                       |.. .|
I (5203) bt_tx: 0x3ffd3e15   04 10 01 00                                       |....|
I (6799) bt_rx: 0x3ffc7abc   01 03 0c 00                                       |....|
I (6799) bt_tx: 0x3ffd3e02   04 0e 04 05 03 0c 00                              |.......|
I (6805) bt_rx: 0x3ffc82bc   01 02 20 00                                       |.. .|
I (6809) bt_tx: 0x3ffd3e15   04 10 01 00                                       |....|
I (8405) bt_rx: 0x3ffc8abc   01 03 0c 00                                       |....|
```
```

zhp0406, 2023-10-25:
@OYTIS 
It seems that your command format is incorrect.
[image]

OGF 0x04 OCF 0x0002 ——>
01 **10 02** Supported_Commands

zhp0406, 2023-10-25:
like the "Command: HCI_Read_Local_Supported_Features":
[image]

zhp0406, 2023-10-26:
[2023-10-26 11:23:39.489]
TX：01030c00
[2023-10-26 11:23:40.513]
RX：04 0E 04 05 03 0C 00 
[2023-10-26 11:23:41.719]
TX：01011000
[2023-10-26 11:23:42.748]
RX：04 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03 
@OYTIS this is my log：
[2023-10-26 11:23:47.004]
TX：01021000
[2023-10-26 11:23:48.039]
RX：04 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[2023-10-26 11:23:48.967]
TX：01030c00
[2023-10-26 11:23:49.980]
RX：04 0E 04 05 03 0C 00 
[2023-10-26 11:23:55.113]
TX：01031000
[2023-10-26 11:23:56.128]
RX：04 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87 

----

zhp0406, 2023-10-26:
@OYTIS 
I was unable to reproduce your issue on my end. I tested it using the example, and it seems to be working fine for now. I suggest that you start by testing it with the example and UART. After that, you can proceed with your own testing.

If you still encounter problems, please let me know. I will try to reproduce your issue using other methods.

OYTIS, 2023-10-26:
Hi @zhp0406,

Thank you for checking, I will look at your example. I can see your chip revision is different, `BT Controller compile version` is the same as mine though. Do you know what hardware error 01 can refer to? The code is implementation-specific, and the sources for VHCI implementation don't seem to be available.

zhp0406, 2023-10-31:
@OYTIS 
Hi,have your problem solved?

OYTIS, 2023-11-02:
Hi @zhp0406 ,

Sorry, can't make my hardware take commands from UART, will have to get a separate breakout device to try that demo. But in the end of the day I need to make it work with SDIO, so what I am really interested in is getting more information on what can cause VHCI to return a hardware error, and what code 01 would mean.

OYTIS, 2023-11-03:
Thank you, @zhp0406, I think this particular error is in esp-hosted: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/network_adapter/main/slave_bt.c#L98

I am not sure what wan the intention here, but messing with payload ruins it, so I just deleted those in our project, and it looks much better now.

rahult-github, 2024-02-06:
Closing this issue . Please feel free to reopen , if needed.