espressif/esp-idf #12435: Deleted Task is reported as eReady instead of eDeleted

timoxd7, 2023-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-Wrover-E (Module)

### Power Supply used.

USB

### What is the expected behavior?

Calling `eTaskGetState(taskHandle);` returns `eDeleted` on a deleted Task.

### What is the actual behavior?

Calling `eTaskGetState(taskHandle);` returns `eReady` on a deleted Task.

### Steps to reproduce.

1. Spawn "monitor_task" from app_main
2. Call `vTaskDelete(NULL);` in app_main
3. Do `eTaskState state = eTaskGetState(_taskHandle);` in "monitor task", while _taskHandle is the Handle to the Task of app_main

### Debug Logs.

```plain
I (1265) MONITOR_TASK: Main task state: 2
I (2262) MAIN: Deleting main task now
I (2265) MONITOR_TASK: Main task state: 1
```

0xjakob, 2023-10-20:
@timoxd7 I think using `eTaskGetState()` on an already deleted task is like undefined behavior. Internally, the task handle refers to the task's TCB (task control block). As soon as calling `vTaskDelete()`, that TCB is either deleted directly or can basically be deleted at any time. In other words, any task handle of a deleted task becomes invalid and should not be used.

I don't really understand in which situation `eTaskGetState()` could legitimately be called and return `eDeleted`, though. We'll check this and come back here.

Regarding your `eReady` result: are you sure that the `MAIN` task has deleted itself already when `MONITOR_TASK` is calling `eTaskGetState()`?

igrr, 2023-10-20:
See also https://github.com/espressif/esp-idf/issues/4920.

timoxd7, 2023-10-20:
@0xjakob During normal use, this does not imply any problems. However, while debugging heap memory, while using `heap_caps_get_per_task_info(&heap_info);` to get the information about the Heap per Task usage, the reported Tasks include also deleted tasks. They still have a handle saved in the returned `heap_info.totals[i].task` instead of `NULL`, so it is not possible to check if a task name can be safely obtained or not for reporting. Either `eTaskGetState()` needs to be callable even after the task was deleted and report correctly `eDeleted`, or the returnded data from `heap_caps_get_per_task_info(&heap_info);` needs to set the Task Handle to `NULL` for deleted Tasks.

timoxd7, 2023-10-20:
This line will then have undefined behaviour after deleting a Task: https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/examples/system/heap_task_tracking/main/heap_task_tracking_main.c#L43

0xjakob, 2023-10-23:
@timoxd7 I think you're right. That code looks quite dangerous. We'll take a look at this and come back here.

SoucheSouche, 2023-11-02:
@timoxd7,
I am working on an updated behavior of the task tracking feature. The `heap_task_totals_t` structure no longer includes the `TaskHandle_t` but includes the name of the task and the status of the task (Boolean value to show if the task is running or deleted). In that way you won't need to interact with the task handle itself to get the name of the task.

timoxd7, 2023-11-02:
@SoucheSouche perfect, that sounds great. I still think that the handle might be useful to do other things while the task is still alive to obtain things like stack high watermark to gather information in one loop, but it could be set to NULL if it got deleted or by checking the now included state of the task for eDeleted.

SoucheSouche, 2023-11-02:
You are right, I kept it in `heap_task_stat_t` and set it to null when the task is deleted, but I forgot to propagate it to `heap_task_totals_t`. I will rectify that. Thanks :)

The fix is ready but it is a part of a bigger MR that also adds per task peak memory usage. I will keep you posted when the MR has been merged.

AxelLin, 2024-11-06:
@SoucheSouche How is the status of the fix mentioned in https://github.com/espressif/esp-idf/issues/12435#issuecomment-1790265969 ?

SoucheSouche, 2024-11-06:
The fix will be included as part of a bigger MR. This MR is about to go for second round of review.
The fix on its own cannot exist without the rework of the task tracking feature unfortunately.