espressif/esp-idf #12437: example system/console/advanced_usb_cdc not working

aircable, 2023-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0 and master

### Espressif SoC revision.

ESP32-S3-PICO-1 

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

As the description says: show up with a command prompt and allow enter commands.

### What is the actual behavior?

Connection is only through the internal USB port.
The example is flashed and monitored with the IDF command

### Steps to reproduce.

In the example directory console/advanced_usb_cdc
1. idf.py set-target esp32s3
2. idf.py build
3. idf.py -p /dev/ttyACM0 flash monitor

### Debug Logs.

```plain
Output: but no prompt or any reaction on input

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c406
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce38d8,len:0x11a8
load:0x403c9700,len:0xa5c
load:0x403cc700,len:0x2bec
entry 0x403c98a4
I (0) cpu_start: App cpu up.
I (201) cpu_start: Pro cpu start user code
I (201) cpu_start: cpu freq: 160000000 Hz
I (201) cpu_start: Application information:
I (202) cpu_start: Project name:     console_usb
I (202) cpu_start: App version:      v5.0.4-246-g00d58a92dc
I (202) cpu_start: Compile time:     Oct 20 2023 19:58:31
I (202) cpu_start: ELF file SHA256:  8f1290321078600d...
I (203) cpu_start: ESP-IDF:          v5.0.4-246-g00d58a92dc
I (203) cpu_start: Min chip rev:     v0.0
I (203) cpu_start: Max chip rev:     v0.99 
I (203) cpu_start: Chip rev:         v0.1
I (204) heap_init: Initializing. RAM available for dynamic allocation:
I (204) heap_init: At 3FCA0010 len 00049700 (293 KiB): DRAM
I (204) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (205) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (205) heap_init: At 600FE03C len 00001FAC (7 KiB): RTCRAM
I (216) spi_flash: detected chip: gd
I (217) spi_flash: flash io: dio
W (217) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (218) sleep: Configure to isolate all GPIO pins in sleep state
I (219) sleep: Enable automatic switching of GPIO sleep configuration
I (219) app_start: Starting scheduler on CPU0
I (220) app_start: Starting scheduler on CPU1
I (220) main_task: Started on CPU0
I (230) main_task: Calling app_main()

This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
```
```

### More Information.

Tried many different way on different chips with Release 5.0 and with Master branch.
Often with even less success, like stopping in the boot process not showing debug.
```
I (728) esp_psram: Adding pool of 8178K of PSRAM memory to heap allocator
I (739) spi_flash: detected chip: gd
I (739) spi_flash: flash io: qio
I (739) sleep: Configure to isolate all GPIO pins in sleep state
I (739) sleep: Enable automatic switching of GPIO sleep configuration
I (742) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
```
AND STOP
When typing, this happens: Writing to serial is timing out.

I tried other TinyUSB test examples. printf to USB CDC works, but no input.
There is no example using esp_vfs_dev_cdcacm_register anywhere. 
Can you guy build a good USB console and TinyUSB serial input/output example?

aircable, 2023-10-21:
In addition to the plain example, I added this code.
I couldn't even get it running after the sleep config:
```
        esp_vfs_dev_cdcacm_register();

        fclose( stdin );
        fclose( stdout );
        // switch stdio
        const char *default_cdcacm_dev = "/dev/cdcacm/0";
        _GLOBAL_REENT->_stdin  = fopen(default_cdcacm_dev, "r");
        _GLOBAL_REENT->_stdout = fopen(default_cdcacm_dev, "w");

        if(( _GLOBAL_REENT->_stdin == NULL)||( _GLOBAL_REENT->_stdout == NULL )){
            fprintf( stderr, "stdin/out error\n" );
        } else {
            fprintf( stderr, "stdin/out init\n" );
        }
        printf( "USB finished\n");
```
Debug output:
```
I (215) cpu_start: Pro cpu start user code
I (215) cpu_start: cpu freq: 160000000 Hz
I (215) cpu_start: Application information:
I (215) cpu_start: Project name:     console_usb
I (216) cpu_start: App version:      v5.2-dev-3318-g8fc8f3f479-dirty
I (216) cpu_start: Compile time:     Oct 20 2023 20:33:10
I (216) cpu_start: ELF file SHA256:  7377d521b...
I (216) cpu_start: ESP-IDF:          v5.2-dev-3318-g8fc8f3f479-dirty
I (217) cpu_start: Min chip rev:     v0.0
I (217) cpu_start: Max chip rev:     v0.99 
I (217) cpu_start: Chip rev:         v0.1
I (217) heap_init: Initializing. RAM available for dynamic allocation:
I (218) heap_init: At 3FCA09C8 len 00048D48 (291 KiB): DRAM
I (218) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (218) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (219) heap_init: At 600FE070 len 00001F78 (7 KiB): RTCRAM
I (230) spi_flash: detected chip: gd
I (230) spi_flash: flash io: dio
W (230) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (231) sleep: Configure to isolate all GPIO pins in sleep state
I (232) sleep: Enable automatic switching of GPIO sleep configuration
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```

0xjakob, 2023-10-23:
@aircable I could reproduce your issue. It seems that the example only runs on ESP32-S2. We'll take a look to see what's wrong here.

aircable, 2023-10-23:
Hi @0xjakob, thanks for looking into it. If I can help you with anything, please let me know. I'm keen on using the VFS with esp_vfs_dev_cdcacm_register().

igrr, 2023-10-25:
Hi @aircable, sorry about the issue. When I have initially added support for USB_OTG CDC console on ESP32-S3, I was using a development board with PHY_SEL efuse burned — which made USB_OTG peripheral connected to the internal USB PHY by default. However by default, this efuse is not burned, and a different peripheral (USB_SERIAL_JTAG) is connected to the USB PHY. 
Here's the patch which seems to fix the issue by switching the internal USB PHY to USB_OTG at startup:
[26653.patch.txt](https://github.com/espressif/esp-idf/files/13163625/26653.patch.txt)

aircable, 2023-10-25:
@igrr, thank you for looking into it. I was able to get the console working, but not without modification of the example.
I added:
```
        esp_vfs_dev_cdcacm_register();

        fclose( stdin );
        fclose( stdout );
        // switch stdio
        const char *default_cdcacm_dev = "/dev/cdcacm/0";
        _GLOBAL_REENT->_stdin  = fopen(default_cdcacm_dev, "r");
        _GLOBAL_REENT->_stdout = fopen(default_cdcacm_dev, "w");

        if(( _GLOBAL_REENT->_stdin == NULL)||( _GLOBAL_REENT->_stdout == NULL )){
            fprintf( stderr, "%s: stdin/out error\n", TAG );
        } else {
            //fprintf( stderr, "%s: stdin/out init\n", TAG );
        }
        //printf( "USB finished\n");

        /* Disable buffering on stdin */
        setvbuf(stdin, NULL, _IONBF, 0);
        /* Enable non-blocking mode on stdin and stdout */
        fcntl( fileno( stdout ), F_SETFL, 0);
        fcntl( fileno( stdin ), F_SETFL, 0);    
```
after setting the line endings and also had to remove the dumb mode on the console:
```
    linenoiseSetDumbMode( false );
```
to get the command line working nicely on the USB. In dumb mode, flush does not work.

Otherwise your fix worked.
I think you can close the thread.

closed 2023-11-07 by espressif-bot