espressif/esp-idf #12438: Wrong #if guard for BT_CONTROLLER_DISABLED=y configuration

AxelLin, 2023-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Since the Kconfig allows setting BT_CONTROLLER_DISABLED for Bluetooth Host only usecases,
The #if SOC_ESP_NIMBLE_CONTROLLER is not correct when
CONFIG_BT_CONTROLLER_DISABLED=y (CONFIG_BT_CONTROLLER_ENABLED is not set).

I think all below #if guard
 #if SOC_ESP_NIMBLE_CONTROLLER
should be replaced to
 #if SOC_ESP_NIMBLE_CONTROLLER && CONFIG_BT_CONTROLLER_ENABLED

And also changes all
 #if !SOC_ESP_NIMBLE_CONTROLLER
to
 #if !(SOC_ESP_NIMBLE_CONTROLLER && CONFIG_BT_CONTROLLER_ENABLED)

### What is the actual behavior?

The #if SOC_ESP_NIMBLE_CONTROLLER does not prevent the controller related code to be built even with CONFIG_BT_CONTROLLER_DISABLED=y (CONFIG_BT_CONTROLLER_ENABLED is not set).

### Steps to reproduce.

Build with BT_CONTROLLER_DISABLED=y.

rahult-github, 2023-11-20:
Hi @AxelLin  , thanks . The suggestion of modifying macro cannot be done in all places, since the way the macro SOC_ESP_NIMBLE_CONTROLLER is used at multiple places have little different meaning and hence need to be considered as per code and chipset. 

I think attached patch should help you overcome compilation failures when the support for BLE controller is disabled.  

There is still some work that needs to be done to make this change generic and be pushed into repository though. 

path  --> patch

$IDF_PATH : idf_compilation_fix.txt
$IDF_PATH/components/bt/host/nimble/nimble:  submodule_compilation_fix 

[submodule_compilation_fix.txt](https://github.com/espressif/esp-idf/files/13408166/submodule_compilation_fix.txt)
[idf_compilation_fix.txt](https://github.com/espressif/esp-idf/files/13408168/idf_compilation_fix.txt)

AxelLin, 2023-11-21:
> Hi @AxelLin , thanks . The suggestion of modifying macro cannot be done in all places, since the way the macro SOC_ESP_NIMBLE_CONTROLLER is used at multiple places have little different meaning and hence need to be considered as per code and chipset.

SOC_ESP_NIMBLE_CONTROLLER is used at multiple places have little different meaning?
This is a surprise and difficult to understand.

BTW, my proposed changed is basically very simple from the logic.
SOC_ESP_NIMBLE_CONTROLLER is only meaningful when CONFIG_BT_CONTROLLER_ENABLED=y.
For host-only configuration, the user don't need any code built with SOC_ESP_NIMBLE_CONTROLLER guard.
I think even though SOC_ESP_NIMBLE_CONTROLLER have little different meaning in different places
the above statement is still true.
Actually I had tested my proposed changes, and it works for my host-only setup.

> 
> I think attached patch should help you overcome compilation failures when the support for BLE controller is disabled.

I have tested your patch but it does not work.
I need more time to figure out what is wrong.

AxelLin, 2023-11-21:
BTW,
    choice BT_HOST
          config BT_NIMBLE_ENABLED
            bool "NimBLE - BLE only"
            depends on BT_CONTROLLER_ENABLED   << This dependency is obvious wrong and needs fix first.

No idea why this dependency is changed in v5.2, it was ok in v5.1.
In v5.1, I can set BT_CONTROLLER_ONLY=n && BT_NIMBLE_ENABLED=y to use external BT controller.

AxelLin, 2023-12-10:
> Hi @AxelLin , thanks . The suggestion of modifying macro cannot be done in all places, since the way the macro SOC_ESP_NIMBLE_CONTROLLER is used at multiple places have little different meaning and hence need to be considered as per code and chipset.

Hi @rahult-github 
Could you share the meaning of SOC_ESP_NIMBLE_CONTROLLER in different places?
(I think I need better understanding about SOC_ESP_NIMBLE_CONTROLLER before checking why your patch does not work)

BTW, I can run host+controller_enabled using nimble with C3 and C6 (using esp32c3 or c6's bt controller).
For C3, the SOC_ESP_NIMBLE_CONTROLLER is not defined, for C6 SOC_ESP_NIMBLE_CONTROLLER=1.
So what's the difference? (Why the SOC_ESP_NIMBLE_CONTROLLER is not defined for C3?)

rahult-github, 2023-12-14:
Hi @AxelLin ,  

  The internal BT controller architecture was revamped for new generation of chipsets. So the previous ( viz legacy chipsets - esp32 / c3 / s3 ) need to be differentiated from the newer generation of chipsets. One way to do so was to use this macro which was enabled for relevant SOCs.  If you check the nimble  porting code in IDF, you can see different handling , primarily with npl related functions and usage of dynamic memory instead of static which is part of the new architecture, that is driven by presence of this macro. 

There is some part of the macro usage inside BT controller code which i cannot explain as it is proprietary, but that should not be a concern, since presence or absence of the macro just highlights the chipset being used. 

Coming to the original query in this issue, we will take it up as a request to have standalone ( NimBLE  host only ) mode to work.

AxelLin, 2023-12-24:
> Hi @AxelLin ,
> 
> Coming to the original query in this issue, we will take it up as a request to have standalone ( NimBLE host only ) mode to work.

Then I hope you can support uart hci transport for using external bt.
Most of the required code are already available in nimble.
e.g. ble_hci_uart.c (in nimble/transport/uart) , hal_uart.c (in nimble/porting)

rahult-github, 2024-02-06:
Hi @AxelLin , recently a commit was pushed into [master](https://github.com/espressif/esp-idf/commit/51efb2051655d4197f1bfaef8e68d53ef52b1e7c) to address this . Please help check if the fix works for you now.

closed 2024-02-13 by AxelLin