espressif/esp-idf #12439: Is nested functions supported? Looks no

valkuc, 2023-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi.
I'm trying to use callback functions declared as nested functions and looks like they are not working.
As per GCC documentation it is possible to use nested functions - https://gcc.gnu.org/onlinedocs/gcc/Nested-Functions.html
When I'm trying to do in the similar fashion I'm getting INVALID MEM_TYPE error.

Here is few code snippets.
Working (without nesting):
```
static httpd_req_t *req_glob;
static void test(event_descriptor_t *evt) {
    ESP_LOGI("!!!", "=== %d", evt->type);
}

esp_err_t cgi_get_events(httpd_req_t *req)
{
	req_glob = req;
	event_mgr_list(test, 100);
	return ESP_OK;
}
```

Not working (nested):
```
esp_err_t cgi_get_events(httpd_req_t *req)
{
	void test(event_descriptor_t *evt) {
                ESP_LOGI("!!!", "=== %d", evt->type);
	}

	event_mgr_list(test, 100);
	return ESP_OK;
}
```

I'm using ESP IDF 5.0.4.
@igrr I know you know the answers to any questions there :))

valkuc, 2023-10-21:
Sorry, but example I created is very bad and probably will not demonstrate the problem. Actually, simple nested functions is working fine, but if the nested function has a bit harder logic (for ex., call other functions) - app failing with either `INVALID MEM_TYPE` or `InstructionFetchError` (depend on scenario).

If it will be needed to create a sample app to demonstrate the problem - please ping me.

valkuc, 2023-10-21:
Okay, I figured it out. Until your nested function use any variable defined in parent function - all is fine. But if you decide to use in nested function any variable defined in parent func (or variable from parent func arguments) - you will get `INVALID_MEM_TYPE`.
If this is expected behavior - the ticket can be closed, but anyway it will be nice to mention this behavior in ESP IDF documentation (I didn't find anything in ESP IDF docs regarding nested functions).

0xjakob, 2023-10-23:
@valkuc What is `event_mgr_list()` doing? Is it putting `void test()` on a list to be called later? If yes, there is a pretty big warning on the GCC doc page for nested functions you reference earlier:
> If you try to call the nested function through its address after the containing function exits, all hell breaks loose.

They also explicitly mention out-of-scope variables that are accessed (like `buf` here):
>  If you try to call it after a containing scope level exits, and if it refers to some of the variables that are no longer in scope, you may be lucky, but it’s not wise to take the risk. If, however, the nested function does not refer to anything that has gone out of scope, you should be safe.

This sounds pretty much like what you encountered here. In my opinion, "it’s not wise to take the risk" should be replaced by "do not do this". Referencing out-of-scope variables is undefined behavior.

valkuc, 2023-10-23:
@0xjakob The `event_mgr_list()` calls callback method defined in first parameter in the same thread/task (so, on the same stack). Is the `buf` out-of-scope here? In my understanding - no, because it's defined in `cgi_get_events` and the `test` method is also defined same here.

0xjakob, 2023-10-23:
If the callback is called after `cgi_get_events()` has returned, then `buf` is out-of-scope, because it is only defined inside `cgi_get_events()`. The scope of a variable of the stack is not directly related to the lifetime of a thread/task. Once `cgi_get_events()` has returned, another function in the same thread/task can use that very same memory that was just allocated for `buf`.

valkuc, 2023-10-23:
> If the callback is called after `cgi_get_events()` has returned, then `buf` is out-of-scope

Yep, in this scenario all will go wrong, but in my case all operations happening inside the scope of `cgi_get_events()`.
The flow is like that:
`begin cgi_get_events` --> `event_mgr_list` --> `test` (invoked multiple times by the logic inside event_mgr_list) --> `end cgi_get_events`.

About internals of event_mgr_list. The method use [FlashDB ](https://github.com/armink/FlashDB) `fdb_tsl_iter_reverse` to iterate over data.

0xjakob, 2023-10-23:
> begin cgi_get_events --> event_mgr_list --> test (invoked multiple times by the logic inside event_mgr_list) --> end cgi_get_events.

That looks OK and according to what the documentation says. 

Would you have any minimal reproducing example for this? This would greatly help us analyze the issue to see where the actual error comes from. Furthermore, would you mind letting us know which chip you use?

Thanks!

0xjakob, 2023-10-23:
Is it [this `fdb_tsl_iter_reverse` function](https://github.com/armink/FlashDB/blob/master/src/fdb_tsdb.c#L507) which calls `void test()`?

valkuc, 2023-10-23:
> Is it [this `fdb_tsl_iter_reverse` function](https://github.com/armink/FlashDB/blob/master/src/fdb_tsdb.c#L507) which calls `void test()`?

Yes

valkuc, 2023-10-23:
> That looks OK and according to what the documentation says.
>
> Would you have any minimal reproducing example for this? This would greatly help us analyze the issue to see where the actual error comes [...]

I'm using ESP32-S2-DevKitC-1.
Yes, I will create a reproducible example either today or in the next few days.

valkuc, 2023-10-24:
@0xjakob Here is simple code snippet to reproduce the problem. Just copy paste it in hello-world example.
```
typedef void (*callback_t)(int arg);

void func_which_call_callback(callback_t cb, int limit)
{
	for (int n = 0; n < limit; n++) {
		cb(n);
	}
}

void app_main(void)
{
	char buf[32];
	void callback(int arg) {
		sprintf(buf, "arg=%d", arg);
		ESP_LOGI("!!!", "=== buf: %s", buf);
	}
	func_which_call_callback(callback, 10);
}
```

Pay attention, if you will move `char buf[32];` into `callback` method body - all will be fine.

While I'm preparing this example I catch myself on a mind that looks like this code is not working because the callback method is actually called from the stack of `func_which_call_callback` method and `buf` is not available in this context.

valkuc, 2023-10-24:
UPDATE:
I tried the same example in OnlineGDB - and it's working. Here it is: https://onlinegdb.com/iGiZXmItR

0xjakob, 2023-10-25:
@valkuc The reason this code does not work on our chips is that the compiler generates a piece of executable code on the stack to implement nested functions. This, however, leads to a failure because the stack is marked as non-executable. This happens both on Espressif's Xtensa-based and RISC-V-based chips. So yes, nested functions can not be used in IDF. As you suggested, we'll document this at an appropriate place.

If you use C++, you can use Lambdas. If you use C, you could try to mimic lambdas in C or simply use static functions. If you find any case not covered by the alternatives, please let us know.

jcmvbkbc-issues, 2023-10-25:
> nested functions can not be used in IDF

I would like to clarify this a bit: nested functions cannot be called indirectly through pointers when the stack is not executable. Direct calls to nested functions would still work.

valkuc, 2023-10-25:
Thanks. Great that we figured out that.

closed 2023-11-20 by espressif-bot