espressif/esp-idf #12440: incosistent mbedtls sha256 context/midstate across software/hardware mbedtls sha256 alternative implementations

greenaddress, 2023-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 tdisplay s3

### Power Supply used.

USB

### What is the expected behavior?

Expected behavior: mbedtls_sha256_context state value consistent across esp32/esp32s3 hardware acceleration on/off

I've been testing the idf mbedtls library across esp32 and esp32s3 with both software sha256 and hardware sha256 and I noticed that the state of the sha256 is kept in little endian for esp32 (both software and hardware implementation) while for esp32s3 (and possibly other esp chips) it was kept in big endian for hardware and little endian for software implementation.

Software that relies on accessing the ctx->state data off the mbedtls_sha256_context will break unless it knowns specifically it is running on esp32s3 with hardware acceleration on.

I realize the hw engine may require the state in a particular way but the end user IMHO should be exposed to the correct/consistent state/midstate values across software/hardware.

### What is the actual behavior?

mbedtls_sha256_context state is different on esp32s3 with hw acceleration on.

### Steps to reproduce.

reading ctx->state off a esp32s3 with software vs hw acceleration.

just as an example, a patch like the below fixes it (though __builtin_bswap32 could also be used instead):
 
```
diff --git i/components/mbedtls/port/sha/block/esp_sha256.c w/components/mbedtls/port/sha/block/esp_sha256.c
index eb456afda7..ec61bbd47d 100644
--- i/components/mbedtls/port/sha/block/esp_sha256.c
+++ w/components/mbedtls/port/sha/block/esp_sha256.c
@@ -110,6 +110,14 @@ static void esp_internal_sha_update_state(mbedtls_sha256_context *ctx)
         ctx->sha_state = ESP_SHA256_STATE_IN_PROCESS;
     } else if (ctx->sha_state == ESP_SHA256_STATE_IN_PROCESS) {
         ctx->first_block = false;
+        GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+        GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+        GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+        GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+        GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+        GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+        GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+        GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
         esp_sha_write_digest_state(ctx->mode, ctx->state);
     }
 }
@@ -129,6 +137,14 @@ int mbedtls_internal_sha256_process( mbedtls_sha256_context *ctx, const unsigned
     esp_internal_sha_update_state(ctx);
     esp_sha_block(ctx->mode, data, ctx->first_block);
     esp_sha_read_digest_state(ctx->mode, ctx->state);
+    GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+    GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+    GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+    GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+    GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+    GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+    GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+    GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
     esp_sha_release_hardware();
     return 0;
 }
@@ -184,6 +200,14 @@ int mbedtls_sha256_update( mbedtls_sha256_context *ctx, const unsigned char *inp
             ilen  -= 64;
         }
         esp_sha_read_digest_state(ctx->mode, ctx->state);
+        GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+        GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+        GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+        GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+        GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+        GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+        GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+        GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
 
         esp_sha_release_hardware();
 
@@ -231,7 +255,14 @@ int mbedtls_sha256_finish( mbedtls_sha256_context *ctx, unsigned char *output )
         return ret;
     }
 
-    memcpy(output, ctx->state, 32);
+    PUT_UINT32_BE( ctx->state[0], output,  0 );
+    PUT_UINT32_BE( ctx->state[1], output,  4 );
+    PUT_UINT32_BE( ctx->state[2], output,  8 );
+    PUT_UINT32_BE( ctx->state[3], output, 12 );
+    PUT_UINT32_BE( ctx->state[4], output, 16 );
+    PUT_UINT32_BE( ctx->state[5], output, 20 );
+    PUT_UINT32_BE( ctx->state[6], output, 24 );
+    PUT_UINT32_BE( ctx->state[7], output, 28 );
 
     return ret;
 }
diff --git i/components/mbedtls/port/sha/dma/esp_sha256.c w/components/mbedtls/port/sha/dma/esp_sha256.c
index b2d33b04ac..696a328877 100644
--- i/components/mbedtls/port/sha/dma/esp_sha256.c
+++ w/components/mbedtls/port/sha/dma/esp_sha256.c
@@ -107,6 +107,14 @@ static void esp_internal_sha_update_state(mbedtls_sha256_context *ctx)
         ctx->sha_state = ESP_SHA256_STATE_IN_PROCESS;
     } else if (ctx->sha_state == ESP_SHA256_STATE_IN_PROCESS) {
         ctx->first_block = false;
+        GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+        GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+        GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+        GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+        GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+        GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+        GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+        GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
         esp_sha_write_digest_state(ctx->mode, ctx->state);
     }
 }
@@ -124,6 +132,14 @@ int mbedtls_internal_sha256_process( mbedtls_sha256_context *ctx, const unsigned
     }
 
     esp_sha_read_digest_state(ctx->mode, ctx->state);
+    GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+    GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+    GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+    GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+    GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+    GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+    GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+    GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
     esp_sha_release_hardware();
 
     return ret;
@@ -176,6 +192,14 @@ int mbedtls_sha256_update( mbedtls_sha256_context *ctx, const unsigned char *inp
         }
 
         esp_sha_read_digest_state(ctx->mode, ctx->state);
+        GET_UINT32_BE( ctx->state[0], (unsigned char*)ctx->state,  0 );
+        GET_UINT32_BE( ctx->state[1], (unsigned char*)ctx->state,  4 );
+        GET_UINT32_BE( ctx->state[2], (unsigned char*)ctx->state,  8 );
+        GET_UINT32_BE( ctx->state[3], (unsigned char*)ctx->state, 12 );
+        GET_UINT32_BE( ctx->state[4], (unsigned char*)ctx->state, 16 );
+        GET_UINT32_BE( ctx->state[5], (unsigned char*)ctx->state, 20 );
+        GET_UINT32_BE( ctx->state[6], (unsigned char*)ctx->state, 24 );
+        GET_UINT32_BE( ctx->state[7], (unsigned char*)ctx->state, 28 );
 
         esp_sha_release_hardware();
     }
@@ -222,7 +246,14 @@ int mbedtls_sha256_finish( mbedtls_sha256_context *ctx, unsigned char *output )
         return ret;
     }
 
-    memcpy(output, ctx->state, 32);
+    PUT_UINT32_BE( ctx->state[0], output,  0 );
+    PUT_UINT32_BE( ctx->state[1], output,  4 );
+    PUT_UINT32_BE( ctx->state[2], output,  8 );
+    PUT_UINT32_BE( ctx->state[3], output, 12 );
+    PUT_UINT32_BE( ctx->state[4], output, 16 );
+    PUT_UINT32_BE( ctx->state[5], output, 20 );
+    PUT_UINT32_BE( ctx->state[6], output, 24 );
+    PUT_UINT32_BE( ctx->state[7], output, 28 );
 
     return ret;
 }
```

Harshal5, 2023-10-25:
Hi @greenaddress ,

In the case of ESP32, the SHA hardware peripheral expects and processes the data in big-endian format, whereas, the SHA peripheral of the other chips uses the little-endian format.
We have defined a SOC capability in the case of ESP32, `SOC_SHA_ENDIANNESS_BE` that can be used to differentiate between the SHA peripherals' endianness.

I agree there would be a discrepancy in endianness when accessing the intermediate states of the SHA operation. To maintain consistency, we would need to swap endianness multiple times (whenever `write_digest` and `read_digest` are called), which could lead to a performance hit.

For a normal SHA operation, the intermediate states are not accessed by the use, and thus this is a trade-off between consistency (which normally is not needed) and performance. 

We once came across a case where these intermediate states were accessed and this consistency issue had to be addressed; while porting the fastpbkdf2 library for our system [here](https://github.com/espressif/esp-idf/commit/c82a792bc349f889bdff5a012af1496b6e4f0a39#diff-732de94c4db5354cfa27ecb44a0c741b386255a24b9a3c20428af2880bc59ba2R253-R258). There we handled this endianness swap in its application layer, and rather if we had handled it in our internal layer there would be extremely large number of swaps, leading to a great performance hit.

If possible could you share your use case, would definitely like to discuss and evaluate it.

Harshal5, 2023-11-06:
@greenaddress closing the issue for now, please feel free to re-open if needed.

closed 2023-11-06 by Harshal5

AxelLin, 2023-11-06:
> Hi @greenaddress ,
>
> In the case of ESP32, the SHA hardware peripheral expects and processes the data in big-endian format, whereas, the SHA peripheral of the other chips uses the little-endian [...]

But SOC_SHA_ENDIANNESS_BE does not exist in any release branches, it only available in master tree now.