espressif/esp-idf #12441: ESP32C6 iTWT example auto light sleep unexpected wake ups

GitJomi, 2023-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.
v5.1.1

### Operating System used.
Windows 11

### How did you build your project?
VisualStudio Code with ESP-IDF extension v.1.6.5

### Development Kit.
ESP32-C6-DevKitC-1

### Power Supply used.
Nordic PowerProfiler Kit II (see picture below)

### Project description.
I am using the WiFi itwt example to measure the power consumption during automatic light sleep when connected to an AP using TWT.

### What is the expected behavior?
I expect the ESP32C6 to wake up every 4sec (TWT wake interval) and go to around 35µA current consumption during light sleep and stay at that level for the total of 4sec until the TWT wake interval expires.

### What is the actual behavior?
The TWT interval is set to 4sec and every 4 sec the current consumption peaks can be seen on screenshot_1. 
The devices is also able to go down to 32-35µA during the sleep as expected but not for the entire 4sec period. 

1.	The current measurements show that every 1sec the current consumption rises from 35µA to around 47.5mA peak and 16.4mA average for a duration of around 3.2ms.
(See screenshots below)

2.	Around every 60sec before a TWT wake up an extended period with increased current consumption can be seen. 
Current measurements are 42mA peak and 16mA average for a duration from 500ms to several seconds.
(See screenshots below)

According to the documentation 
„The duration of Light-sleep will be determined by:
-	FreeRTOS tasks blocked with finite timeouts
-	Timers registered with High resolution timer APIs“
I haven’t changed the example code in any way except for changing settings in the sdkconfig to be able to go down to 35µA in auto light sleep and I cannot figure out if either of the two wake up sources triggers these 1sec and 60sec wake ups. 

### Questions:

1. What do I need to change in the example code or the config to prevent these wake ups or are they needed to do some housekeeping under the hood? 
2. How can I see the set light sleep duration and the next wake up trigger?

Many thanks in advance!

### Attachments:
Picture_1 - Hardware setup. ESP32C6 powered via J5 (3.3v) coming from PowerProfiler Kit
Screenshot_1 - 2min power measurements showing the wake ups
Screenshot_2 - Current consumption zoomed in on 1sec wake interval
Screenshot_3 - Current consumption zoomed in on 60sec wake interval
sdkconfig.txt

[image]
[image]
[image]
[image]

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13063202/sdkconfig.txt)

xuxiao111, 2023-10-23:
Hi @GitJomi, thanks for reporting this issue. 
QA1:
The 1-second wakeup event was triggered by lwip timer ARP_TMR_INTERVAL which is in the file "components/lwip/lwip/src/include/lwip/etharp.h". You can change the number from 1000 to 10000 to see the difference. And the 60s wakeup issue has been fixed by this commit (21360b9ef3a2c5747b2effdb3213182db47dd2a7). 
QA2:
Until now, we don't have a way to get the light sleep duration and the next wake up trigger. And we will enable a twt wakeup event, which is currently in progress.

GitJomi, 2023-10-24:
Hi @xuxiao111, thank you for your swift reply.

QA1.1: I can confirm that the 1sec wake ups are caused by the ARP Timer. I tried different timings and it seems that the max timeout is 10000ms can you confirm that? I would like to set it to a longer timeout.
Are you planning to sync the ARP wake up with the TWT wake ups in the future? This way the device would only need to wake up on the TWT intervals and could handle everything that is pending including the ARP tasks. 

QA1.2: Great that this gets fixed. Unfortunately I cannot find the commit based on the key you have provided. Could you post a link to the commit?

QA2: Ok understood, thanks. The twt wakeup event will be very very much appreciated because right now I dont see how the TWT and application wake ups can be synced in a good way. 

Thanks,
Jomi

xuxiao111, 2023-10-24:
Hi @GitJomi, 
QA1.1: In addition to the ARP timer, there is also a DHCP timer (DHCP_COARSE_TIMER_MSECS) which is in the file "components/lwip/lwip/src/include/lwip/dhcp.h". And the default value is 10s, you also need to change it.
QA2: We will do it ASAP.

GitJomi, 2023-10-27:
Hi @xuxiao111 

thank you for the tip to check the DHCP timer. 
When looking into the setting the default is 60sec not 10sec. 
[image]

I changed the DHCP timer it to see if it makes any difference but seems not to. Difficult for me to say what exactly is going on because right now I dont have my WiFi sniffer with me but as soon as I get it I will test in more depth. 

I changed all timer values (ARP, itwt, DCHP to very long timeouts > 60sec to see if the 10sec wake ups are still happening and they do. 
Do you have other ideas what I could check?

Thanks,
Jomi

xuxiao111, 2023-10-27:
Hi @GitJomi, 
"DHCP_COARSE_TIMER_SECS" is defined in the file "components/lwip/port/include/lwipopts.h".
[image]
You can directly modify the definition in this to 60.

GitJomi, 2023-10-27:
Hi @xuxiao111 ,

thanks a lot that was it. Actually I changed that timer in the sdkconfig.h via the menuconfig but what I did not realize is that the max is 10sec and if set to >10sec it gets defaulted to 1sec. 
[image]

Anyway, now it works. Again, thanks for your support.
Jomi

closed 2023-11-03 by GitJomi