espressif/esp-idf #12444: cJSON invalid initializer

tBubus, 2023-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

working cJSON lib

### What is the actual behavior?

i can't get cJSON working...
[image]

example is very simple:

[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

i have even tried to add REQUERIES json in CMakeList, no change

erhankur, 2023-10-22:
@tBubus, the errors you've encountered are a result of incorrect usage of cJSON's APIs. To resolve these issues, you can utilize the following code snippet:

```c
cJSON *root = cJSON_CreateObject();
cJSON_AddStringToObject(root, "Info", "Connected");
const char *ptr = cJSON_PrintUnformatted(root);
if (ptr) {
    ESP_LOGI(TAG, "json: %s", ptr);
    cJSON_free((void *)ptr);
}
```

It's essential to ensure that you include the cJSON header file in the C files where you intend to utilize cJSON's API. There's no need to make modifications to the CMakeLists.txt file.

tBubus, 2023-10-23:
thx, you are right, forgot that pointer 😵

closed 2023-10-23 by tBubus