espressif/esp-idf #12445: unexpected .eh_frame and .eh_frame_hdr sections in ELF prevent binary from running

john30, 2023-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

when CONFIG_ESP_SYSTEM_USE_EH_FRAME is not configured (i.e. not set to `CONFIG_ESP_SYSTEM_USE_EH_FRAME=y`), there are no .eh_frame or .eh_frame_hdr sections in the result ELF file

### What is the actual behavior?

.eh_frame and .eh_frame_hdr sections are part of the result ELF file and as such lead to esptool being unable to merge the sections into a single DROM (becuase there is a gap of 4 bytes in between). As a consequence, 2 DROM sections will be part of the image and flashing such an image will prevent it from running as the bootloader supports only a single DROM section.

in case the undesired .eh_frame* sections are generated, the output of `riscv32-esp-elf-readelf.exe -S output.elf` includes lines similar to these:
```
[17] .eh_frame         PROGBITS        3c1825f0 1835f0 03b130 00   A  0   0  8
[18] .eh_frame_hdr     PROGBITS        3c1bd720 1be720 00b1c4 00   A  0   0  8
```

### Steps to reproduce.

I don't know if a small sized application will lead to the same behaviour. mine is very huge and also includes c++ parts, which I suspect being the root cause in the first place.

other than that, compiling the app regularly works without failures, but flashing it and trying to run it will lead to the bootloader failing with a message of "multiple DROM segments"

### More Information.

changing [sections.ld.in](https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/components/esp_system/ld/esp32c3/sections.ld.in) to not include the two .eh_frame* sections solves the issue reproducably, i.e. my suggestion is to wrap the [relevant parts in of sections.ld.in](https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/components/esp_system/ld/esp32c3/sections.ld.in#L360-L378)
```
#if CONFIG_ESP_SYSTEM_USE_EH_FRAME
...
#endif
```
the same probably should also be done for other targets

0xjakob, 2023-10-23:
@john30 We'll take a look at this.

o-marshmallow, 2023-10-25:
Hello @john30 ,

I was unable to reproduce the issue on my side with the C++ examples. The analysis you made seems to be correct to me: `esptool` cannot merge sections beacuse they are not adjacent.

Keep in mind that `.eh_frame` section is also used by C++ for exception handling, not only by ESP-IDF for backtracing/unwinding. Moreover, the linker script templates `.ld.in` in ESP-IDF are not passed throught the pre-processors, so they don't have access to the configuration macros.

The lines you showed for `.eh_frame` sections does describe adjacent sections:
```
[17] .eh_frame         PROGBITS        3c1825f0 1835f0 03b130 00   A  0   0  8
[18] .eh_frame_hdr     PROGBITS        3c1bd720 1be720 00b1c4 00   A  0   0  8
```
Indeed, `3c1825f0` + `03b130` = `3c1bd720`, so there is no gap in-between.

Can you provide all the lines shown by `readelf -S yourelf.elf` when the problem appears? I understand it may be hard to do but is it possible to try reproducing the issue with a smaller project that I could test? The issue may be due to alignment of somes sections created by some C++ libraries?

o-marshmallow, 2023-11-27:
Closing this issue due to lack of activity, feel free to reopen if you still see this issue.

closed 2023-11-27 by o-marshmallow