espressif/esp-idf #12446: Unable to understand the audio i.e buffer streamed from hands free profile via websocket.

rdev06, 2023-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi I am working on a project to make a **Wifi Headphone**.
For that first I connected with WiFi -> Websocket Server -> Bluetooth as _hfp_hf_ client via _vHCI_ and _Wide Band Speech_ enabled. (I followed this example https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_hf/main).
First I tried to run above example exactly how it is and it was working that means for further change I am missing something...
As suggested, the function `esp_hf_client_register_data_callback(incoming_cb, outgoing_cb);` accepts two callback functions.
My `incoming_cb` looks like below: This function stores buffer data to ring buffer.
```
void incoming_cb(const uint8_t *buf, uint32_t sz)
{
    if (!m_rb)
    {
        return;
    };
    BaseType_t done = xRingbufferSend(m_rb, (uint8_t *)buf, sz, 0);
    if (!done)
    {
        ESP_LOGE(BT_HF_TAG, "rb send fail");
    }

    esp_hf_client_outgoing_data_ready();
}
```
And then I am using `xTaskCreate(confirmWSAndFreeMemo, "udp_receive_task", 2048, NULL, 5, NULL);` which looks like below
```
void confirmWSAndFreeMemo(void *arg)
{
    uint32_t item_size = 0;
    while (1)
    {
        if (!m_rb)
        {
            continue;
        }
        uint8_t *data = xRingbufferReceiveUpTo(m_rb, &item_size, 0, 120);
        if(item_size == 0){
            continue;
        }
        int done = esp_websocket_client_send_bin(ws_client, (const char *)data, item_size, portMAX_DELAY);
        if(done > 0){
            vRingbufferReturnItem(m_rb, data);
        }
        item_size = 0;
    }

    vTaskDelete(NULL);
}
```

Using this logic I able to receive buffer on websocket server, but later I understood after going through documentation the buffer I received was mSBC encoded and looks like they need to be decoded in PCM format. Surprisingly, _I did not found any example over internet to convert this_. And the data which I received in the other end is completely distorted when I tried to decode it with `ffmpeg` using ` ffmpeg -f s16le -i input.msbc output.wav`.
It looks like sound is completely corrupted and distorted. I tried multiple ways but none of them worked.
I am attaching my main logic file **bluetooth.c** and **sdkconfig** for further analysis.
[bluetooth.c.txt](https://github.com/espressif/esp-idf/files/13066657/bluetooth.c.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13066660/sdkconfig.txt)

I am using Node.js server to capture websocket buffer, henceforth I am attaching my **index.js** and **package.json** files as well.
[package.json](https://github.com/espressif/esp-idf/files/13066675/package.json)
[index.js.txt](https://github.com/espressif/esp-idf/files/13066678/index.js.txt)

I understand that I am missing something but don't know what. I am OK to convert that buffer inside esp32 or in node.js server or if I can get any reference of the sample project who is doing same.

rdev06, 2023-10-23:
For mSBC buffer capture the simple index.js file looks like below
[index.js.txt](https://github.com/espressif/esp-idf/files/13066771/index.js.txt)

boblane1, 2024-02-23:
Hi @rdev06,

The audio data received in `incoming_cb` has been decoded in PCM format. Since you have no outgoing audio data, you can delete the invoked function `esp_hf_client_outgoing_data_ready`. Please just play the audio which is received on websocket server to see if this issu is still exist. Thanks.

rdev06, 2024-02-27:
Hi @boblane1 
I tried your suggestion but audio is not playable.
This time I appended the buffer to `input.wav` file.
It will be helpful if I can get any sample project like this.

Source code for my project is [here](https://github.com/rdev06/vTCP)

boblane1, 2024-02-27:
Hi @rdev06,

I have tested your project with example `hfp_ag`, it was ok. You can not just append the pcm data into `input.wav`, since the PCM data is in binary format. You can use the software [`audacity`](https://www.audacityteam.org/download/windows/) to play the raw PCM audio data. BTW, it is better to send audo data into ring buffer and retrieve them in the sending task. You can apply the following patch and try again to see if the issue is still exist.

[test.zip](https://github.com/espressif/esp-idf/files/14416076/test.zip)

closed 2025-03-10 by Alvin1Zhang