espressif/esp-idf #12447: I2C Slave Clock Stretching

AndreCilia, 2023-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am currently in the process of developing a smart battery application that uses SMBus as a slave device.

The workflow for communications is as follows:

1. Master WRITES a one byte command to the Slave
2. Slave holds the SCL line low to indicate that the data is being loaded into the buffer but its not ready yet **(CLOCK STRETCHING)**
3. Slave interprets the received command and loads the data into the buffer
4. Slave releases the SCL line to indicate that the data is now ready in the buffer
5. Master reads out the data in the buffer

The issue at hand is that the I2C slave peripheral of the ESP32-S3 is skipping Step 2 of the above flow and since there is only random data in the buffer it sends out that.

Subsequent requests send out the previously written data.

mythbuster5, 2023-10-23:
That means you are using esp32s3 i2c as slave? not master? I don't understand your step?

AndreCilia, 2023-10-23:
Yes the ESP32S3 is being used as a slave.

mythbuster5, 2023-10-23:
IMO, i2c slave clock stretching only occur in several occasions(and slave on esp32s3 doesn't support stretch in software).

AndreCilia, 2023-10-23:
Hi Simon,

With your comment that "slave on ESP32s3 doesn't support stretch in software" do you mean that this is a limitation from the Espressif library?

If the Espressif library is the issue, if we re-write or modify the library ourselves would the I2C hardware be able to do clock stretching?

Thanks.

mythbuster5, 2023-10-23:
From my understanding. You want a master send a specific command, then slave do a stretch and prepare data and write into fifo. After that, release SCL bus and master reads. So, if you read esp32s3 TRM, you can see our stretch is a hardware-motivated stuff only in few occasions, rather than it can be controlled by software. So maybe this might not satisfy your requirement. 

Our currently software doesn't support the stretch as well. It's more like an error handling on esp32s3.

AndreCilia, 2023-10-30:
Hi Simon,

I am in contact with Espressif directly with regards to this issue. I have no idea in what state the issue is at since I am not getting any reply from who is taking care of the issue when I ask for updates.

If you can refer to ref:!00D7F0yCNM.!500A903xysz:ref please let me know whether there has been any update.

Thanks and regards,
Andre

eccentricOrange, 2024-09-08:
Hey @AndreCilia did you get any updates on this? I'm running into a similar problem as you, and need slave clock stretching.

AndreCilia, 2024-09-24:
> Hey @AndreCilia did you get any updates on this? I'm running into a similar problem as you, and need slave clock stretching.

Hi @eccentricOrange,
Sorry I couldn't answer before.
Espressif gave an unofficial library which works here: 
https://github.com/leeebo/esp_i2c_slave

eccentricOrange, 2024-09-24:
Thanks a lot @AndreCilia ! This is helpful for now, though I'd really like for them to release an official solution

schveiguy, 2025-01-04:
Hm... I made my own i2c_slave device driver, I wish I had seen this, I would have tried it first. This is mine, we are using it in our system with little issue. https://github.com/schveiguy/espi2cslave

IMO, the whole slave driver needs redesigning.