espressif/esp-idf #12448: LightSleep not working after update to v4.4.6 with FREERTOS_IDLE_TIME_BEFORE_SLEEP set to 2

ztefanjo, 2023-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Stable lightsleep function.

### What is the actual behavior?

After update from v4.4.5 to v4.4.6, LightSleep is no longer working, we get "`assert failed: vTaskStepTick tasks.c:2959 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)`" more or less directly when LightSleep is enabled, see attached coredump.

[vTaskStepTick assert.txt](https://github.com/espressif/esp-idf/files/13068748/vTaskStepTick.assert.txt)

If increasing CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP from 2 to 3 (the default value) it seems to work much better, we have not been able to reproduce with this setting.

Note: with earlier v4.4.x releases, we have seen this problem as well, but very, very seldom, see https://github.com/espressif/esp-idf/issues/9797.

### Steps to reproduce.

For us, it happens more or less directly when LightSleep is entered/exited and CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP is set to 2.

### Debug Logs.

```plain
assert failed: vTaskStepTick tasks.c:2959 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)
```

AxelLin, 2024-05-26:
@0xjakob Any update?

sudeep-mohanty, 2025-01-24:
Fix merged at https://github.com/espressif/esp-idf/commit/35cdecb1214c775ef2274280a6dbf05c28e811da. Details of the issue can be found in #9797. Closing this issue. Please feel free to reopen the ticket or open a new one for further support.

closed 2025-01-24 by sudeep-mohanty