espressif/esp-idf #12453: ESP32-C6低功耗电流是怎么才能达到手册上面的180uA?

king592, 2023-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

目前我使用light_sleep例程，电流要440uA,我应该怎么弄才能达到手册上面描述的180uA呢？

king592, 2023-10-24:
然后我使用deep_sleep例程测试，验证一下电路是否有漏电，待机电流为7uA,那么这样是不是能够说明我的电路是正确的
IDF版本为5.1
模组是ESP32-C6-Wroom-1

cuevana8, 2023-10-24:
+1 Like ..[.](https://github.com/flslib/fl-studio-cracked)[@@@](https://github.com/espressif/esp-idf/issues/12453#issue-1958602431)

esp-wzh, 2023-10-24:
Hi~ @king592 

1. 两百多uA的漏电值怀疑是从 IO 上泄露的，可否找颗所有IO都浮空的芯片再次测试？
2. Deepsleep 会关闭外设电源域，其中包括 GPIO，并不能说明电路上没有问题。
3. sdkconfig 是否为默认配置？如果不是能否附上 sdkconfig 文件？

king592, 2023-10-24:
SDK文件是这样的，目前我的模组上面是没有接任何外部电路的，是
否还会出现你说的漏电呢？

king592, 2023-10-24:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13112804/sdkconfig.txt)

king592, 2023-10-24:
@esp-wzh 我这边试了一下找了一颗ESP32-C3的模组，也是跑light_sleep这个例程，功耗是能够达到手册上写的130uA的，实测出来是132uA

huming2207, 2023-10-25:
440uA比标称多260uA这个数看着更像是GPIO在往外漏，如果是某个内部peripheral没休眠一般都会比这个高或者会跳动，如果电流不咋跳，或者呈缓慢（几十几百毫秒，因电容充放电导致的那种）变化，那基本就是GPIO在漏。

总之休眠前把各个GPIO该pull的要pull，确保这个GPIO和外部peripheral之间没有大的电压差（比如一头是3.3v一头0v那肯定就会漏）；或者`gpio_reset_pin()`一下配置成analog模式，不然GPIO脚的能量会漏到外部peripheral上，或者外部peripheral的能量会逆流到GPIO里导致耗电。另外别忘了CS信号要正确配置，比如SPI flash和PSRAM。

huming2207, 2023-10-25:
检查一下休眠前一刻的Flash的CS脚有没有正确配置（要拉高），以及检查一下其他的GPIO上下拉配置

king592, 2023-10-25:
乐鑫这边能否提供一下固件，我直接烧录进去看看电流数据是否正常，模组是ESP32-C6-Wroom-1  M4N4

esp-wzh, 2023-10-25:
Hi, @king592, 

esp32c6 在 light sleep 可配置下电的电源域有 3 个，分别为 CPU 电源域，Modem 电源域和外设电源域

由于light sleep是不会丢失运行上下文的休眠，则在休眠期间如果配置下电某电源域，则需要作相应的初始化以支持进出睡眠时对掉电电源域内外设的上下文进行保存和恢复。

Datasheet 中的 180uA 为在休眠期间下电 CPU 与 Modem 模块时的电流数据。

当前 IDF 如果使用esp_pm组件进行电源管理，是支持自动下电这些电源域的，而[此例程](https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep) 并未支持自动retention。（即当前用户调用esp_light_sleep_start时，CPU 电源域，Modem 电源域和外设电源域并不受menuconfig中的配置控制，这个问题很快会修复）

如果期望在esp_light_sleep_start时下电CPU，可以在之前包含头文件`#include "esp_private/sleep_cpu.h"`并调用`sleep_cpu_configure(true);`， CPU下电后将降低功耗30~40uA。

如果应用不使用wifi/bt等modem模块，可以将`esp-idf/components/esp_hw_support/sleep_modem.c`中的`modem_domain_pd_allowed` 函数直接 `return true` 以允许休眠时下电modem电源域，modem 下电后将降低功耗 90~120uA。

如果应用需使用wifi/bt等modem模块，请参考对应example中的power_save例程，如 [wifi power_save](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save)

esp-wzh, 2023-10-25:
> 乐鑫这边能否提供一下固件，我直接烧录进去看看电流数据是否正常，模组是ESP32-C6-Wroom-1 M4N4

[PowerPerf_firmware.tar.gz](https://github.com/espressif/esp-idf/files/13162165/PowerPerf_firmware.tar.gz)

king592, 2023-10-25:
我这边测试了一下电流分别是，38uA和200uA左右，是基本对得上手册上面的参数了感谢？IDF大约什么时候能够修复这个问题呢？

AxelLin, 2023-11-20:
> Modem 电源域默认不允许下电的问题将在v5.2中修复并backport到v5.1。

When will this fix available in v5.1?

> 
> 但对于用户应用通过 `esp_lightsleep_start` 主动调用的lightsleep，短期内没有支持下电 CPU / Modem / Peripheral 的计划。如果对功耗有要求，请使用 `esp_pm` 组件托管 lightsleep，refer: [Power Management](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/power_management.html)

You should support powerdown CPU / Modem / Peripheral when use esp_lightsleep_start which
is a much simpler use case. (I think this is a BUG if it does not work).

esp-wzh, 2023-11-20:
@AxelLin Thanks for care about this.

> When will this fix available in v5.1?

The fix for master is in internal reviewing and it will backport to v5.1, but it's a big change and involves many peripherals, we need more time to discuss and reviewing.  If everything goes well, it is expected to be fixed on the master this week.

> You should support powerdown CPU / Modem / Peripheral when use esp_lightsleep_start which
is a much simpler use case. (I think this is a BUG if it does not work).

Yes, this is indeed a reasonable request, after discussion, we have added support for this feature into the support plan, but the peripherals of each power domain have different retention strategies. We still need time to abstract the API and organize documents to clearly explain all this.

AxelLin, 2023-11-20:
@esp-wzh 
Do you have update for the peripherals that are not yet supported for Light-sleep Peripheral Power Down?
Which peripherals will be supported in the near future?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/power_management.html#light-sleep-peripheral-power-down

esp-wzh, 2023-11-20:
GDMA/UART1 is in progress, will be supported soon. In the driver of crypto peripherals, the module will be re-initialized for each calculation, so there is no need to support retention for them,  and ETM/TIMG1/ SPI2/I2C/I2S will support soon also,    There are plans to support PCNT/TWAI/LEDC/MCPWM/RMT/SARADC/SDIO/PARL_IO but the priority won't be high. There's no plan to support ASSIST_DEBUG / Trace / USB-Serial-JTAG retention.

AxelLin, 2023-12-19:
> @AxelLin Thanks for care about this.
>
>
> The fix for master is in internal reviewing and it will backport to v5.1, but it's a big change and involves many peripherals, we need more time to discuss and [...]

@esp-wzh I wait yet another month but still don't find this in recent esp-idf update.

esp-wzh, 2023-12-19:
Things didn't go as planned, during the review process, some larger refactoring work was proposed. I don't think it can be integrated into master soon.

I want to know what are the specific problems you encounter when you need this feature, maybe I can provide you with a workaround solution at the application layer.

AxelLin, 2024-01-05:
> GDMA/UART1 is in progress, will be supported soon. In the driver of crypto peripherals, the module will be re-initialized for each calculation, so there is no need to support retention for them, and ETM/TIMG1/ SPI2/I2C/I2S will support soon also, 

But there is no update so far...

esp-wzh, 2024-05-15:
https://github.com/espressif/esp-idf/commit/0c68b9b06c7df83ba3edc1873cc3506e79cbc769 should solved this problem, in applications that do not use a modem peripheral, `modem_domain_pd_allowed` will not block the peripheral power domain from powering down.

AxelLin, 2024-05-15:
> [0c68b9b](https://github.com/espressif/esp-idf/commit/0c68b9b06c7df83ba3edc1873cc3506e79cbc769) should solved this problem, in applications that do not use a modem peripheral, `modem_domain_pd_allowed` will not block the peripheral power domain from powering down.

@esp-wzh 
See the regressions in https://github.com/espressif/esp-idf/issues/13683#issuecomment-2109298560 and https://github.com/espressif/esp-idf/issues/13288#issuecomment-2047151358 .
I'm still waiting for the regression fixes.

esp-wzh, 2024-05-15:
@AxelLin 

The MR to fix issue  https://github.com/espressif/esp-idf/issues/13683#issuecomment-2109298560  has been proposed in the same way as in patch and will be merged soon.

We are debugging the https://github.com/espressif/esp-idf/issues/13288#issuecomment-2047151358 issue, this doesn't seem to be a very tricky issue and will be fixed soon, thank you very much for reporting it, you can fall back to the version that is fine to use!

Indastri, 2024-05-15:
Closing this issue as solved (via [0c68b9b](https://github.com/espressif/esp-idf/commit/0c68b9b06c7df83ba3edc1873cc3506e79cbc769)). If the issue will persist feel free to reopen this or create a new issue.

closed 2024-05-15 by Indastri